zephyrproject-rtos/zephyr #66298: Bluetooth: conn.c: deferred_work() blocks system work queue

BryanZhuAM, 2023-12-08:
**Describe the bug**:
I am developing a vendor specific HCI interface and inside it I am using system work queue to delay HCI interrupt read opeartion(in this work item, I am using I2C driver to read some status from controller device at first), and found the 0x2013(BT_HCI_OP_LE_CONN_UPDATE) HCI command is getting timeout after connection.
Fail rate is 100%.
I tried to create a dedicate HCI RX WQ instead of system WQ to handle this HCI interrupt, and do not meet this issue.

**Root cause**
After connection some operations are submited in a delayed work item in system WQ(in conn.c), the work handler is static void deferred_work(struct k_work *work);
in this handler, it calls some blocking APIs: send_conn_le_param_update(), which calls bt_hci_cmd_send_sync(), this send_sync() API waits for command response, and as a result it blocks system work queue, due to this reason, my HCI RX operation submitted into system WQ never get invoked and caused timeout.

**To Reproduce**
Sample used: samples:bluetooth:peripheral_hr
Need to rework hci driver to execute RX operation in a system work queue item

**Expected behavior**
Expected system work queue is not blocked due to work item handler's blocking call

**Impact**
Due to system WQ is a global WQ for all user's usage, This may introduce risk to system WQ's general usage in anywhere after enabled BT stack

**Logs and console output after connection and wait seconds for triggering the issue**
...
[00:00:22.161,071] [0m<dbg> bt_hci_core: hci_tx_thread: Calling k_poll with 3 events[0m

[10:19:11.741]收←◆ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:331
	command opcode 0x2013 timeout with err -11
[00:00:31.316,101] [1;31m<err> os: esf_dump: r0/a1:  0x00000003  r1/a2:  0x00000000  r2/a3:  0x00000002[0m
[00:00:31.316,101] [1;31m<err> os: esf_dump: r3/a4:  0x10010ee8 r12/ip:  0x00000010 r14/lr:  0x0001c8d3[0m
[00:00:31.316,131] [1;31m<err> os: esf_dump:  xpsr:  0x41000000[0m
[00:00:31.316,162] [1;31m<err> os: esf_dump: Faulting instruction address (r15/pc): 0x0001c8de[0m
[00:00:31.316,223] [1;31m<err> os: z_fatal_error: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0[0m
[00:00:31.316,284] [1;31m<err> os: z_fatal_error: Current thread: 0x10011a48 (unknown)[0m
[00:00:31.323,181] [1;31m<err> os: k_sys_fatal_error_handler: Halting system[0m

**Suggested fix**
I would like to suggest to create a BT stack dedicated WQ instead of using system WQ or maybe you have a better idea.

henrikbrixandersen, 2023-12-19:
@jhedberg Please triage this in the Bluetooth WG meeting.

jhedberg, 2023-12-21:
I've recategorised this as an enhancement. It's true that the Zephyr Bluetooth host stack relies on the system workqueue for various functions right now, and may e.g. do a `bt_hci_cmd_send_sync()` from that context (like you've seen with LE_Conn_Update).

To improve the situation we should look at the overall design of the host stack and see if we can completely stop using shared system threads (like the system work queue). This should be behind a Kconfig option however, since the intention of the current design is to save memory when possible. Another way to improve things would be to add support for asynchronous completion of HCI commands, i.e. the ability to have a callback for the command completion status and parameters.

In the short term, I'd suggest to follow the same design solutions that existing HCI drivers do, i.e. use a normal RX thread rather than the system workqueue to process data from the controller.

BryanZhuAM, 2023-12-22:
> I've recategorised this as an enhancement. It's true that the Zephyr Bluetooth host stack relies on the system workqueue for various functions right now, and may e.g. do a `bt_hci_cmd_send_sync()` [...]

@jhedberg Good to know your are taken this behavior into consideration, and change options are reasonable while I was just think how to fix it as simple as possible.
and yes, I am using a private thread for my HCI driver to avoid this blocking.

carlescufi, 2024-01-04:
@BryanZhuAM we've discussed this in the weekly Bluetooth meeting. While this issue is known and actually present in others subsystem (USB is a good example), we do want to address it mid-term. In the meantime, in your particular case, you may be able to get away with using the sys workq in your driver if you disable `CONFIG_BT_GAP_AUTO_UPDATE_CONN_PARAMS`.

BryanZhuAM, 2024-01-05:
> @BryanZhuAM we've discussed this in the weekly Bluetooth meeting. While this issue is known and actually present in others subsystem (USB is a good example), we do want to address it mid-term. In [...]

@carlescufi Thanks for reminder, I got it.