zephyrproject-rtos/zephyr #66306: stm32h7: the CS line in the case of spi_write is set to a high state before sending all the data.

robertkowalewski, 2023-12-08:
**Description**
Yesterday I encountered a strange issue when trying to configure BMI323 sensor using [STM32H743ZI](https://www.st.com/en/microcontrollers-microprocessors/stm32h743zi.html) MCU. I was able to read data from sensor, but wasn't able to write any data to it (in case of writing data to sensor, we send 3 bytes to it and don't expect any data from sensor), after reading written data back I got defaults. Along with my colleague, we connected an oscilloscope and noticed that CS line was set to high state before sending all data on MOSI line. After some debugging we found that function `spi_stm32_shift_m` are different for stm32h7 family and others (as `spi_stm32_can_use_fifo` returns `true` for stm32h7 family). It looks like driver doesn't wait until all data is sent, and go to next steps (like changing CS line state).

**Possible workarounds**

1. Changed `spi_stm32_can_use_fifo` to return always `false`.
2. Changed `bosch_bmi323_spi_write_words` to expect 3 bytes to be read.

Both workarounds made that we're able to write data to sensor (confirmed by reading written data back).

**To Reproduce**
Integrate BMI323 driver into your app. My **overlay** for **nucleo_h743zi.dts**:
```
&spi1 {
    status = "okay";

    bmi323: bmi323@0 {
        compatible = "bosch,bmi323";
        reg = <0>;
        spi-max-frequency = <8000000>;
        int-gpios = <&gpiof 3 GPIO_ACTIVE_HIGH>;
    };
};
```
You may add additional code in `bosch_bmi323_enable_feature_engine` (bmi323.c) to read back written registers.

**Expected behavior**
Expectation is that CS line will be changed to high state after all 3 bytes on MOSI line are sent.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Zephyr main

erwango, 2023-12-11:
Could it be linked to https://github.com/zephyrproject-rtos/zephyr/issues/66326 ?

GeorgeCGV, 2023-12-11:
@erwango CS is a part of the problem in that issue case. But PR is required for CS relocation.

PR #66381 (2023-12-13): drivers: spi: stm32: disable cs only after spi is done
Move CS control to de-assert the CS line after the SPI is done.

The issue has been discovered in #66326. This PR applies a fix proposed by @dgastonochoa (_validated in #66326_).

Fixes #66306.
closed 2023-12-13 by carlescufi