zephyrproject-rtos/zephyr #66309: nucleo_h563zi: SRAM3 cannot be used and memory corruption when using SRAM1 and 2

alelec13, 2023-12-08:
I'm trying to relocate a buffer in SRAM3 on my nucleo_h563zi.

I've followed : https://docs.zephyrproject.org/latest/kernel/code-relocation.html

Steps to reproduce the behavior:
1. CONFIG_CODE_DATA_RELOCATION=y in prj.conf.
2. zephyr_code_relocate(FILES src/firmware.c LOCATION SRAM3) in cmakelists
3. firmware.c only containing: uint8_t firmware_image[10] = {0,1,2,3,4,5,6,7,8,9};

**Expected behavior**
Compilation successfull, buffer stored in SRAM3 and being able to read it.

**Impact**
I'm trying to setup firmware update on the dev board so I need to store the fw image on SRAM3 as it is the only one big enough. Later I'm gonna use an external flash to buffer the image.

**Logs and console output**
linker_zephyr_pre0.cmd:252 cannot move location counter backwards (from 2009bce0 to 20050000)

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain Zephyr SDK
 - Commit : ed610400bcb74338d25e4b29f948709efe12f3a3

**Additional context**
It's compiling correctly if I set SRAM1 or 2. 

The problem then is that if I declare for example my buffer like this at compile time :
uint8_t firmware_image[10] = {0,1,2,3,4,5,6,7,8,9};
When I read it then the output is: 
[00:00:30.005,000] <inf> main: fw[0] = 00
[00:00:30.005,000] <inf> main: fw[1] = 01
[00:00:30.005,000] <inf> main: fw[2] = 02
[00:00:30.005,000] <inf> main: fw[3] = 03
[00:00:30.005,000] <inf> main: fw[4] = 00
[00:00:30.005,000] <inf> main: fw[5] = 80
[00:00:30.005,000] <inf> main: fw[6] = E1
[00:00:30.005,000] <inf> main: fw[7] = A4
[00:00:30.005,000] <inf> main: fw[8] = F9
[00:00:30.005,000] <inf> main: fw[9] = 32

If I set it at runtime:
for (uint32_t i = 0U; i <10; i++) {
        firmware_image[i] = i;
}

then when reading it:
[00:00:30.005,000] <inf> main: fw[0] = 00
[00:00:30.005,000] <inf> main: fw[1] = 01
[00:00:30.005,000] <inf> main: fw[2] = 02
[00:00:30.005,000] <inf> main: fw[3] = 03
[00:00:30.005,000] <inf> main: fw[4] = 04
[00:00:30.005,000] <inf> main: fw[5] = 05
[00:00:30.005,000] <inf> main: fw[6] = 06
[00:00:30.005,000] <inf> main: fw[7] = 07
[00:00:30.005,000] <inf> main: fw[8] = 08
[00:00:30.005,000] <inf> main: fw[9] = 09

So 2 problems : 
SRAM3 not compiling
memory corrupted if initialised at compile time

erwango, 2023-12-12:
About the linkage issue, it has something to do with the stm32h5 sections.ld script.
Compilation completes if you comment following line:
https://github.com/zephyrproject-rtos/zephyr/blob/b84bab36b94164b87257894b3de873f60bcee5e3/soc/arm/st_stm32/stm32h5/CMakeLists.txt#L7

erwango, 2023-12-12:
With this modification, I've modified `samples/application_development/code_relocation_nocopy/src/sram_code.c` and it works as expected. Can you have a check using following branch?
https://github.com/erwango/zephyr/tree/wip_h5_linker_script

alelec13, 2023-12-12:
It compiles now but I declared a buffer of 100KB and it seems to be stored in RAM and not in SRAM3.

[image]

Same when I try to declare it in the other SRAM

erwango, 2023-12-12:
Works on my side:
```
Memory region         Used Size  Region Size  %age Used
        EXTFLASH:          48 B 18446744073577446232 B      0.00%
           FLASH:       33724 B         2 MB      1.61%
             RAM:        6360 B       256 KB      2.43%
           SRAM1:          0 GB       256 KB      0.00%
           SRAM2:          0 GB        64 KB      0.00%
           SRAM3:         272 B       320 KB      0.08%
        IDT_LIST:          0 GB         2 KB      0.00%
```

(don't mind the EXTFLASH)

alelec13, 2023-12-12:
Sorry my bad... Forgot to re-enable CONFIG_CODE_DATA_RELOCATION as I used RAM while waiting for a fix...

Works good to me ! 👍