zephyrproject-rtos/zephyr #66312: tests: crypto: rand32 test issue regression

Desvauxm-st, 2023-12-08:
Describe the bug:

in our bench tester many board test fail now  with:

tests/crypto/rand32
for example:

west build -p always -b   nucleo_f746zg  tests/crypto/rand32
[image]

test is blocked

it is due to this commit:
https://github.com/zephyrproject-rtos/zephyr/commit/78af1c1e3878c410067bd49846771300fa86b840

entropy: stm32: suspend when pool is full
78af1c1e3878c410067bd49846771300fa86b840
author: Dawid Niedzwiecki <dawidn@google.com>  2023-12-01 11:27:37
Committer: Maureen Helm <maureen.helm@analog.com>  2023-12-05 23:27:53

thanks  to have a look

niedzwiecki-dawid, 2023-12-11:
I've sent a fix to review. Would you be able to confirm it works? 

Sorry for the bug.

closed 2023-12-14 by fabiobaltieri

niedzwiecki-dawid, 2024-05-09:
@Desvauxm-st sorry for a late question. Are the results of your test runs available somewhere or it is an internal thing?

erwango, 2024-05-13:
@niedzwiecki-dawid Tests results might be available here: https://github.com/zephyrproject-rtos/test_results/
Updated once/week. 
Regarding internal archives, they are not deep enough to be able to retrieve the info that late. Would be easier to re-run test

niedzwiecki-dawid, 2024-05-13:
Thanks for the info. I don't need archive of this test, just asking about access to test results :).

niedzwiecki-dawid, 2024-05-14:
One more question, is there any easy way to read the result? E.g. If I would like to get 5 last results of a specific test for a specific board. Or I just need to grep commits.

erwango, 2024-05-14:
TBH, I don't know. @hakehuang might be able to answer as Testing group chairman

PR #66373 (2023-12-14): stm32 entropy fix
Enable the RNG module for the get_entropy_isr function if needed.

Fixes #66312