zephyrproject-rtos/zephyr #66315: linker: support linking vector table etc. into user-defined sections

raffi-g, 2023-12-08:
## Introduction

I would like to be able to link certain blocks, e.g. the vector table or `isr_wrapper` to a section of my choice. For example linking the vector table into ram, could speed up irq execution.

### Problem description

Currently these blocks are linked using the cmake function `zephyr_linker_sources`, resp. `zephyr_linker_sources_ifdef`. However, these are hard-coded into architecture-specific or even common CMakeLists.txt files (e.g. arch/common/CMakeLists.txt). I see no possibility to "undo" these steps in my application and link them somewhere else. (except of course to not use these generated linker snippet files at all in my application specific ld file...)

### Proposed change

I mainly see 2 solutions:

- add a kconfig which controls whether these blocks are linked at all. This kconfig would be `default y` of course. If I set it to `n` in my app, I'm then resposible to put them into my own CMakeLists.txt
- add a kconfig with an enum. This would then control where the blocks are linked, so `ROM_START`, `RAMFUNC_SECTION`, etc.

These kconfigs would then have to exist for every block, we want to control like this. So the vector table, `isr_wrapper` and possibly others. This is true at least for the second approach. For the first we could also just have one kconfig for vector table and wrapper together.

For simplicity, I would go with the first one.

nordicjm, 2023-12-08:
`CONFIG_SRAM_VECTOR_TABLE` ?

raffi-g, 2023-12-08:
> `CONFIG_SRAM_VECTOR_TABLE` ?

yes, something like this. But as far as I can tell, `CONFIG_SRAM_VECTOR_TABLE`  is only implemented for the stm32f0.

And I would like to be even more free in placing these blocks. 

In my specific example, I have tightly coupled memory (itcm and dtcm, resp. ilm and dlm). I ultimately want to place the isr wrapper into itcm. And because the vector table in my case is an array of addresses rather than instructions, it should go into the dtcm.

raffi-g, 2023-12-14:
> In my specific example, I have tightly coupled memory (itcm and dtcm, resp. ilm and dlm). I ultimately want to place the isr wrapper into itcm. And because the vector table in my case is an array of addresses rather than instructions, it should go into the dtcm.

btw the support for itcm and dtcm is added in PR https://github.com/zephyrproject-rtos/zephyr/pull/66254

raffi-g, 2023-12-22:
I found another solution: I adjusted the cmake function `zephyr_linker_sources` s.t. you can actually call it multiple times with the same ld files. It then only uses the last occurrence for each ld file and removes everything else again.

Like this I can simply call `zephyr_linker_sources` again in my app-specific CMakeLists.txt at the very end.

PR is under way.

PR #66889 (2024-01-26): cmake: zephyr_linker_sources: Link each file only once
If an ld file is included multiple times using `zephyr_linker_sources`, only the last occurrence is actually used. Everything else is removed again from the generated snippet files.

This allows to relocate certain blocks, e.g. the vector table, in an application-specific CMakeLists.txt (or for an arch or soc), thereby solving my problem statet in #66315.

I think, this would also make  `CONFIG_SOC_IT8XXX2_EXCEPTIONS_IN_RAM `  introduced in ab2515ad267a2caac70ed19f79ab3d1967d95251 obsolete.

closes #66315
commit e78aacdf (2024-01-26): cmake: zephyr_linker_sources: Link each file only once

If an ld file is included multiple times using `zephyr_linker_sources`,
only the last occurrence is actually used. Everything else is removed
again from the generated snippet files.

This allows to relocate certain blocks, e.g. the vector table, in an
application-specific CMakeLists.txt (or for an arch or soc).

Fixes: #66315
closed 2024-01-26 by carlescufi