zephyrproject-rtos/zephyr #66322: hello world sample broken with llvm toolchain

tguenth, 2023-12-08:
**Describe the bug**
Building the hello world sample with llvm toolchain (shipped with docker image zephyrprojectrtos/ci:v0.26.6) breaks on linking.

compilation succeeds until linking

reproduce with following docker file
``` dockerfile
FROM zephyrprojectrtos/ci:v0.26.6

RUN west init -m https://github.com/zephyrproject-rtos/zephyr
RUN west update -o=--depth=1 -n

# OK: build hello world with zephyr sdk
RUN west build -b qemu_cortex_m3 -p always zephyr/samples/hello_world

# FAIL: build hello world with llvm
# (TOOLCHAIN)_TOOLCHAIN_PATH -
#   https://docs.zephyrproject.org/latest/develop/env_vars.html#envvar-ZEPHYR_TOOLCHAIN_VARIANT
# > set ZEPHYR_TOOLCHAIN_VARIANT=llvm
# > set LLVM_TOOLCHAIN_PATH
RUN LLVM_TOOLCHAIN_PATH=/usr/lib/llvm-${LLVM_VERSION}/ \
    west build -b qemu_cortex_m3 -p always zephyr/samples/hello_world -- \
    -DZEPHYR_TOOLCHAIN_VARIANT=llvm -DCONFIG_LLVM_USE_LLD=y
```

**Expected behavior**
compilation should pass

**Logs and console output**
```
 > [5/5] RUN LLVM_TOOLCHAIN_PATH=/usr/lib/llvm-16/     west build -b qemu_cortex_m3 -p always zephyr/samples/hello_world --     -DZEPHYR_TOOLCHAIN_VARIANT=llvm -DCONFIG_LLVM_USE_LLD=y:
0.629 -- west build: making build dir /build pristine
0.705 -- west build: generating a build system
0.717 Loading Zephyr default modules (Zephyr base).
0.717 -- Application: /zephyr/samples/hello_world
0.717 -- CMake version: 3.22.1
0.978 -- Found Python3: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter
0.989 -- Cache files will be written to: /root/.cache/zephyr
0.993 -- Zephyr version: 3.5.99 (/zephyr)
1.012 -- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
1.528 -- Board: qemu_cortex_m3
1.634 -- Found toolchain: host (clang/ld)
1.636 -- Found Dtc: /usr/bin/dtc (found suitable version "1.6.1", minimum required is "1.4.6")
1.636 -- Found BOARD.dts: /zephyr/boards/arm/qemu_cortex_m3/qemu_cortex_m3.dts
1.788 -- Generated zephyr.dts: /build/zephyr/zephyr.dts
1.788 -- Generated devicetree_generated.h: /build/zephyr/include/generated/devicetree_generated.h
2.111 -- Including generated dts.cmake file: /build/zephyr/dts.cmake
3.023 Parsing /zephyr/Kconfig
3.023 Loaded configuration '/zephyr/boards/arm/qemu_cortex_m3/qemu_cortex_m3_defconfig'
3.023 Merged configuration '/zephyr/samples/hello_world/prj.conf'
3.023 Merged configuration '/build/zephyr/misc/generated/extra_kconfig_options.conf'
3.023 Configuration saved to '/build/zephyr/.config'
3.023 Kconfig header saved to '/build/zephyr/include/generated/autoconf.h'
3.262 -- Found LlvmLld: /usr/lib/llvm-16/bin/ld.lld (found suitable version "16.0.6", minimum required is "14.0.0")
3.333 -- The C compiler identification is Clang 16.0.6
3.370 -- The CXX compiler identification is Clang 16.0.6
3.437 -- The ASM compiler identification is Clang with GNU-like command-line
3.438 -- Found assembler: /usr/lib/llvm-16/bin/clang
7.596 -- Using ccache: /usr/bin/ccache
7.643 -- Configuring done
...
10.36 [118/127] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mutex.c.obj
10.42 [119/127] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sched.c.obj
10.46 [120/127] Linking C static library zephyr/kernel/libkernel.a
10.50 [121/127] Linking C executable zephyr/zephyr_pre0.elf
10.50 FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /build/zephyr/zephyr_pre0.map
10.50 : && ccache /usr/lib/llvm-16/bin/clang --target=armv7m-none-eabi  --specs=nosys.specs zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/soc/soc/arm/ti_lm3s6965/libsoc__arm__ti_lm3s6965.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L  -L/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  --config  /zephyr/cmake/toolchain/llvm/clang_libgcc.cfg  -fshort-enums  -mcpu=cortex-m3  -mthumb  -mabi=aapcs  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -fuse-ld=lld  -Wl,--orphan-handling=warn  -Wl,--no-pie && cd /build/zephyr && /usr/bin/cmake -E true
10.50 clang: warning: argument unused during compilation: '--specs=nosys.specs' [-Wunused-command-line-argument]
10.50 ld.lld: error: unable to find library -lgcc
10.50 clang: error: ld.lld command failed with exit code 1 (use -v to see invocation)
10.50 ninja: build stopped: subcommand failed.
10.50 FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /build
```

nordicjm, 2023-12-08:
Doesn't seem like a bug, the docker image isn't shipped with llvm for arm targets, unless @stephanosio says otherwise

tejlmand, 2023-12-11:
this is not a bug.

you would need a runtime library for cross-compiling to arm.
With llvm, that would be `compiler-rt`, however you must build compiler-rt yourself.

For a bit more info as to why this is the case, you may checkout this thread: https://discord.com/channels/636084430946959380/636732781086638081/1144584586387538002

Note, it is planned to create a small doc page on how to compile a basic / bare-metal compiler-rt lib for a simple Zephyr use-case, however distributing a compiler-rt with Zephyr has until recently been challenging due to lack of multilib support in llvm.
(multilib support seems to have been included with llvm 18, but I have yet to look into the details and new possibilities)

tguenth, 2023-12-14:
> this is not a bug.
>
> you would need a runtime library for cross-compiling to arm. With llvm, that would be `compiler-rt`, however you must build compiler-rt yourself.
> For a bit more info as to why [...]
 
Thanks for the clarification. Would you mind checking the URL of the discussion on Discord? Unfortunately, I can't see the discussion there, even though I've already joined the zephyrproject discord server.

> Note, it is planned to create a small doc page on how to compile a basic / bare-metal compiler-rt lib for a simple Zephyr use-case, however distributing a compiler-rt with Zephyr has until recently [...]

tejlmand, 2024-01-03:
> Would you mind checking the URL of the discussion on Discord? Unfortunately, I can't see the discussion there, even though I've already joined the zephyrproject discord server.

because the discussion is on the LLVM discord server, not the Zephyr one.
For ease, i've copied some of the comments here:
From https://discord.com/channels/636084430946959380/636732781086638081/1144584586387538002
> carles — 08/25/2023 12:50 PM
 Hi all, posted this on Discourse before I realized there is a Discord server for LLVM (also, there seems to be no compiler-rt channel so posting it here instead).
Is there a reason why there seems to be no binary distribution of the compiler-rt builtins (libclang_rt.builtins-<arch>.a) for all supported architectures? The reason I ask is that clang is a cross-compiler out of the box, but without the builtins one typically cannot link. Is the user expected to build their own copy of the builtins for their target arch? Note: I do realize that the builtins are provided in the default clang/llvm distro, but only for the host arch. 
tobiashieta — 08/25/2023 2:04 PM
`@carles` yeah to build the runtime's for a specific target you need to have a copy of the headers for that target.
`@carles` so maintaining a set of "correct" headers for different platforms not knowing what you want to target is pretty timeconsuming and hard.
`@carles` so I say it's expected that you'll do that yourself for something that fits your usecase.
carles — 08/25/2023 2:14 PM
`@tobiashieta` thanks for the response. Note that I am talking about a bare metal target here, Linux is just the host. Aren't the headers needed to build the runtime part of the regular LLVM/Clang source distro?
tobiashieta — 08/25/2023 2:16 PM
`@carles` I think the current set of binaries are just a easy way to get into clang / llvm for the mainstream platform it's compiled for. if you want anything more than just a quick start you would need to setup your own compile. Which I recommend in any case to get the highest performance and it targeted specifically for what you need.