zephyrproject-rtos/zephyr #66326: SPI devices fail after H7 SPI FIFO introduction

GeorgeCGV, 2023-12-08:
The SPI devices started to fail (_periodically_) on STM32H730 after the merge of #63173. 

The SPI driver gets stuck in `spi_stm32_complete` after data reception. The `CR2` is `0`, obviously the `SPI_CR2_TSIZE` is `0` as well.
The `SR` (_checked in `LL_SPI_IsActiveFlag_TXC`_) is usually `0`, but there were cases of it being `2`. The `TXC` bit is `0`.
Because `TXC` is `0`, the `ll_func_spi_is_busy` returns `true` and it loops forever. At that stage nothing happens on the SPI lines.

Different SPI peripherals fail at different times. To trigger the issue the board has to be restarted multiple times. When it gets stuck it is always within `spi_stm32_complete`. The context rx/tx count, len are 0 as well as rx/tx buffers are set to 0x0. Sometimes it works, but then devices fail due to corruption of transmitted and/or received data.

After reducing the clock the issue started to appear every time. With higher clock the issue required board restart to trigger it. Sized down connected zoo to one device per SPI (_to eliminate possible issue(s) from having multiple devices_).

SPI without FIFO, the device works:
[image]

SPI with FIFO, the device doesn't work and it is stuck as described above (_within `ll_func_spi_is_busy`, `SR=0`_):
[image]

It is clearly visible that the CS (_chip select_) is dramatically off. It is software controlled by the driver. Most probably that is where the problems are coming from.
 
**Impact**
The #63173 breaks previously working SPI. At least on H730.

**Environment:**
- Toolchain: arm gnu toolchain 12.3 REL1
- Zephyr version is pretty much up to date: [f69641f7d204864aa26f8bdd9fecab259e535da2](https://github.com/zephyrproject-rtos/zephyr/commit/f69641f7d204864aa26f8bdd9fecab259e535da2).

**Additional context**
It is a custom device based on H730. Multiple SPIs are used where each has multiple targets (_slaves_).
I still have to check if existing loopback test would detect the problem.

dgastonochoa, 2023-12-08:
@GeorgeCGV Could you try to move `spi_stm32_cs_control(dev, false);` to below `while (ll_func_spi_is_busy(spi)) {`?

That is, convert this:

```
static void spi_stm32_complete(const struct device *dev, int status)
{
	const struct spi_stm32_config *cfg = dev->config;
	SPI_TypeDef *spi = cfg->spi;
#ifdef CONFIG_SPI_STM32_INTERRUPT
	struct spi_stm32_data *data = dev->data;

	ll_func_disable_int_tx_empty(spi);
	ll_func_disable_int_rx_not_empty(spi);
	ll_func_disable_int_errors(spi);
#endif

	spi_stm32_cs_control(dev, false);

#if DT_HAS_COMPAT_STATUS_OKAY(st_stm32_spi_fifo)
	/* Flush RX buffer */
	while (ll_func_rx_is_not_empty(spi)) {
		(void) LL_SPI_ReceiveData8(spi);
	}
#endif

	if (LL_SPI_GetMode(spi) == LL_SPI_MODE_MASTER) {
		while (ll_func_spi_is_busy(spi)) {
			/* NOP */
		}
	}
	/* BSY flag is cleared when MODF flag is raised */
	if (LL_SPI_IsActiveFlag_MODF(spi)) {
		LL_SPI_ClearFlag_MODF(spi);
	}

	ll_func_disable_spi(spi);

#ifdef CONFIG_SPI_STM32_INTERRUPT
	spi_context_complete(&data->ctx, dev, status);
#endif
}
```
Into this:
```
static void spi_stm32_complete(const struct device *dev, int status)
{
	const struct spi_stm32_config *cfg = dev->config;
	SPI_TypeDef *spi = cfg->spi;
#ifdef CONFIG_SPI_STM32_INTERRUPT
	struct spi_stm32_data *data = dev->data;

	ll_func_disable_int_tx_empty(spi);
	ll_func_disable_int_rx_not_empty(spi);
	ll_func_disable_int_errors(spi);
#endif

	// -----------------------

#if DT_HAS_COMPAT_STATUS_OKAY(st_stm32_spi_fifo)
	/* Flush RX buffer */
	while (ll_func_rx_is_not_empty(spi)) {
		(void) LL_SPI_ReceiveData8(spi);
	}
#endif

	if (LL_SPI_GetMode(spi) == LL_SPI_MODE_MASTER) {
		while (ll_func_spi_is_busy(spi)) {
			/* NOP */
		}
               **spi_stm32_cs_control(dev, false);**
	}
	/* BSY flag is cleared when MODF flag is raised */
	if (LL_SPI_IsActiveFlag_MODF(spi)) {
		LL_SPI_ClearFlag_MODF(spi);
	}

	ll_func_disable_spi(spi);

#ifdef CONFIG_SPI_STM32_INTERRUPT
	spi_context_complete(&data->ctx, dev, status);
#endif
}
```

GeorgeCGV, 2023-12-08:
@dgastonochoa that for sure corrects the CS signal.

Unfortunately, it is still possible to achieve mentioned SPI condition. Especially when slave(s) are configured to run on frequencies < 8-9MHz. With frequencies higher than 14MHz it takes several restarts to stall.

Disabling fifo by returning `false` from `spi_stm32_can_use_fifo` works fine on mentioned frequencies.

dgastonochoa, 2023-12-08:
> Especially when slave(s) are configured to run on frequencies < 8-9MHz. With frequencies higher than 14MHz it takes several restarts to stall.

Could you provide the frequencies at which master and slaves are configured? I mean both the peripheral frequency and the SCK frequency.

GeorgeCGV, 2023-12-08:
@dgastonochoa overall, all slaves are unstable. It is easy to get SPI to stall at lower frequencies. The situation is a bit better when closer to max allowed (_based on datasheets_).

* PLL3P -> SPI 1,2,3 - 160Mhz
   - D1 (-, 32]  (_operating interval in MHz_) - fails at 5 (_div32_), periodically fails at 20 (_div8_)
   - D2 (-, 50] - periodically fails at 40 (_div4_)
   - D3 (-, 16] - periodically fails at 10 (_div16_)
* PCLK2 - SPI 4,5 - 120MHz
   - D1 (-, 25] - fails at 7 (_div16_), periodically fails at 15 (_div8_)
   - D2 (-, 43] - periodically fails at 30 (_div4_)

GeorgeCGV, 2023-12-11:
> @dgastonochoa that for sure corrects the CS signal.

Forgot to attach how it looks now. Two captures below, one that is stuck and the other that worked.

Works
[image]

Stuck after restart
[image]

Captures look pretty much identical.

SPI context:
```
{
    <...>
    lock = {
        wait_q = {waitq = {{head = 0x24002954, next = 0x24002954}, {tail = 0x24002954, prev = 0x24002954}}},
        count = 0x0,
        limit = 0x1,
        poll_events = {{head = 0x24002964, next = 0x24002964}, {tail = 0x24002964, prev = 0x24002964}},
        _obj_track_next = 0x0
    },
    sync = {
        wait_q = {waitq = {{head = 0x24002970, next = 0x24002970}, {tail = 0x24002970, prev = 0x24002970}}},
        count = 0x1,
        limit = 0x1,
        poll_events = {{head = 0x24002980, next = 0x24002980}, {tail = 0x24002980, prev = 0x24002980}},
        _obj_track_next = 0x0
    },
    sync_status = 0x0,
    current_tx = 0x24022474,
    tx_count = 0x0,
    current_rx = 0x2402246c,
    rx_count = 0x0,
    tx_buf = 0x0,
    tx_len = 0x0,
    rx_buf = 0x0,
    rx_len = 0x0
}
```

SPI peripheral state:
```
CR1 = 0x1000,
CR2 = 0x0,
CFG1 = 0x30070007,
CFG2 = 0x4400000,
IER = 0x0,
SR = 0x2,
IFCR = 0x0,
RESERVED0 = 0x0,
TXDR = 0x0,
RESERVED1 = {0x0, 0x0, 0x0},
RXDR = 0x0,
RESERVED2 = {0x0, 0x0, 0x0},
CRCPOLY = 0x107,
TXCRC = 0x0,
RXCRC = 0x0,
UDRDR = 0x0,
I2SCFGR = 0x0
```

Redacted stack backtrace:
```
#0  LL_SPI_IsActiveFlag_TXC (SPIx=0x40013400) at <...>/modules/hal/stm32/stm32cube/stm32h7xx/drivers/include/stm32h7xx_ll_spi.h:1832
#1  ll_func_spi_is_busy (spi=spi@entry=0x40013400) at <...>/zephyr/drivers/spi/spi_ll_stm32.h:181
#2  <...> in spi_stm32_complete (dev=dev@entry=<...> <__device_dts_ord_167>, status=status@entry=0) at <...>/zephyr/drivers/spi/spi_ll_stm32.c:482
#3  <...> in spi_stm32_isr (dev=<...> <__device_dts_ord_167>) at <...>/zephyr/drivers/spi/spi_ll_stm32.c:519
#4  <...> in _isr_wrapper () at <...>/zephyr/arch/arm/core/cortex_m\isr_wrapper.S:128
#5  <signal handler called>
#6  <...> in spi_context_total_rx_len (ctx=<optimized out>) at <...>/zephyr/drivers/spi/spi_context.h:449
#7  spi_context_wait_for_completion (ctx=ctx@entry=0x24002944 <spi_stm32_dev_data_1>) at <...>/zephyr/drivers/spi/spi_context.h:157
#8  <...> in transceive (dev=<...> <__device_dts_ord_167>, config=<...> <some_dev_config_0+8>, tx_bufs=0x2402245c <z_main_stack+12188>, rx_bufs=0x24022454 <z_main_stack+12180>, asynchronous=asynchronous@entry=false, cb=cb@entry=0x0, userdata=userdata@entry=0x0) at <...>/zephyr/drivers/spi/spi_ll_stm32.c:759
#9  <...> in spi_stm32_transceive (dev=<optimized out>, config=<optimized out>, tx_bufs=<optimized out>, rx_bufs=<optimized out>) at <...>/zephyr/drivers/spi/spi_ll_stm32.c:1011
#10 <...> in z_impl_spi_transceive (rx_bufs=0x2402240c <z_main_stack+12108>, tx_bufs=0x24022414 <z_main_stack+12116>, config=<...> <some_dev_config_0+8>, dev=<optimized out>) at <...>/zephyr/include/zephyr/drivers/spi.h:761
#11 spi_transceive (rx_bufs=0x2402240c <z_main_stack+12108>, tx_bufs=0x24022414 <z_main_stack+12116>, config=<...> <some_dev_config_0+8>, dev=<optimized out>) at <...>/ext-uif-app/build/ext_uif_h730vb/app/zephyr/include/generated/syscalls/spi.h:38
#12 spi_transceive_dt (spec=spec@entry=<...> <some_dev_config_0+4>, tx_bufs=tx_bufs@entry=0x2402245c <z_main_stack+12188>, rx_bufs=rx_bufs@entry=0x24022454 <z_main_stack+12180>) at <...>/zephyr/include/zephyr/drivers/spi.h:786
#13 <...>
```

dgastonochoa, 2023-12-11:
@GeorgeCGV I think the problem might be that the master uses the FIFO but the slave doesn't; I will do some tests today.

GeorgeCGV, 2023-12-11:
@dgastonochoa what is noticeable is that the amount of bytes in and out differs from what is given to `spi_transceive`. Trying to do `spi_transceive` with 9 bytes, but observing more. It looks like it accumulates with every next `transceive`. Eventually leading to a stall.

dgastonochoa, 2023-12-11:
@GeorgeCGV I have tried to reproduce this error with 2 separated nucleo boards, but I couldn't. I have used a way lower frequency (190 kHz) than the minimum you specified (~8MHz, right?).

I have tried sending from a master SPI peripheral running @ 48MHz, with SCK @ 190 kHz, to a slave SPI peripheral running @ 96MHz (in a different board). I have tried both with and without interrupts, sending 128 bytes of data in a row, waiting for 1 second between sends. 

When you say:
 - PLL3P -> SPI 1,2,3 - 160Mhz [...]
 - PCLK2 - SPI 4,5 - 120MHz [...]

160 and 120 MHz are the SPI peripheral clocks I imagine, correct? Which ones are masters and which ones slaves? Also, do SPI masters and slaves belong to the same MCU, or are them in separated ones? 

Note: during my tests, I forgot to enable `CONFIG_SPI_SLAVE` at the beginning, I mention it here in case the same happened to you by any chance.

GeorgeCGV, 2023-12-11:
@dgastonochoa  

> 160 and 120 MHz are the SPI peripheral clocks I imagine, correct?

yes

> Which ones are masters and which ones slaves? Also, do SPI masters and slaves belong to the same MCU, or are them in separated ones?

in all cases mcu acts as master with different kind of slaves connected

Will need to find a way to run existing tests on the board.

dgastonochoa, 2023-12-11:
> in all cases mcu acts as master on all spi buses with different kind of slaves connected

Different kind of slaves connected? Then, the slaves are not H7's ones? They are completely different slaves that belong to other devices?

GeorgeCGV, 2023-12-11:
@dgastonochoa the slaves are other ICs like eeprom, nor-flash, and others.

dgastonochoa, 2023-12-12:
> the slaves are other ICs like eeprom, nor-flash, and others.

@GeorgeCGV I see, I thought they were H7 SPI slaves. In that case, there is no way I can reproduce your problem.

However, what I can see from your captures and stacktrace etc. is that the driver seems to be stuck reading TXC **even though the CS line has been de-asserted**. This makes me think that the `complete` function might have been called twice (assuming those captures have been taken after applying #66381).

GeorgeCGV, 2023-12-12:
@dgastonochoa 

> ...given that you have enabled CRC that would be expected...

The CRC bit (_CFG1 bit 22_) is not set in the SPI peripheral. Software part is used. I don't expect single CRC byte to increase the whole transaction by a factor of 2.

> assuming those captures have been taken after applying #66381

Yes.

Anyway, modified existing loopback test to perform incremental length `spi_transceive` in a loop.

```
#define SPI_OP(frame_size) SPI_OP_MODE_MASTER | SPI_TRANSFER_MSB | SPI_WORD_SET(frame_size) | SPI_LINES_SINGLE
...
static int spi_complete_multiple(struct spi_dt_spec *spec)
{
	int ret;

	uint8_t tx_data[18];
	uint8_t rx_data[18];

	for (unsigned i = 0; i < sizeof(tx_data); ++i) {
		tx_data[i] = i;
	}

	while(1) {
		for(unsigned len = 1; len <= sizeof(tx_data); ++len) {
			(void)memset(rx_data, 0, sizeof(rx_data));

			struct spi_buf tx_buf = { .buf = tx_data, .len = len };
			struct spi_buf rx_buf = { .buf = rx_data, .len = len };
			const struct spi_buf_set tx = { .buffers = &tx_buf, .count = 1U };
			const struct spi_buf_set rx = { .buffers = &rx_buf, .count = 1U };

			ret = spi_transceive_dt(spec, &tx, &rx);
			if (ret) {
				LOG_ERR("Code %d, len %u", ret, len);
				continue;
			}

			if (memcmp(tx_data, rx_data, len)) {
				LOG_ERR("Buffer contents are different, rx&tx len %u", len);
			}

			k_sleep(K_MSEC(1000));
		}

		k_sleep(K_MSEC(5000));
	}
}
```

#### Captures

Please ignore incorrect data values, probably bad connections. What matters is the clock generation. The clock is active for way longer than expected.

##### With fifo
[image]

###### With fifo 8 byte transceive
[image]

##### Without fifo
[image]

###### Without fifo 8 byte transceive
[image]

GeorgeCGV, 2023-12-12:
If CS control is not moved as it is done in #66381, and de-asserted before busy check then we can see that it is de-asserted as soon as all data is transmitted/received.

[image]

However, the clock continues to operate.

emil-lindq, 2023-12-13:
@MaureenHelm Hi!

In response to Maureen's priority: low I'd like to add that for me the SPI is unusable since the FIFO merge, it gets stuck where the Issue describes
> The SPI driver gets stuck in spi_stm32_complete after data reception due to ll_func_spi_is_busy being true all the time.

not sometimes but every single boot. It freezes. I have to set `spi_stm32_can_use_fifo()` to false to get it to work properly. In addition, I get regularly SPI overrun error on TX-only mode.

My point is, the priority doesn't feel low, with all due respect and all that. Thanks!

Using STM32H743BIT6

erwango, 2023-12-13:
@emil-lindq Thanks for reporting this.

Would you mind providing more inputs on your use case ? Ideally a test code that reproduces the issue would greatly help?
Current issue is that @dgastonochoa is not able to reproduce the error, which is blocking the investigation.

erwango, 2023-12-13:
Some ideas to progress on this:

@emil-lindq Would you mind having a try to revert https://github.com/zephyrproject-rtos/zephyr/pull/63173 and apply https://github.com/zephyrproject-rtos/zephyr/pull/65918 instead ?

@dgastonochoa Would you be able to introduce a `fifo-enable` device tree prop which would allow to put FIFO as a opt-in feature ? (Similarly to what was done here https://github.com/zephyrproject-rtos/zephyr/pull/65875)

GeorgeCGV, 2023-12-13:
@erwango ordered nucleo h723zg. Hopefully, the issue is reproducible there (_closes available mcu to the h730 within publicly available board_). 

Running the "test" code above and observing the signals shows that something is definitely off. Unfortunately, from the test perspective, both buffers content is going to be equal; and equality check will pass.

dgastonochoa, 2023-12-13:
@GeorgeCGV  I have been able to reproduce the error finally. As you say, SPI loopback tests pass, but the error is visible on the logic analyzer.

I have been investigating what causes it, but I haven't found a reasonable explanation. The H7's datasheet mentions several times that the SPI master can work in "continuous mode", which seems to mean that the clock signal is generated  constantly, not as frames are copied to the TxFIFO (so SCK is still generated since the last Tx frame is copied to the TxFIFO and until SPI gets disabled, which I think explains the problem). However, if this were true, the non-FIFO mode should have the same problem, but it doesn't, which seems strange.

Anyway, quoting the H7's datasheet:

[image]

This problem will require more time of study from my side so, for now, I would suggest to revert the change @erwango and  @MaureenHelm, at least the last part; the PR that enabled the use of FIFO had 3 commits: the first 2 are more "code simplifying", and the last one is the one that actually enables the FIFO, so the first 2 should be valid. I can create a PR to revert the last commit, or all of them. Which options is better?

dgastonochoa, 2023-12-13:
@erwango @MaureenHelm @GeorgeCGV @emil-lindq I think it might have found a solution. I have applied a patch and re-run the SPI loopback tests, including the new test provided by @GeorgeCGV, which runs in second place, after `spi_complete_multiple`. Let me show the code:

@GeorgeCGV 's test adapted to the spi_loopback tests:
```
static int spi_complete_multiple2(struct spi_dt_spec *spec)
{
	int ret;

	uint8_t tx_data[18];
	uint8_t rx_data[18];

	for (unsigned i = 0; i < sizeof(tx_data); ++i) {
		tx_data[i] = i;
	}

	for(unsigned len = 1; len <= sizeof(tx_data); ++len) {
		(void)memset(rx_data, 0, sizeof(rx_data));

		struct spi_buf tx_buf = { .buf = tx_data, .len = len };
		struct spi_buf rx_buf = { .buf = rx_data, .len = len };
		const struct spi_buf_set tx = { .buffers = &tx_buf, .count = 1U };
		const struct spi_buf_set rx = { .buffers = &rx_buf, .count = 1U };

		ret = spi_transceive_dt(spec, &tx, &rx);
		if (ret) {
			LOG_ERR("Code %d, len %u", ret, len);
			continue;
		}

		if (memcmp(tx_data, rx_data, len)) {
			LOG_ERR("Buffer contents are different, rx&tx len %u", len);
		}

		for(unsigned len = 1; len <= sizeof(tx_data); ++len) {

		}
	}

	return 0;
}
```
Order of execution:
```
if (spi_complete_multiple(&spi_slow) || spi_complete_multiple2(&spi_slow) ||
    spi_complete_loop(&spi_slow) ||
    spi_null_tx_buf(&spi_slow) ||
    spi_rx_half_start(&spi_slow) ||
    spi_rx_half_end(&spi_slow) ||
    spi_rx_every_4(&spi_slow) ||
    spi_rx_bigger_than_tx(&spi_slow)
#if (CONFIG_SPI_ASYNC)
    || spi_async_call(&spi_slow)
#endif
    ) {
    goto end;
}
```
The logic analyzer captures are too big to be shown fully in a picture here, still there is a summary. They are identical for both irq-enabled and not-irq-enabled tests:

[image]

The above is a capture of part of the @GeorgeCGV's test. From what I'm seeing, it seems the problem of the extra clocks is solved, isn't it?

Here are the 2 full captures which can be opened with Logic II. It would be great if someone could double check and confirm whether or not that is the expected result:

[captures.zip](https://github.com/zephyrproject-rtos/zephyr/files/13660006/captures.zip)

GeorgeCGV, 2023-12-13:
@dgastonochoa I am not against, and available for testing. Maybe before a PR it is ok to provide a patch here (_git diff <commitid1> <commitid2> > <patch_filename>.patch_)? Just recently updated to the upstream.

dgastonochoa, 2023-12-13:
> Would you be able to introduce a fifo-enable device tree prop which would allow to put FIFO as a opt-in feature ? (Similarly to what was done here https://github.com/zephyrproject-rtos/zephyr/pull/65875)

@erwango Yes, I will as soon as I can.

avisconti, 2023-12-13:
> > Would you be able to introduce a fifo-enable device tree prop which would allow to put FIFO as a opt-in feature ? (Similarly to what was done here #65875)
> 
> @erwango Yes, I will as soon as I can.

Good idea.

GeorgeCGV, 2023-12-13:
A bit off-topic. Is there any reason to flush rx buffers before disabling SPI in case of H7, U5?

Currently, RX buffer is flashed twice for H7. Once on transceive complete, then after the peripheral is disabled.

Based on RM0468 Rev 3 2215/3357:

> Both RxFIFO and TxFIFO content is kept flushed when SPI is disabled (SPE=0). 

If I understand it correctly, that means

https://github.com/zephyrproject-rtos/zephyr/blob/9f3d9f75942f595964447ec685701e808eef1ab3/drivers/spi/spi_ll_stm32.c#L475-L480

and 

https://github.com/zephyrproject-rtos/zephyr/blob/9f3d9f75942f595964447ec685701e808eef1ab3/drivers/spi/spi_ll_stm32.h#L228-L231

are safe to remove (_`st_stm32_spi_fifo` is on at least H7_).

GeorgeCGV, 2023-12-13:
@dgastonochoa does your fix employs setting transfer size (_TSIZE_)?

With some corrections to use TSIZE it looks better:
[image]

dgastonochoa, 2023-12-13:
> @dgastonochoa does your fix employs setting transfer size (_TSIZE_)?
> 
> With some corrections to use TSIZE it looks better: ![image](https://private-user-images.githubusercontent.com/5956504/290232038-97b0de29-2a67-439b-bbd1-8c24c5b08e81.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI0ODU3ODYsIm5iZiI6MTcwMjQ4NTQ4NiwicGF0aCI6Ii81OTU2NTA0LzI5MDIzMjAzOC05N2IwZGUyOS0yYTY3LTQzOWItYmJkMS04YzI0YzViMDhlODEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIxMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMTNUMTYzODA2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmZkOTE0YjU3YjFjZDQwNGFiNjYyNmFhNjYwMTE5YTgyMjc5OTk2M2FiOTk5MDRiOWVjZTNiYzU4M2I5YjY3ZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.QUaTXq2CqxTjNIpXLyo9Tx2ynB2HdW3CBSaibuOCGSI)
> 
> But didn't test with multiple buffers.

Yes, TSIZE and EOT.

erwango, 2023-12-18:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/66595.
@GeorgeCGV feel free to close or we can keep it open if useful for discussion.

GeorgeCGV, 2023-12-19:
@erwango closing. Discussion moved to in #66503.

closed 2023-12-19 by GeorgeCGV