zephyrproject-rtos/zephyr #66360: [CI] `mikroe_mini_m4_for_stm32` is failing `tests/benchmarks/footprints/benchmark.kernel.footprints.pm`

ycsin, 2023-12-11:
**Describe the bug**
The `mikroe_mini_m4_for_stm32` board is failing the `zephyr/tests/benchmarks/footprints/benchmark.kernel.footprints.pm` test.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using
   - `mikroe_mini_m4_for_stm32`
 - What have you tried to diagnose or workaround this issue?
   - `COUNTER_RTC_STM32_SUBSECONDS` is getting y-selected by `PM`, despite `mikroe_mini_m4_for_stm32` not having its RTC enabled.

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout latest main
2. `west build -b mikroe_mini_m4_for_stm32 -p auto -T zephyr/tests/benchmarks/footprints/benchmark.kernel.footprints.pm`
3. See error

**Expected behavior**
Test should pass

**Impact**
Blocking #63973

**Logs and console output**
```
warning: COUNTER_RTC_STM32_SUBSECONDS (defined at drivers/counter/Kconfig.stm32_rtc:47) has direct dependencies !SOC_SERIES_STM32F1X && COUNTER_RTC_STM32 && COUNTER with value n, but is currently being y-selected by the following symbols:
 - PM (defined at soc/arm/silabs_exx32/efr32bg22/Kconfig.defconfig.series:18, soc/arm/silabs_exx32/efr32bg27/Kconfig.defconfig.series:18, soc/arm/silabs_exx32/efr32mg24/Kconfig.defconfig.series:19, soc/arm/st_stm32/stm32f4/Kconfig.defconfig.series:20, subsys/pm/Kconfig:13), with value y, direct dependencies (SOC_SERIES_EFR32BG22 && SOC_FAMILY_EXX32) || (SOC_SERIES_EFR32BG27 && SOC_FAMILY_EXX32) || (SOC_SERIES_EFR32MG24 && SOC_FAMILY_EXX32) || SOC_SERIES_STM32F4X || (SYS_CLOCK_EXISTS && HAS_PM) (value: y), and select condition SOC_SERIES_STM32F4X (value: y)
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 0bf4efc78080ceefaad7becf0d01c8d7fb2a8119

ycsin, 2023-12-11:
Reopened the issue as `96b_aerocore2` also fails the test, currently running the `tests/benchmarks/footprints/benchmark.kernel.footprints.pm` test on vendor `st` with Twister to hopefully find out all ST boards that fail the test

FRASTM, 2023-12-11:
Same fix to be applied to `96b_aerocore2`  I guess

```
diff --git a/boards/arm/96b_aerocore2/96b_aerocore2.dts b/boards/arm/96b_aerocore2/96b_aerocore2.dts
index 4a4e8672f7..dd5f8d8a89 100644
--- a/boards/arm/96b_aerocore2/96b_aerocore2.dts
+++ b/boards/arm/96b_aerocore2/96b_aerocore2.dts
@@ -42,6 +42,10 @@
 
 };
 
+&clk_lsi {
+	status = "okay";
+};
+
 &clk_hse {
 	clock-frequency = <DT_FREQ_M(24)>;
 	status = "okay";
@@ -180,6 +184,12 @@ zephyr_udc0: &usbotg_fs {
 	status = "okay";
 };
 
+&rtc {
+	clocks = <&rcc STM32_CLOCK_BUS_APB1 0x10000000>,
+		 <&rcc STM32_SRC_LSI RTC_SEL(2)>;
+	status = "okay";
+};
+
 &rng {
 	status = "okay";
 };

```

erwango, 2023-12-12:
Ideally we need a better solution than the need to enable RTC on all boards, otherwise this will pop up every now and then.

PR #66359 (2023-12-11): boards: arm: Add RTC clock source for mikroe_mini_m4_for_stm32
The `mikroe_mini_m4_for_stm32` board doesn't have its RTC node enabled, and is failing the following test:

`tests/benchmarks/footprints/benchmark.kernel.footprints.pm`

This board seems to have been missed out from 44b8370, let's enable the rtc & clk_lsi here.

Fixes #66360
PR #66370 (2023-12-18): [Backport v3.5-branch] boards: arm: Add RTC clock source for mikroe_mini_m4_for_stm32
Backport 54573e7c4c77b5b9311108d19c8d9543e3f0b936 from #66359.

Fixes #66360
closed 2023-12-12 by fabiobaltieri