zephyrproject-rtos/zephyr #66361: UART port open/close event in uart_set_callback

pnchatterji, 2023-12-11:
Hello,
I have a large number of bare-metal demo apps that I need to port to Zephyr. All of these apps wait for the host to connect on UART, and then display prompts etc. Unfortunately, Zephyr UART API does not seem to provide any mechanism for this. The only way seems to be to implement a "Wait for DTR" kind of scenario, but that would deviate from the behavior of the bare metal apps, and would add a needless complication for the user. The bare metal apps (based on the Nordic platform) achieve this quite simply using the native Nordic CDC ACM API:
static void cdc_acm_user_ev_handler(app_usbd_class_inst_t const * p_inst, app_usbd_cdc_acm_user_event_t event)
{
....
    switch (event)
    {
        case APP_USBD_CDC_ACM_USER_EVT_PORT_OPEN:
            connected =1;
             ...
           break;
        case APP_USBD_CDC_ACM_USER_EVT_PORT_CLOSE:
            connected =0;
             ...
            break;

I am sure other vendor SDKs would also have similar support.

So it should be fairly straightforward to incorporate this into Zephyr.
I propose that in uart_callback_set(), the enum uart_event_type should support two additional types of events: PORT_OPEN and PORT_CLOSE. 
This would simplify the development of the kind of  interactive, console driven apps that I am talking about.
Thanks
regards
PNC

jfischer-no, 2023-12-11:
> I am sure other vendor SDKs would also have similar support.

Perhaps, but Zephyr RTOS is not "other vendor SDK".

> So it should be fairly straightforward to incorporate this into Zephyr. I propose that in uart_callback_set(), the enum uart_event_type should support two additional types of events: PORT_OPEN and [...]

Not straightforward. uart_callback_set() is part of optional UART ASYNC API, i.e. it will not work if the driver does not support this API. The same applies to uart_irq_callback_user_data_set().

pnchatterji, 2023-12-12:
>>I am sure other vendor SDKs would also have similar support.
>>>Perhaps, but Zephyr RTOS is not "other vendor SDK".
I'm pretty sure you know what I meant, I guess you are just being sarcastic for the sake of it :-( . Since Zephyr drivers directly or indirectly (via HAL) invoke native SDK calls, all I was saying is that this kind of support would be available across most hardware platforms.
>>uart_callback_set() is part of optional UART ASYNC API
Well, where the underlying hardware/vendor-specific driver does not support this feature, the user would be out of luck. But that is fairly common situation in a vendor-neutral platform like Zephyr, no? I know any number of features in different parts of the Zephyr API that are not available on one or the other platform. 
Of course, if you, as the expert on this API, feel that it will not be straightforward to implement, I respect that. I do not have your in-depth knowledge of the implementation. However, I feel this is a sufficiently useful feature that it might be worth the effort. I believe most CDC ACM based applications would benefit from knowing if there is an active connection or not. 
Another way of providing this feature could be to provide a is_port_open() API function, in which case platform-specific implementations could decide the best way to provide this feature, or simply return true if it cannot be supported. This could be handled in the API documentation "This function is not guaranteed to return an accurate value on all platforms. UART ASYNC API needs to be activated to get a meaningful value, where supported by the platform. In other cases, by default, this function returns 'true'. "

tmon-nordic, 2023-12-12:
> I believe most CDC ACM based applications would benefit from knowing if there is an active connection or not. Another way of providing this feature could be to provide a is_port_open() API function

The problem is that the CDC ACM protocol is not really proving such information. The device cannot really know when the host actually *opens* port, there can be some heuristics but they have no real way to know when the port is closed and then reopened.

However, the latest main has the possibility to buffer data even when not connected to host (see https://github.com/zephyrproject-rtos/zephyr/pull/64720 for more information). Would simply enabling hardware flow control in devicetree and relying on the buffering be enough for your use case? I believe this is best we can do for basic text based protocols supposed to be operated by humans.

pnchatterji, 2023-12-12:
Hi tmon-nordic
Yes, I guess it will work... if tx buffer gets full the app will block until a connection is done. Since I am dealing with demo apps at present, this should be tolerable. However, it probably isn't the best solution for real-life apps, which I may have to handle in future. Btw, from your nickname, I presume you are a Nordic engineer. How does the Nordic SDK manage to generate the APP_USBD_CDC_ACM_USER_EVT_PORT_OPEN/CLOSE events if it isn't supported by the CDC ACM protocol?

tmon-nordic, 2023-12-12:
> Btw, from your nickname, I presume you are a Nordic engineer. How does the Nordic SDK manage to generate the APP_USBD_CDC_ACM_USER_EVT_PORT_OPEN/CLOSE events if it isn't supported by the CDC ACM protocol?

SDK was created before I joined Nordic. In nRF5 SDK the CDC ACM line state DTR bit is used to generate `APP_USBD_CDC_ACM_USER_EVT_PORT_OPEN` (when DTR is set) and `APP_USBD_CDC_ACM_USER_EVT_PORT_OPEN` (when DTR is cleared).

pnchatterji, 2023-12-12:
Ah, OK, thanks for the clarification! To be honest, I haven't attempted to understand each line of the customer's bare-metal codebase. I'm reimplementing it in Zephyr at a functional level. So what you are saying is that it is just a way to implement a wait-for-DTR?

jfischer-no, 2024-05-21:
There are no plans to implement this suggestion. For the new USB device support, there are explicit messages for the CDC ACM UART line state and line coding events.
https://github.com/zephyrproject-rtos/zephyr/blob/b3ffe0aede6db181e118eb95db84ad7537bed0b2/samples/subsys/usb/cdc_acm/src/main.c#L73-L84