zephyrproject-rtos/zephyr #66380: tests/net/hostname test issue regression

Desvauxm-st, 2023-12-11:
Describe the bug:

in our bench tester many board test fail now with:

tests/net/hostname
for example:

 west build -p always -b   nucleo_h743zi  tests/net/hostname

test is blocked in the console nothing is displayed

it is due to this commit:
https://github.com/zephyrproject-rtos/zephyr/commit/78af1c1e3878c410067bd49846771300fa86b840

entropy: stm32: suspend when pool is full
https://github.com/zephyrproject-rtos/zephyr/commit/78af1c1e3878c410067bd49846771300fa86b840
author: Dawid Niedzwiecki [dawidn@google.com](mailto:dawidn@google.com) 2023-12-01 11:27:37
Committer: Maureen Helm [maureen.helm@analog.com](mailto:maureen.helm@analog.com) 2023-12-05 23:27:53

thanks to have a look

niedzwiecki-dawid, 2023-12-18:
Can we close this one?

erwango, 2023-12-18:
> Can we close this one?

Yes, closing. Thanks for the support

closed 2023-12-18 by erwango