zephyrproject-rtos/zephyr #66452: `board_runner_args` will not append command arguments for OpenOCD west flash runner

IvanVnucec, 2023-12-12:
I'm trying to append chip erase command arguments to the OpenOCD flash runner.

In `app/CmakeLists.txt` I've added `app_set_runner_args` macro:
```cmake
macro(app_set_runner_args)
    board_runner_args(openocd "-c 'flash erase_sector 0 1 last'")
endmacro()
```
When I try to run `west --verbose flash -r openocd` I don't see my board runner args in the west flash log:
```console
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/ivan/Documents/artronic/temp-mag-sens/artronic-ble-temp-mag-sensor/build/zephyr/merged.hex
runners.openocd: /home/ivan/zephyr-sdk-0.16.1/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/ivan/zephyr-sdk-0.16.1/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -c 'set WORKAREASIZE 0x4000' -c 'source [find interface/jlink.cfg]' -c 'transport select swd' -c 'source [find target/nrf52.cfg]' '-c init' '-c targets' -c 'reset init' -c 'flash write_image erase /home/ivan/Documents/artronic/temp-mag-sens/artronic-ble-temp-mag-sensor/build/zephyr/merged.hex' -c 'reset init' -c 'verify_image /home/ivan/Documents/artronic/temp-mag-sens/artronic-ble-temp-mag-sensor/build/zephyr/merged.hex' -c 'reset run' -c shutdown
Open On-Chip Debugger 0.11.0+dev-00726-ge2b6f5655 (2023-05-11-16:58)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
0x4000
swd
Info : J-Link ARM V8 compiled Jan 31 2018 18:34:52
Info : Hardware version: 8.00
Info : VTarget = 3.345 V
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x2ba01477
Info : [nrf52.cpu] Cortex-M4 r0p1 processor detected
Info : [nrf52.cpu] target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for nrf52.cpu on 3333
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* nrf52.cpu          cortex_m   little nrf52.cpu          running

target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x000020d4 msp: 0x200057c0
Info : nRF52832-QFAA(build code: E0) 512kB Flash, 64kB RAM
Info : Flash write discontinued at 0x00007d52, next section at 0x0000a000
Info : Padding image section 0 at 0x00007d52 with 2 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x00007d54 .. 0x00007fff
Warn : Adding extra erase range, 0x0003d71c .. 0x0003dfff
auto erase enabled
wrote 242800 bytes from file /home/ivan/Documents/artronic/temp-mag-sens/artronic-ble-temp-mag-sensor/build/zephyr/merged.hex in 6.692354s (35.430 KiB/s)

target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x000020d4 msp: 0x200057c0
verified 242798 bytes in 0.658096s (360.293 KiB/s)

shutdown command invoked
```
Also, looking at the `build/mcuboot/zephyr/runners.yaml`, I don't see my flash erase arguments:
```yaml
# Runner specific arguments
args:
  openocd:
    - --cmd-load
    - flash write_image erase
    - --cmd-verify
    - verify_image
    - --cmd-pre-init
    - set WORKAREASIZE 0x4000
    - --cmd-pre-init
    - source [find interface/jlink.cfg]
    - --cmd-pre-init
    - transport select swd
    - --cmd-pre-init
    - source [find target/nrf52.cfg]
```

tejlmand, 2024-01-03:
> In app/CmakeLists.txt I've added app_set_runner_args macro:

when are you defining the `macro` ?

It should be defined before `find_package(Zephyr)`
Like this:
```
macro(app_set_runner_args)
    board_runner_args(openocd "-c 'flash erase_sector 0 1 last'")
endmacro()

find_package(Zephyr ...)
```

tejlmand, 2024-01-03:
Description improved here: #67148

IvanVnucec, 2024-01-03:
> > In app/CmakeLists.txt I've added app_set_runner_args macro:
> 
> when are you defining the `macro` ?
> 
> It should be defined before `find_package(Zephyr)` Like this:
> 
> ```
> macro(app_set_runner_args)
>     board_runner_args(openocd "-c 'flash erase_sector 0 1 last'")
> endmacro()
> 
> find_package(Zephyr ...)
> ```

Indeed, it is defined before `find_pachage(Zephyr)`:
```cmake
macro(app_set_runner_args)
    board_runner_args(dfu-util "--dfuse-modifiers=force:mass-erase")
    board_runner_args(pyocd "--flash-opt=-e=chip")
    board_runner_args(nrfjprog "--erase")
    board_runner_args(openocd "-c 'flash erase_sector 0 1 last'")
endmacro()

find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(app LANGUAGES C VERSION 3.5.0)
```

tejlmand, 2024-01-03:
> Indeed, it is defined before find_package(Zephyr):

@IvanVnucec  interesting. So far I've found no suspicious code in CMake, and feature is working as expected in Zephyr tree.
Can you ping me on discord, so I can get some more trace info from you ?
(I can't seem to find your discord userid)

IvanVnucec, 2024-01-03:
> > Indeed, it is defined before find_package(Zephyr):
> 
> @IvanVnucec interesting. So far I've found no suspicious code in CMake, and feature is working as expected in Zephyr tree. Can you ping me on discord, so I can get some more trace info from you ? (I can't seem to find your discord userid)

I'm not on discord.
I will try to replicate the issue with some simple example and I will post it here once I'm finished.

tejlmand, 2024-01-03:
> I'm not on discord.

you may run the build using trace output, as:
```
# Using CMake
$ cmake -DBOARD=<board> ... --trace-expand 2> issue_66452.trace

# Using west build
$ west build -b <board> --cmake-only ... -- --trace-expand 2> issue_66452.trace
```

and then attach the trace file in a new comment in this issue.

IvanVnucec, 2024-02-20:
After a lot of trial and error I've managed to append OpenOCD arguments for a full chip erase by doing:
```cmake
macro(app_set_runner_args)
    board_runner_args(openocd "--cmd-pre-load=reset halt" "--cmd-pre-load=flash erase_sector 0 0 last")
endmacro()
```

closed 2024-02-20 by IvanVnucec