zephyrproject-rtos/zephyr #66454: Can linker information already be used at compile time?

laroche, 2023-12-12:
The current source code of zephyr in kernel/include/kernel_tls.h is using linker information
stored into variables to look at alignment information. This results in 2 integer divisions
within the function z_tls_data_size().

Even better would be if we could change this into compile time constants and have compiler
optimizations work on them.

The following patch shows the exact code location. Bute the patch has no clean solution to
use some generated header defines for the alignment size, but just uses very cludgy code
via "sizeof(int)" as a really bad workaound.

Anyone have a better idea? Maybe some Kconfig default parameters?

<pre>
--- a/kernel/include/kernel_tls.h
+++ b/kernel/include/kernel_tls.h
@@ -28,8 +28,9 @@
  */
 static inline size_t z_tls_data_size(void)
 {
-       size_t tdata_size = ROUND_UP(__tdata_size, __tdata_align);
-       size_t tbss_size = ROUND_UP(__tbss_size, __tbss_align);
+       /* XXX __tdata_align and __tbss_align should be compile-time constants and not link-time vars. */
+       size_t tdata_size = ROUND_UP(__tdata_size, sizeof(int) /* __tdata_align */);
+       size_t tbss_size = ROUND_UP(__tbss_size, sizeof(int) /* __tbss_align */);

        return tdata_size + tbss_size;
 }
@@ -51,7 +52,7 @@ static inline void z_tls_copy(char *dest)
        memcpy(dest, __tdata_start, tdata_size);

        /* Clear BSS data (tbss) */
-       dest += ROUND_UP(tdata_size, __tdata_align);
+       dest += ROUND_UP(tdata_size, sizeof(int) /* __tdata_align */);
        memset(dest, 0, tbss_size);
 }

</pre>

best regards,

Florian La Roche

henrikbrixandersen, 2023-12-19:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md (Please provide a description of the impact of this issue).

peter-mitsis, 2024-01-15:
If I am following things correctly, the core of the issue is that we are performing what is perceived to be unnecessary division operations in the ROUND_UP() macros, and division operations  are infamously slow.

Taking a step back for a moment, if we were to eliminate the division completely, what would it buy us? The routine z_tls_data_size() seems to be only invoked from arch_tls_stack_setup(), which in turn is only invoked when a thread is created. Thus, perfect optimisation here will gain us a slight performance boost when threads are created. Conceivably, we could also get a small footprint reduction if there is no other division and division is implemented with software and not a hardware instruction. This seems like small gains, but depending upon one's application, needs and aims it could still be valuable.

It is the generic nature of the ROUND_UP() macro that introduces the division. That is, if one wants to round to the nearest multiple of 11, it will do so. However, unless I missed it, there does not appear to be a fast version of ROUND_UP() and ROUND_DOWN() for powers of two--something like ALIGN_UP() and ALIGN_DOWN(). Having macros intended for use only with powers of two could be useful.

Something along the lines of the following ...

#define ALIGN_UP(value, align)  ((value + align - 1) & ~(align - 1))
#define ALIGN_DOWN(value, align) (value & (~align - 1))

This would replace the division with fast bit-twiddling. It might not be a perfect solution to the original problem, but it should definitely be an improvement.

henrikbrixandersen, 2024-01-16:
@peter-mitsis Please assign a priority for this issue.

PR #67677 (2024-02-26): kernel: Add ROUND2_UP() and ROUND2_DOWN() macros
Fixes #66454

Current Summary
==============
Re-worked this PR to just update __tdata_xx and __tbss_xx linker variables instead of trying to go the ROUND_UP/ROUND_DOWN() route (whose summary follows just for reference). Instead of rounding up __tdata_size and __tbss_size at runtime each time arch_tls_stack_setup() is called, the round up calculation has been moved into the linker script thread-local-storage.ld so that we do not need to perform the calculation at runtime at all.

Old Obsolete Summary
===================
Changed to ROUND2_UP and R [... cut]
closed 2024-02-26 by nashif