zephyrproject-rtos/zephyr #66461: doc: Unavailable libpython3.8-dev package following installation guide 'develop/getting_started/installation_linux'

IvanVeloz, 2023-12-13:
As of December 12 2023, in a fully updated Ubuntu Desktop 22.04 (jammy), the package `libpython3.8-dev` package is not available. `apt-get` suggest substituting with `libpython3-dev`.

```
ivan@jammy-build:~$ sudo apt-get install --no-install-recommends git cmake ninja-build gperf   ccache dfu-util device-tree-compiler wget   python3-dev python3-pip python3-setuptools python3-tk python3-wheel xz-utils file libpython3.8-dev   make gcc gcc-multilib g++-multilib libsdl2-dev libmagic1
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package libpython3.8-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libpython3-dev
```

**Environment**

* Page: `develop/getting_started/installation_linux`
* Version: main
* SHA-1: 915f8a10b62b24b244d2da00793e2b670f677763

kartben, 2023-12-13:
@IvanVeloz thanks for the report! It looks like we should remove the `libpython3.8-dev` entry altogether since `python3-dev` is already taking care of installing it. Agreed @nordicjm?

nordicjm, 2023-12-13:
No, ubuntu 20.04 LTS is still supported and that uses the package name `libpython3.8-dev` from what I can see, though maybe another tab for "ubuntu 22.04" package list?

nordicjm, 2023-12-13:
Or... and maybe this is not such a good idea, but a new repo under zephyr with .sh files which will install the required packages for specific releases? Could have one for arch, one for ubuntu 20.04, one for 22.04, even one for windows, no clue on mac. Then in the documentation can link people to that repo, or even add it to zephyr itself as an optional repo which is disabled by default but under a different group "setup" perhaps

kartben, 2023-12-13:
My point was that python3-dev already installs libpython3-dev, no?
https://packages.ubuntu.com/focal/python3-dev

IvanVeloz, 2023-12-13:
Hi everyone! Thank you for looking into it. Indeed, on 20.04 `libpython3-dev` pulls `libpython3.8-dev`.

```
root@test:~# apt install python3-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
  libfreetype6
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  libc-dev-bin libc6-dev libcrypt-dev libexpat1-dev libpython3-dev libpython3.8-dev linux-libc-dev manpages-dev python3.8-dev zlib1g-dev
Suggested packages:
  glibc-doc
The following NEW packages will be installed:
  libc-dev-bin libc6-dev libcrypt-dev libexpat1-dev libpython3-dev libpython3.8-dev linux-libc-dev manpages-dev python3-dev python3.8-dev zlib1g-dev
0 upgraded, 11 newly installed, 0 to remove and 0 not upgraded.
```

kartben, 2023-12-14:
Marking as low priority since apt provides a clear recommendation that should unblock most users.

nordicjm, 2023-12-15:
Change to `libpython3-dev`

kartben, 2023-12-15:
> Change to `libpython3-dev`

@nordicjm but why not just have `python3-dev`? I would think keeping the dependency list concise and to the point is a good thing to do. Some folks tend to complain about the initial setup being too complex, and a list of dependencies longer than necessary doesn't help with this initial feeling.

nordicjm, 2023-12-15:
`python3-dev` is fine with me

PR #66552 (2023-12-15): doc: remove explicit libpython3.8-dev from Ubuntu deps
Having libpython3.8-dev explicitly mentioned causes issues if "python3-dev" pulls a Python version that's not 3.8. What's more, python3-dev already pulls the correct "versionless" libpython3-dev anyway.

Fixes #66461
commit c2dae16a (2023-12-15): doc: remove explicit libpython3.8-dev from Ubuntu deps

Having libpython3.8-dev explicitly mentioned causes issues if
"python3-dev" pulls a Python version that's not 3.8. What's more,
python3-dev already pulls the correct "versionless" libpython3-dev
anyway.

Fixes #66461
closed 2023-12-15 by carlescufi