zephyrproject-rtos/zephyr #66469: net: socket: tcp: net.socket.tcp and net.socket.tcp.preempt fail

katgiadla, 2023-12-13:
**Describe the bug**
The tests `net.socket.tcp` and `net.socket.tcp.preempt` fail.

Observed for `nrf5340dk_nrf5340_cpuapp_ns`

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf5340dk_nrf5340_cpuapp_ns` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/net/socket/tcp -p nrf5340dk_nrf5340_cpuapp_ns --device-testing --device-serial /dev/ttyACM1 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
Error: Image size (0x2ffee) + trailer (0x1b0) exceeds requested size 0x30000
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.1
- Commit SHA or Version used: zephyr-v3.5.0-2927-g03b702690ae7

carlescufi, 2023-12-13:
@joerchan this is TF-M related, correct?

joerchan, 2023-12-13:
> @joerchan this is TF-M related, correct?

@carlescufi It's a size problem. test application does not fit into 192 KB. Easiest fix it to disable the bootloader to give the application more space.

carlescufi, 2023-12-13:
@joerchan probably better to disable the bootloader for now then please.
cc @nordicjm @de-nordic

PR #66493 (2023-12-15): tests: net: socket: tcp: Fix build for nrf5340dk_nrf5340_cpuapp_ns
Fix build for nrf5340dk_nrf5340_cpuapp_ns.
Test takes up almost all of the flash, leaving no room for the image to be signed. Disable the bootloader to free up flash-usage.

Fixes: #66469
commit cb13e21e (2023-12-15): tests: net: socket: tcp: Fix build for nrf5340dk_nrf5340_cpuapp_ns

Fix build for nrf5340dk_nrf5340_cpuapp_ns.
Test takes up almost all of the flash, leaving no room for the image
to be signed. Disable the bootloader to free up flash-usage.

Fixes: #66469
closed 2023-12-15 by carlescufi