zephyrproject-rtos/zephyr #66480: Bluetooth: Mesh: Directed Forwarding

LingaoM, 2023-12-13:


LingaoM, 2023-12-13:
Let's you known @PavelVPV , any new progress will be publish.

PavelVPV, 2023-12-13:
Are you planning or working on it already?

LingaoM, 2023-12-13:
Yeah, have free time do it now.

PavelVPV, 2023-12-13:
I think it would be good if we could discuss and review architecture first.

LingaoM, 2023-12-13:
I have also considered this, and a basic version will be submitted in time

alxelax, 2023-12-14:
@LingaoM, are you going to add Direct Forwarding PTS tests in the Bluetooth tester and make the feature qualifiable?

LingaoM, 2023-12-14:
It will be delayed and basic functions need to be completed first.

LingaoM, 2023-12-14:
My plan is to complete the basic functions first, then complete the Doc and API details, then complete the Demo show(maybe add into mesh_shell first), then complete the bbsim test, and finally the PTS.

LingaoM, 2023-12-15:
[image]

Happy Moment

LingaoM, 2023-12-18:
Hi @omkar3141 

I have a question for you about Mesh Protocol 1.1,  if there only two node establish a path(no directed rely node),
I'm found the path origin can't set `Backward_Path_Validated =1` , and didn't find any relevant specification explaining this.

There is only one place where this is stated:
Mesh Protocol 1.1 3.6.8.5.5 PATH_CONFIRMATION message received

> If a Forwarding Table entry that corresponds to the PATH_CONFIRMATION message exists, the Backward_Path_Validated field in the table entry shall be set to 1, and the Forwarding Table Update Identifier shall change (see Section 4.2.29.1).

As I said at the beginning, if there are only two nodes to establish the path, how can path origin set this flag?

LingaoM, 2023-12-21:
[image]

@alxelax PTS testing is in progress and will be submitted to the community after completion. Some Doc and API work may require your help. This is not my strong point. Thank you.

PavelVPV, 2024-01-25:
@LingaoM, I've updated task description with the list of what is done and what is not. Hope I didn't forgot or incorrectly (not)marked anything.

alxelax, 2024-07-05:
Hi @LingaoM, please keep this task open. We have plans to pick up your results and continue work. There is no exact date for this, but this ticket is a good place for keeping information.

xieshren, 2024-07-23:
@PavelVPV 
Has the feature been supported? If not supported yet, when can it be supported?

PavelVPV, 2024-08-28:
> @PavelVPV Has the feature been supported? If not supported yet, when can it be supported?

Hi @xieshren, no, it is not supported and not in the first priority atm. We will update this issue once we resume work on it.

schneider-engineering, 2025-05-09:
@PavelVPV / @alxelax any idea if Directed Forwarding will be supported by zepyhr in the foreseeable Future?

PavelVPV, 2025-05-09:
> [@PavelVPV](https://github.com/PavelVPV) / [@alxelax](https://github.com/alxelax) any idea if Directed Forwarding will be supported by zepyhr in the foreseeable future?

Hi @schneider-engineering , no changes since last update from us. Perhaps @LingaoM can answer this?