zephyrproject-rtos/zephyr #66481: Odd latency_measure numers after 1a7e5c6c8d

fabiobaltieri, 2023-12-13:
**Describe the bug**
Hi, the tests/benchmarks/latency_measure is reporting some unreasonable numbers on few risc-v platforms since https://github.com/zephyrproject-rtos/zephyr/pull/63636.

Verified on an esp32c3

`west build -p -b xiao_esp32c3 tests/benchmarks/latency_measure`

```
...
Give a semaphore (no waiters) from kernel thread    :  429494 cycles , 26843395 ns :
Take a semaphore (no blocking) from kernel thread   :  429490 cycles , 26843152 ns :
Take a semaphore (context switch K -> K)            :      49 cycles ,     3062 ns :
Give a semaphore (context switch K -> K)            :      56 cycles ,     3524 ns :
Lock a mutex from kernel thread                     :  429491 cycles , 26843239 ns :
Unlock a mutex from kernel thread                   :  429487 cycles , 26842982 ns :
...
```

and an fe310 board as well

`west build -p -b hifive1_revb tests/benchmarks/latency_measure`

```
...
Lock a mutex from kernel thread                     :       0 cycles ,     2310 ns : 
Unlock a mutex from kernel thread                   :  429496 cycles , 222297965 ns : 
Average time for heap malloc                        :       4 cycles ,   141601 ns : 
...
```

Reverting 1a7e5c6c8d on top of main seems to fix the numbers.

@peter-mitsis can you take a look?

**To Reproduce**
See above

**Expected behavior**
Reasonable numbers

**Impact**
@nashif reporting "something is wrong somewhere" in #release

**Environment (please complete the following information):**
Linux, Zephyr SDK, c758f1cc14

peter-mitsis, 2023-12-13:
Initial thoughts based on the numbers ...

It is curious that the wonky numbers are approximately UINT_MAX / 10000 (each test was done 10000 times) as this suggests either one of two possibilities.

1. The ordering of the threads involved got out of sync somehow.
2. Or the timestamp bias correction step resulted in a negative number.

Between the two, I am leaning towards the second idea.

peter-mitsis, 2023-12-13:
Did some more digging .... Those semaphore and mutex sub-tests that had the wonky numbers were erroneously applying the timestamp bias adjustment.

PR #66509 (2023-12-15): tests: Fix bias adjustment in latency_measure
Updates the non-blocking semaphore and mutex benchmark sub-tests in the latency_measure benchmark to remove the timestamp bias adjustment. As the timestamps are not sampled during each iteration of those sub-tests, there is no need to correct for any bias from the sampling.

Fixes #66481
closed 2023-12-15 by fabiobaltieri