zephyrproject-rtos/zephyr #66486: SPI on STM32U5 mcu get stuck after  50f64eaeba commit (#63173)

avisconti, 2023-12-13:
After #63173 was merged I started to see an issue on my sensortile_box_pro board, based on stm32u5 mcu. In particular, reverting the 50f64eaeba commit seems to fix the issue. Looking at that commit I found out that, since stm32u5 mcu is based on stm32h7 spi controller, the new code gets stuck into spi_stm32_shift_fifo() routine.

erwango, 2023-12-13:
Closing as this is the same issue as https://github.com/zephyrproject-rtos/zephyr/issues/66326.
But don't hesitate to share in this issue if you have a simple way to reproduce the issue.

avisconti, 2023-12-13:
> Closing as this is the same issue as #66326. But don't hesitate to share in this issue if you have a simple way to reproduce the issue.

@erwango I actually saw issue #66326 but I'm not sure it is the same one (it might be). In my case it gets stuck in a different routine.

dgastonochoa, 2023-12-13:
> > Closing as this is the same issue as #66326. But don't hesitate to share in this issue if you have a simple way to reproduce the issue.
> 
> @erwango I actually saw issue #66326 but I'm not sure it is the same one (it might be). In my case it gets stuck in a different routine.

I have opened a PR that (at least in my setup) fixes the problem:  #66503. 

I'm not sure if the issue is the exact same one as you mention; it would be great if you could test it on your side to be 100% sure.

avisconti, 2023-12-14:
> I have opened a PR that (at least in my setup) fixes the problem: #66503.
>
> I'm not sure if the issue is the exact same one as you mention; it would be great if you could test it on your side to be [...]

Yes, it seems to fix this issue as well. Thanks very much