zephyrproject-rtos/zephyr #66490: nrf52840dongle_nrf52840: default i2c0 pins does not exist on the board

osaether, 2023-12-13:
The default i2c0 pins assigned in the file "boards/arm/nrf52840dongle_nrf52840/nrf52840dongle_nrf52840-pinctrl.dtsi" doesn't exist on the physical board. Here is the relevant section in the file:

`	i2c0_default: i2c0_default {
		group1 {
			psels = <NRF_PSEL(TWIM_SDA, 0, 26)>,
				<NRF_PSEL(TWIM_SCL, 0, 27)>;
		};
	};

	i2c0_sleep: i2c0_sleep {
		group1 {
			psels = <NRF_PSEL(TWIM_SDA, 0, 26)>,
				<NRF_PSEL(TWIM_SCL, 0, 27)>;
			low-power-enable;
		};
	};
`
Please see attached schematic.

[PCA10059_Schematic_And_PCB.pdf](https://github.com/zephyrproject-rtos/zephyr/files/13661128/PCA10059_Schematic_And_PCB.pdf)

osaether, 2023-12-13:
I see now that there are other pins that are wrong as well so maybe a full review of the pins should be done.

gmarull, 2023-12-13:
I'm not sure if the dongle has any _preferred_ pins for e.g. I2C or other peripherals. @anangl do you know?

romannurik, 2024-02-24:
Also related, it seems like the NRF52840 dongle *does* support `arduino_i2c`, but it's not in the YAML config file and more importantly, it's missing from the `.dts` (a simple copy/paste from the NRF52840 DK seems to be all that's needed).

gmarull, 2024-02-28:
> Also related, it seems like the NRF52840 dongle _does_ support `arduino_i2c`, but it's not in the YAML config file and more importantly, it's missing from the `.dts` (a simple copy/paste from the NRF52840 DK seems to be all that's needed).

dongles does not have any arduino-like connector, so it doesn't make sense to add `arduino_i2c` in there.

romannurik, 2024-02-28:
> dongles does not have any arduino-like connector, so it doesn't make sense to add `arduino_i2c` in there.

Oh I see. So I was able to get an SSD1306 I2C display working with the nRF52840 Dongle by using arbitrary pins, but to do so, the dongle's `.dts` needed to be patched to include an `arduino_i2c` section, otherwise it wouldn't compile. Is there a more correct way to do this, or is this configuration not really supported?

gmarull, 2024-02-28:
> Oh I see. So I was able to get an SSD1306 I2C display working with the nRF52840 Dongle by using arbitrary pins, but to do so, the dongle's `.dts` needed to be patched to include an `arduino_i2c` [...]

If a sample or shield relies on boards having arduino connector then no. You can still add a board specific overlay defining your own arduino_i2c, but this can’t live upstream as I assume it requires a custom wiring.

romannurik, 2024-02-28:
> If a sample or shield relies on boards having arduino connector then no. You can still add a board specific overlay defining your own arduino_i2c, but this can’t live upstream as I assume it requires a custom wiring.

Got it... for what it's worth, I wanted to do this entirely in an overlay, but the device tree merge order (board dts -> shield dts -> board overlay -> shield overlay) didn't allow the built-in shield to work. I suppose another approach is to not use the built-in shield at all. (BTW, if this is best discussed outside this bug thread, happy to take it elsewhere)

closed 2024-05-13 by github-actions[bot]