zephyrproject-rtos/zephyr #66494: Allocate LVGL heap and VDB into external memory

dolence, 2023-12-13:
MCUs used to drive large displays commonly have little built-in SRAM. To get around this problem they generally have an much bigger external SDRAM  module attached to it. 

Would be nice to have an option to move the heap and buffer used by LVGL to an external address.

Below is an example. I have an STM32H757 with 8 MB external SDRAM being used for LTDC as framebuffer. Since my display resolution is 1280x800 and color depth is 16 bits, only 2048 KB is in use. LVGL heap and VDB are occupying precious internal RAM while I have plenty of external memory unused. If this could be allocated to the remaining portion of external RAM I could have plenty of free internal RAM and would be able to do things like use double full screen buffering.

```
Memory region         Used Size  Region Size  %age Used
           FLASH:      232940 B         1 MB     22.21%
             RAM:      441088 B       512 KB     84.13%
            QSPI:          0 GB       256 MB      0.00%
           SRAM1:          0 GB       128 KB      0.00%
           SRAM2:          0 GB       128 KB      0.00%
           SRAM3:          0 GB        32 KB      0.00%
           SRAM4:          0 GB        64 KB      0.00%
          SDRAM2:       2000 KB         8 MB     24.41%
        IDT_LIST:          0 GB         2 KB      0.00%
```

pdgendt, 2023-12-13:
There's a PR that could help, https://github.com/zephyrproject-rtos/zephyr/pull/66270, however I think this needs some more work to get right.
Maybe @faxe1008 can comment on this.

dolence, 2023-12-13:
> There's a PR that could help, #66270, however I think this needs some more work to get right. Maybe @faxe1008 can comment on this.

This only applies to heap or VDB buffer too?

faxe1008, 2023-12-13:
VDB static allocation can already be placed into a custom section. There are already kconfig flags for it.

PR #70542 (2024-06-22): modules: lvgl: support linking memory pool to custom section
This patch adds a kconfig option `LV_Z_MEMORY_POOL_CUSTOM_SECTION` which allows to place the buffer the memory pool is backed by into a section with the label ".lvgl_heap".

Fixes #66494.
closed 2024-06-22 by nashif