zephyrproject-rtos/zephyr #66500: [v2.7-branch] cannot use CONFIG_COMPILER_WARNINGS_AS_ERRORS

cfriedt, 2023-12-13:
**Describe the bug**
It would be nice to have a stable way of configuring whether warnings are promoted to errors in LTSv2. 

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? Any
 - What have you tried to diagnose or workaround this issue? Looked at #53305 and realized it was an easy backport
 - Is this a regression? No

**To Reproduce**
Steps to reproduce the behavior:
1. `git checkout origin/v2.7-branch`
2. `west build -p auto -t run samples/hello_world -- -DCONFIG_COMPILER_WARNINGS_AS_ERRORS=y`
3. See (build) error due to undefined Kconfig option
4. add an unused variable in `samples/hello_world/src/main.c`
5. apply change in linked PR
6. see expected error when building (due to warning being promoted to error).

**Expected behavior**
* Build should succeed (there are no warnings in `hello_world`
* Warnings should be promoted to errors.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
Build error because Kconfig option is not defined
```shell
 warning: attempt to assign the value 'y' to the undefined symbol COMPILER_WARNINGS_AS_ERRORS
```
Expected error when promoting warnings to errors
```shell
error: unused variable 'foo' [-Werror=unused-variable]
```

**Environment (please complete the following information):**
 - OS: Any
 - Toolchain: Zephyr SDK v0.14.2
 - Commit SHA: e677cfd61d20c043fdd554ea59b97deffa816326 

**Additional context**
Helpful for those migrating from LTSv2 to (pre) LTSv3

stephanosio, 2023-12-18:
> It would be nice to have a stable way of configuring whether warnings are promoted to errors in LTSv2.

That sounds more like an enhancement than a bug.

cfriedt, 2023-12-24:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/66501

PR #66501 (2023-12-19): [v2.7] toolchain: Move extra warning options to toolchain abstraction
Backport of #53305
Fixes #66500
**Note: Compliance failure is a false positive (those lines are not modified with this commit)**

Move extra warning option from generic twister script into compiler-dependent config files. ARCMWDT compiler doesn't support extra warning options ex. "-Wl,--fatal-warnings". To avoid build fails flag "disable_warnings_as_errors" should be passed to twister.

This allows all warning messages and make atomatic test useles.
closed 2023-12-24 by cfriedt