zephyrproject-rtos/zephyr #66512: twister code coverage report base directory not setting

carriegong, 2023-12-14:
**Describe the bug**
passing in --coverage-basedir ../XXX/src/ 
leads to 
```

Traceback (most recent call last):
  File "/opt/nordic/ncs/v2.4.0/zephyr/./scripts/twister", line 201, in <module>
    ret = main(options)
  File "/opt/nordic/ncs/v2.4.0/zephyr/scripts/pylib/twister/twisterlib/twister_main.py", line 212, in main
    run_coverage(tplan, options)
  File "/opt/nordic/ncs/v2.4.0/zephyr/scripts/pylib/twister/twisterlib/coverage.py", line 278, in run_coverage
    coverage_tool.generate(options.outdir)
  File "/opt/nordic/ncs/v2.4.0/zephyr/scripts/pylib/twister/twisterlib/coverage.py", line 101, in generate
    ret = self._generate(outdir, coveragelog)
  File "/opt/nordic/ncs/v2.4.0/zephyr/scripts/pylib/twister/twisterlib/coverage.py", line 208, in _generate
    subprocess.call(["gcovr", "-r", self.base_dir, "--gcov-executable",
  File "/usr/local/Cellar/python@3.10/3.10.13/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 345, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/local/Cellar/python@3.10/3.10.13/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 971, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/local/Cellar/python@3.10/3.10.13/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 1796, in _execute_child
    self.pid = _posixsubprocess.fork_exec(
TypeError: expected str, bytes or os.PathLike object, not NoneType
```
Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? qemu_cortex_m3
 - What have you tried to diagnose or workaround this issue? I edited coverage.py to check if self.base_dir == None and it is none though even i set it in the python script or pass it through calling twister 
**To Reproduce**
Steps to reproduce the behavior:
`./scripts/twister --coverage --coverage-basedir ../XXX/src/ -p qemu_cortex_m3  -c --coverage-tool gcovr  -T ../XXX/tests `

**Expected behavior**
code coverage report should generate 

**Environment (please complete the following information):**
 - OS: Mac
 - Toolchain zephyr sdk 0.16.3
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

nashif, 2023-12-14:
not able to reproduce

carriegong, 2023-12-18:
does  --coverage-basedir need to be in quotes?

carriegong, 2023-12-19:
seems like passing --gcov-tool gcov does the trick, but coverage report is blank

carriegong, 2023-12-20:
base directory still not setting. showing as '.'

nashif, 2024-01-30:
Sorry, I still can't reproduce this, please provide exact steps which lead to this error

````
i9:zephyr(topic/twister/predicate): ./scripts/twister --coverage --coverage-basedir tests/kernel/threads/thread_init/src/ -p qemu_cortex_m3  -c --coverage-tool gcovr  -T tests/kernel/threads/thread_init/
Deleting output directory /home/nashif/zephyrproject/zephyr/twister-out
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.5.0-4936-gc84df7ded81a
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/nashif/zephyrproject/zephyr/twister-out/testplan.json
INFO    - JOBS: 20
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    1/   1  100%  skipped:    0, failed:    0, error:    0
INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 1 of 1 test configurations passed (100.00%), 0 failed, 0 errored, 0 skipped with 0 warnings in 10.17 seconds
INFO    - In total 3 test cases were executed, 1 skipped on 1 out of total 675 platforms (0.15%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Generating coverage files...
INFO    - Using gcov tool: /home/nashif/bin/zephyr-sdk-0.16.4/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcov
INFO    - HTML report generated: /home/nashif/zephyrproject/zephyr/twister-out/coverage/index.html
INFO    - Saving reports...
INFO    - Writing JSON report /home/nashif/zephyrproject/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/nashif/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/nashif/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed

````

reopen with how you get the issue with the upstream tree please.