zephyrproject-rtos/zephyr #66528: upgrading cpp/c mixed project from 3.2 to 3.4 fail to build

eyalsol, 2023-12-14:
we are trying to upgrade the project, we work on, from ZephyrOS 3.2 to 3.4
the prj.conf in version 3.2 was:

#```
enable c++ support
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y
CONFIG_STD_CPP20=y
CONFIG_NEWLIB_LIBC=y

the prj.conf in version 3.4:

CONFIG_CPP=y
CONFIG_REQUIRES_FULL_LIBCPP=y
CONFIG_REQUIRES_FULL_LIBC=y
CONFIG_STD_CPP20=y

The Error we get:
in file included from ~/zephyr_workspace/zephyr/include/zephyr/types.h:11,
from ~/zephyr_workspace/zephyr/include/zephyr/kernel_includes.h:17,
from ~/zephyr_workspace/zephyr/include/zephyr/kernel.h:17,
from ~/zephyr_workspace/zephyr/arch/riscv/core/offsets/offsets.c:16:
~/tools/syntacore/sc-dt-2023.11/riscv-gcc/lib/gcc/riscv64-unknown-elf/12.2.1/include/stdint.h:9:16: fatal error: stdint.h: No such file or directory
9 | # include_next <stdint.h>
| ^~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.

we have tried other config parameters but it was still failing with other errors.
Any assistance will be appreciated

eyalsol, 2023-12-14:
duplicate of #66527 - closing

ndrs-pst, 2023-12-16:
@eyalsol I've been working on a mixed C/C++ project since Zephyr 3.2 and am currently using version 3.5. My configuration for the **ARM platform** (arm-zephyr-eabi) is as follows:
```
CONFIG_CPP=y
CONFIG_STD_CPP20=y
CONFIG_GLIBCXX_LIBCPP=y
CONFIG_NEWLIB_LIBC=y
CONFIG_NEWLIB_LIBC_NANO=y
```
Additionally, I was able to compile successfully using your prj.conf in Zephyr 3.5.
Hope this helps!

eyalsol, 2023-12-16:
@ndrs-pst  
thanks for tying to build it. 
1. I am using syntacore riscv toolchain
2. i wonder if you can set <functional> in cpp file and <stdint.h> in a c file and test if that work for you

Thanks

eyalsol, 2023-12-16:
I am trying with your configuration and get

```
~/demo/main.cpp:18:10: fatal error: functional: No such file or directory
   18 | #include <functional>
```

eyalsol, 2023-12-16:
Issue resolved.

the toolchain in use is syntacore: sc-dt-2023.11, riscv-gcc, riscv64-unknown-elf.

I had to explicitly set this in CMakeLists.txt:

`set(TOOLCHAIN_HAS_NEWLIB ON CACHE BOOL "True if toolchain supports newlib")
`
the prj.conf is set to:

```
CONFIG_CPP=y
CONFIG_REQUIRES_FULL_LIBCPP=y
CONFIG_NEWLIB_LIBC=y
CONFIG_STD_CPP20=y

```

closing the issue