zephyrproject-rtos/zephyr #66530: Emulators & simulators documentation issues

aescolar, 2023-12-14:
This is a list of Zephyr device emulators and simulators, as well as native_sim specific drivers  which lack documentation, or whose documentation is in a bad state, and other related documentation issues:

- [ ] ADC emulator:
  * No docs beyond source, and almost not mentioned.
  * References:
    * https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_ADC_EMUL
    * https://docs.zephyrproject.org/latest/build/dts/api/bindings/adc/zephyr,adc-emul.html
- [ ] CAN loopback driver
  * No docs beyond source, no mentions in docs.
  * References:
    * https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_CAN_LOOPBACK
    * https://docs.zephyrproject.org/latest/build/dts/api/bindings/can/zephyr,can-loopback.html
- [ ] CAN native Linux driver (connect to linux host CAN driver)
  * No docs beyond source, and almost not mentioned.
  * References:
    * https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_CAN_NATIVE_LINUX
    * https://docs.zephyrproject.org/latest/build/dts/api/bindings/can/zephyr,native-linux-can.html
- [ ] EEPROM *emulator*: (EPROM emulator on a flash partition):
  * No docs beyond source.
  * Mixed in other docs with the EEPROM *simulator*
  * References:
    * https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_EEPROM_EMULATOR
    * https://docs.zephyrproject.org/latest/build/dts/api/bindings/mtd/zephyr,emu-eeprom.html
- [ ] EEPROM *simulator*: (EPROM emulated on RAM, or (for native_sim also on a file in the host)):
  * No docs beyond source.
  * Mixed in other docs with the EEPROM *emulator*
  * References:
    * https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_EEPROM_SIMULATOR
    * https://docs.zephyrproject.org/latest/build/dts/api/bindings/mtd/zephyr,sim-eeprom.html
- [ ] Flash *simulator*: (FLASH emulated on RAM, or, for native_sim, on a file in the host):
  * docs: Only in the context of the [native_simulator](https://docs.zephyrproject.org/latest/boards/posix/native_sim/doc/index.html#peripherals) (this ignores that we can use it in any other platform)
  * References:
    * https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_FLASH_SIMULATOR
    * https://docs.zephyrproject.org/latest/build/dts/api/bindings/flash_controller/zephyr,sim-flash.html)
- [ ] GPIO emulator:
  * No docs, mention as pending in https://docs.zephyrproject.org/latest/hardware/emulator/index.html
  * References:
    * https://docs.zephyrproject.org/latest/build/dts/api/bindings/gpio/zephyr,gpio-emul.html
    * https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_GPIO_EMUL
- [ ] GPIO SDL emulator:
  * No docs
  * References:
    * https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_GPIO_EMUL_SDL
    * https://docs.zephyrproject.org/latest/build/dts/api/bindings/gpio/zephyr,gpio-emul-sdl.html
- [ ] I2C emulator:
  * Only docs in https://docs.zephyrproject.org/latest/hardware/emulator/index.html, maybe enough?
  * References:
    * https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_I2C_EMUL
    * https://docs.zephyrproject.org/latest/build/dts/api/bindings/i2c/zephyr,i2c-emul-controller.html#std-dtcompatible-zephyr-i2c-emul-controller
- [ ] Input SDL touch:
  * No docs
  * References:
    * https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_INPUT_SDL_TOUCH
    * https://docs.zephyrproject.org/latest/build/dts/api/bindings/input/zephyr,input-sdl-touch.html
- [ ] RTC emulator: Emulates an RTC with a timed workq item
  * No docs beyond source
  * References:
    * https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_RTC_EMUL
    * https://docs.zephyrproject.org/latest/build/dts/api/bindings/rtc/zephyr,rtc-emul.html
- [ ] SPI emulator:
  * Only docs in https://docs.zephyrproject.org/latest/hardware/emulator/index.html, maybe enough?
  * References:
    * https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_SPI_EMUL
    * https://docs.zephyrproject.org/latest/build/dts/api/bindings/spi/zephyr,spi-emul-controller.html

- [ ] [Emulators general docs](https://docs.zephyrproject.org/latest/hardware/emulator/index.html): 
  * The docs seem to refer mostly to bus emulators (I2C/SPI/GPIO) that would then have peripheral emulators, with focus on real peripheral HW emulators, and testing of their drivers
  * But the EPROM emulator (not of this type) and the CAN lopback driver are also mentioned.
  * => Probably this doc title and the content should be changed so it is clear it refers only to this type of bus emulators and its emulated peripherals.
  * This doc use .pngs with "native_posix" in them => The png should be converted to svg (they seem to be generated drawio), and the text corrected.

Maintainers: Feel free to correct and add to this list

CC: @kartben @henrikbrixandersen @bjarki-trackunit @Laczen @alexanderwachter @cfriedt @ceolin @jeremybettis @pdgendt @sjg20 @teburd @aaronemassey @fabiobaltieri @fgrandel

kartben, 2023-12-14:
Note re: PNG diagrams. Recreating them in SVG would certainly be great, it might be even better to have them directly in in-lined Graphviz, assuming getting a similar layout is not overly complicated. See e.g. https://github.com/zephyrproject-rtos/zephyr/blob/af0336ed19228e1350fcc7fe7de4321ffd0ac830/doc/contribute/documentation/generation.rst#L42-L66 or https://github.com/zephyrproject-rtos/zephyr/blob/af0336ed19228e1350fcc7fe7de4321ffd0ac830/doc/security/reporting.rst#L24-L51

cfriedt, 2023-12-14:
@aescolar - there is also dma_emul and uart_emul

aescolar, 2024-01-05:
CC @alevkoy  @niedzwiecki-dawid