zephyrproject-rtos/zephyr #66543: dhcp client is unable to acquire an address at times

chmorgan, 2023-12-14:
I'd appreciate input on how to further debug the issue.

**Describe the bug**
About half of the time at startup the dhcp client is unable to acquire an address.

dhcpv4 retries are unsuccessful in getting an address and Wireshark monitoring indicates that while in dhcpv4_send_discover() there are no packets observed from the system's Mac address during these retries.

Rebooting the system several times can resolve the issue, dhcp leases occur correctly 10% of the time.

** Configuration

Custom STM32H723.

**To Reproduce**

I'm unable to reproduce this behavior with the nucleo_h743zi development board and the dhcpv4_client example. Aligned my code with this example with no improvement in behavior.

**Expected behavior**
dhcp client should be able to acquire an address

**Impact**
Unable to consistently get a dhcp address.

**Logs and console output**
*** working case ***
```
[00:00:00.002,000] <dbg> net_dhcpv4: net_dhcpv4_init: (main): 
*** Booting Zephyr OS build zephyr-v3.5.0 ***
[00:00:00.002,000] <wrn> net_if: iface 0x24002840 is down
[00:00:00.002,000] <err> net_pkt: *** ERROR *** pkt 0x2400ef48 is freed already by igmp_send():108 (igmp_send_generic():280)
[00:00:00.002,000] <inf> dhcp: Starting dhcpv4 client...
[00:00:00.002,000] <inf> dhcp: Start on ethernet@40028000: index=1
[00:00:00.002,000] <dbg> net_dhcpv4: dhcpv4_start_internal: (main): iface 0x24002840 state=init
[00:00:00.002,000] <dbg> net_dhcpv4: dhcpv4_start_internal: (main): wait timeout=1s
[00:00:00.002,000] <dbg> net_dhcpv4: dhcpv4_set_timeout: (main): sched timeout dhcvp4=0x240028e8 timeout=1s
[00:00:00.002,000] <inf> main: registering shell command
[00:00:00.002,000] <dbg> net_dhcpv4: dhcpv4_manage_timers: (sysworkq): iface 0x24002840 dhcpv4=0x240028e8 state=init timeleft=1
[00:00:00.002,000] <dbg> net_dhcpv4: dhcpv4_timeout: (sysworkq): Waiting for 1s
[00:00:00.010,000] <dbg> net_dhcpv4: dhcpv4_iface_event_handler: (net_mgmt): Interface 0x24002840 coming up
[00:00:00.010,000] <dbg> net_dhcpv4: dhcpv4_immediate_timeout: (net_mgmt): force timeout dhcpv4=0x240028e8
[00:00:00.011,000] <dbg> net_dhcpv4: dhcpv4_manage_timers: (sysworkq): iface 0x24002840 dhcpv4=0x240028e8 state=init timeleft=0
[00:00:00.011,000] <dbg> net_dhcpv4: dhcpv4_enter_selecting: (sysworkq): enter state=selecting
[00:00:00.011,000] <dbg> net_dhcpv4: dhcpv4_set_timeout: (sysworkq): sched timeout dhcvp4=0x240028e8 timeout=4s
[00:00:00.011,000] <dbg> net_dhcpv4: dhcpv4_send_discover: (sysworkq): send discover xid=0x526224e8 timeout=4s
[00:00:00.011,000] <dbg> net_dhcpv4: dhcpv4_timeout: (sysworkq): Waiting for 4s
[00:00:00.011,000] <dbg> net_dhcpv4: dhcpv4_manage_timers: (sysworkq): iface 0x24002840 dhcpv4=0x240028e8 state=selecting timeleft=4
[00:00:00.011,000] <dbg> net_dhcpv4: dhcpv4_timeout: (sysworkq): Waiting for 4s
--- 7 messages dropped ---
[00:00:04.013,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]):   ciaddr=0.0.0.0
--- 16 messages dropped ---
[00:00:04.013,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]):   yiaddr=192.168.1.152
[00:00:04.013,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]):   siaddr=0.0.0.0
[00:00:04.013,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]):   giaddr=0.0.0.0]
[00:00:04.013,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_server_id: 192.168.1.1
[00:00:04.013,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_lease_time: 86039
[00:00:04.013,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_subnet_mask 255.255.255.0
[00:00:04.013,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_router: 192.168.1.1
[00:00:04.015,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_subnet_mask 255.255.255.0
[00:00:04.015,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_router: 192.168.1.1
[00:00:04.015,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): option unknown: 6
[00:00:04.015,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): option unknown: 44
[00:00:04.016,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): option unknown: 15
[00:00:04.016,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): option unknown: 125
[00:00:04.016,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_end
[00:00:04.016,000] <dbg> net_dhcpv4: dhcpv4_handle_reply: (rx_q[0]): state=requesting msg=ack
[00:00:04.016,000] <inf> net_dhcpv4: Received: 192.168.1.152
[00:00:04.016,000] <dbg> net_dhcpv4: dhcpv4_enter_bound: (rx_q[0]): enter state=bound renewal=43019s rebinding=75284s
[00:00:04.016,000] <dbg> net_dhcpv4: dhcpv4_set_timeout: (rx_q[0]): sched timeout dhcvp4=0x240028e8 timeout=43019s
[00:00:04.016,000] <inf> dhcp:    Address[1]: 192.168.1.152
[00:00:04.016,000] <inf> dhcp:     Subnet[1]: 255.255.255.0
[... 110 lines cut ...]
--- 31 messages dropped ---
[00:00:10.508,000] <dbg> net_pkt: net_pkt_skip: (sysworkq): pkt 0x2400fc48 skip 8
[00:00:10.508,000] <dbg> net_pkt: net_pkt_get_reserve_data_debug: (sysworkq): TDATA (?) [0] frag 0x2400df7c ref 1 (ethernet_fill_header():515)
[00:00:10.508,000] <dbg> net_pkt: net_pkt_frag_insert_debug: (sysworkq): pkt 0x2400fc48 frag 0x2400df7c (ethernet_fill_header:565)
[00:00:10.508,000] <dbg> net_pkt: net_pkt_read: (sysworkq): pkt 0x2400fc48 data 0x30004100 length 311
[00:00:10.508,000] <dbg> net_pkt: net_pkt_frag_unref_debug: (sysworkq): TDATA (?) [0] frag 0x2400df7c ref 0 (ethernet_remove_l2_header():597)
[00:00:10.508,000] <dbg> net_pkt: net_pkt_unref_debug: (sysworkq): TX [13] pkt 0x2400fc48 ref 0 frags 0x2400df28 (ethernet_send():742)
[00:00:10.508,000] <dbg> net_pkt: net_pkt_unref_debug: (sysworkq): TDATA (?) [0] frag 0x2400df28 ref 0 frags 0x2400df44 (ethernet_send():742)
[00:00:10.508,000] <dbg> net_pkt: net_pkt_unref_debug: (sysworkq): TDATA (?) [0] frag 0x2400df44 ref 0 frags 0x2400df60 (ethernet_send():742)
[00:00:10.508,000] <dbg> net_pkt: net_pkt_unref_debug: (sysworkq): TDATA (?) [0] frag 0x2400df60 ref 0 frags (nil) (ethernet_send():742)
[00:00:10.508,000] <dbg> net_pkt: net_pkt_frag_unref_debug: (sysworkq): TDATA (?) [0] frag 0x2400df28 ref 0 (net_pkt_unref_debug():597)
[00:00:10.508,000] <dbg> net_dhcpv4: dhcpv4_set_timeout: (sysworkq): sched timeout dhcvp4=0x240028e8 timeout=15s
[00:00:10.509,000] <dbg> net_dhcpv4: dhcpv4_send_discover: (sysworkq): send discover xid=0x73a6d41a timeout=15s
[00:00:10.509,000] <dbg> net_dhcpv4: dhcpv4_timeout: (sysworkq): Waiting for 15s
[00:00:10.509,000] <dbg> net_dhcpv4: dhcpv4_manage_timers: (sysworkq): iface 0x24002840 dhcpv4=0x240028e8 state=selecting timeleft=15
[00:00:10.509,000] <dbg> net_dhcpv4: dhcpv4_timeout: (sysworkq): Waiting for 15s
```

**Environment (please complete the following information):**
 - OS: OSX
 - Toolchain: zephyr-idk-0.16.4
 - Zephyr v3.5.0

**Additional context**
Nothing yet.

rlubos, 2024-01-10:
From the logs it seems that DHCPv4 service seems to work fine, as it tries to resend Discover over time, so it seems that it might not be an issue with DHCP itself, but rather something lower level, especially given that you say that packets does not end up on the wire.

It'd be hard to pinpoint the problem out of the blue, but personally I'd check 2 things first:
* I don't know what your application does, but perhaps some component hogs the TX net buffers? Note that Ethernet L2 needs to allocate one extra net_buf for the L2 header, if that fails, the packet won't be transmitted. Would be good to see `net mem` to see the net pkt/buf configuration and actual usage (with `CONFIG_NET_BUF_POOL_USAGE` enabled).
* If that's not the case, perhaps there's some issue with the driver/actual hardware, failed initialization or so? Maybe check if you can transmit anything at all - for example you could confiigure IP address manually in case DHCP fails (`net ipv4 add 1 192.168.1.152 255.255.255.0`, update the IP address accordingly) and then try to ping the host?

carlescufi, 2024-02-07:
Closing due to inactivity. @chmorgan please reopen if this is not fixed for you.

closed 2024-02-07 by carlescufi

ashutoshpandey-eaton, 2025-01-06:
@rlubos  @erwango Similar type of problem I am getting:
1.Our client is able to get the IP from mobile hotspot (acting as WIFI-AP)  [
[working_log.log](https://github.com/user-attachments/files/18319889/working_log.log)

2.**Our client is not able to get the IP from any router (acting as WIFI-AP): NOTE: dhcp DISCOVER message is not seen on Wireshark, in this case.**
[not_working_log.log](https://github.com/user-attachments/files/18319892/not_working_log.log)

In the non- working case it is seen in the logs [not_working_log.log] **dhcp client tries to resend Discover over time** however same DISCOVER message in not seen in the Wireshark, only it received ARP packet "net_core: process_data: (0x24003028): Packet 0x2402d028 discarded by L2"

Using:
zephyr OS : 3.6.0
Host board(client): nucleo-h753zi 

What could be the cause (by looking at the logs) , in one case dhcp client is working while in another case its not working??

erwango, 2025-01-06:
@ashutoshpandey-eaton Please don't post on closed tickets, but open new ones instead, or alternately [discussions](https://github.com/zephyrproject-rtos/zephyr/discussions). Also please note that V3.6.0 will not be supported anymore shortly, so it would be nice if you can confirm that the issue is still visible on V3.7.0