zephyrproject-rtos/zephyr #66546: Adding Xtensa MPU Support

dcpleung, 2023-12-14:
Since we already have MMU support for Xtensa, we need MPU to complete the circle.

jcmvbkbc-issues, 2023-12-14:
It should be possible. I have a couple LX configurations with MPU, the most relevant seems to be the sample_controller version configured with MPU. I can submit it for QEMU 9 and publish its overlay so that you could make a toolchain and QEMU for it.

dcpleung, 2023-12-14:
That's awesome. Thank you very much.

jcmvbkbc-issues, 2023-12-15:
I've pushed the original and the preprocessed overlays for the sample_controller32 here: https://github.com/jcmvbkbc/xtensa-toolchain-build/commit/435c5e8b108de520565886f3f2349cc6be3fe712
and QEMU branch with this core and a couple minor issues fixed is available here: https://github.com/OSLL/qemu-xtensa/commits/xtensa-9.0-mpu/

closed 2024-03-20 by nashif

dcpleung, 2024-04-02:
@jcmvbkbc Is it possible to have a MPU capable SoC configuration for Zephyr? We wish to incorporate testing of the MPU code in CI, which means the overlay and QEMU patches will need to be publicly available (even if we are to keep them with Zephyr trees).