zephyrproject-rtos/zephyr #66547: wio-terminal board does not support PWM?

maxhbr, 2023-12-14:
Hey,

I am currently looking into the wio-terminal, which is listed as [having support for pwm](https://github.com/zephyrproject-rtos/zephyr/blob/29b19bcf8bc8f97069be00a9012c4cf1b7dbfb8b/boards/arm/wio_terminal/doc/index.rst?plain=1#L88-L90). But the devicetree is built as follows:

1. dts: [boards/arm/wio_terminal/wio_terminal.dts](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/wio_terminal/wio_terminal.dts)
2. via `#include <atmel/samd5xx19.dtsi>` this includes: [dts/arm/atmel/samd5xx19.dtsi](https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/arm/atmel/samd5xx19.dtsi)
3. via `#include <atmel/samd5x.dtsi>` this includes: [dts/arm/atmel/samd5x.dtsi](https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/arm/atmel/samd5x.dtsi)

and in the file [samd5x.dtsi](https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/arm/atmel/samd5x.dtsi) all nodes `tcc*` are `compatible = "atmel,sam0-tcc";` instead of `"atmel,sam0-tcc-pwm"` which, as far as I understand means that they do not support PWM?

kartben, 2023-12-14:
@joelguittet

joelguittet, 2023-12-15:
Hello

PR done to integrate the wio terminal: https://github.com/zephyrproject-rtos/zephyr/pull/54953

Obviously I haven't indicate the PWM was tested. We should verify the datasheet but I will be surprise if this MCU has no PWM at all.

Have you tried to modify the TCC compatible value ? With an overlay for example.

Joel

maxhbr, 2023-12-16:
> We should verify the datasheet but I will be surprise if this MCU has no PWM at all.

The [documentation of the IC](https://ww1.microchip.com/downloads/aemDocuments/documents/MCU32/ProductDocuments/DataSheets/SAM-D5x-E5x-Family-Data-Sheet-DS60001507.pdf) lists
> * PWM Modes using TC and TCC peripherals:
>   * Up to six PWM channels on each 24-bit TCC
>   * Up to three PWM channels on each 16-bit TCC
>   *  Up to two PWM channels on each 16-bit TC

But I am sadly not smart enough to extract the necessary parameter from the documentation (I think page 40ff). My current draft is (with wrong numbers) is:

```
&tcc0 {
	compatible = "atmel,sam0-tcc-pwm";
	prescaler = <8>;
	#pwm-cells = <2>;
};

&tcc1 {
	compatible = "atmel,sam0-tcc-pwm";
	prescaler = <8>;
	#pwm-cells = <2>;
};

&tcc2 {
	compatible = "atmel,sam0-tcc-pwm";
	prescaler = <8>;
	#pwm-cells = <2>;
};

&tcc3 {
	compatible = "atmel,sam0-tcc-pwm";
	prescaler = <8>;
	#pwm-cells = <2>;
};

&tcc4 {
	compatible = "atmel,sam0-tcc-pwm";
	prescaler = <8>;
	#pwm-cells = <2>;
};
```

joelguittet, 2023-12-17:
Your draft looks not bad, should check the details (there are other samd board in zephyr to compare)

However, I don t think you should configure all TCC this way : is it suitable even if you don't want to use it as PWM ? (Default configuration should not be PWM except if the pin is dedicated to something that need it, brightness for example why not).

maxhbr, 2023-12-18:
The pin `PC27` is on https://wiki.seeedstudio.com/Wio-Terminal-IO-Overview/ listed as `PWM0`. My goal would be to get it online.

maxhbr, 2023-12-19:
The pin `PC27` does not seem to have a TC or TCC connected to it:
[image]

I think I will switch to `PWM3`, which is also `BCM13, D6, A6`. And it is related to TCC0:
[image]
But I have no idea how to choose IOSET 3.

joelguittet, 2024-01-02:
Hello @maxhbr 
I'm on it to try making the blinky_pwm sample working on it using the USER LED.
Have you made any progress on your side too ?
Joel

maxhbr, 2024-01-02:
Hey, sorry, I sadly made no progress on the PWM front as the related issue https://github.com/zephyrproject-rtos/zephyr/issues/67049 (maybe, as you have a wio terminal in hand, you could also verify if you see the same behaviour)

joelguittet, 2024-01-02:
Yes I'm on both tickets actually

joelguittet, 2024-01-02:
Got it. I will make a PR soon.

joelguittet, 2024-01-02:
@maxhbr here is the PR: https://github.com/zephyrproject-rtos/zephyr/pull/67126

Can you check on your side if this is working ? Build the blinky_pwm sample and run this on the board. You should have the Blue USER LED near to the USB connector blinking. First slowly, and then faster.

Details to permit you modification/adaptation for your own use case:
- PA15 is used and set at the `pwm_default` entry in the dts. PA15 has PWM using TCC1 channel 3 (PA15G_TCC1_WO3 defined at https://github.com/zephyrproject-rtos/hal_atmel/blob/b0cc87f2a494d8b3c383292d1d6c7d10323932bd/include/dt-bindings/pinctrl/samd51p-pinctrl.h#L593).
- `tcc1` is used and compatible value is changed in the dts to enable PWM support
- `pwm_led0` is declared with TCC1 channel 3 in the dts

You should be able to choose another pin with PWM support and adapt the right tcc/channel :)

closed 2024-01-15 by nashif