zephyrproject-rtos/zephyr #66556: New config: "DEVELOPMENT" as an optional stage before "EXPERIMENTAL"

koffes, 2023-12-15:
## Introduction

This issue proposes to add a new config, for the time being coined `DEVELOPMENT`, which acts as an optional config a developer can use before switching to the `EXPERIMENTAL` config.

In SW development, key to productivity is avoiding stale code and time spent on merge conflicts by use of short lived, small branches. I.e. making small increments of functionality and getting them into main. This also reduces the size and complexity of PRs. 
(I will not go into depth as this is mostly well known).

### Problem description
I quite often witness large, complex additions to code bases as long PRs. This then followed by talking more time to get reviews/reviewers and poorer review quality.
This again I believe stems from that users/customers start using anything that is available to them, despite there being a "EXPERIMENTAL" tag on it. Thus, the developer wants to fully complete the implementation of feature A, without splitting into to chunks.

### Proposed change
A "DEVELOPMENT" tag could do the following:

- Let the developer tag a finished subset of feature A (A1) with `DEVELOPMENT`. This code piece is ready, but incomplete, and may not work on its own.
- This `DEVELOPMENT=y` will throw an error on build - It should, as the unfinished feature A is hidden behind a `kconfig enabled=n` which is never turned on by default.
- There may be an `ALLOW_DEVELOPMENT` global kconfig which can be set to `y`. This will let the build pass, but print a yellow warning.
- There must be accompanying documentation for this kconfig which _clearly_ states that this may be untested and volatile code.

Once the developer has merged A1, A2 and A3, there can be a separate PR which changes `DEVELOPMENT` to `EXPERIMENTAL` which transitions the code over to the next stage before being in production, and finally after X time, `DEPRECATED`.

https://github.com/zephyrproject-rtos/zephyr/blob/main/Kconfig.zephyr#L812C6-L861

### Concerns and Unresolved Questions

I am very open to discuss if this only moves the problem rather than solving it.

## Alternatives

An alternative is to use the `EXPERIMENTAL` config for the same thing, but I do not feel this gives the developer the required safety net. This just prints a small warning and lets users continue using it without properly warning them.

jukkar, 2023-12-18:
I am not sure we should merge code under development into main. We have the collab branches for that purposes as documented here https://docs.zephyrproject.org/latest/project/code_flow.html

keith-zephyr, 2024-06-26:
Process WG consensus: EXPERIMENTAL tag is sufficient.  There are no guarantees about API compatibility with EXPERIMENTAL tag.  DEVELOPMENT tag doesn't provide enough additional benefit to complicate the API lifecycle.

closed 2024-06-26 by keith-zephyr