zephyrproject-rtos/zephyr #66570: bug: stamp_c3 board has incorrect `CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC` in defconfig

Yaxit, 2023-12-15:
**Describe the bug**
k_msleep() in this code only waits a fraction of a second.
```
int main(void){
    int i = 0;
    while(1){
        printk("Hello World %u %d! %s\n",CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC, i, CONFIG_BOARD);
        k_msleep(1000);
        i++;
    }
}
```

When overriding this value with 16000000 (default value if I comment away the config in the def_conf file of the board) in my prj.conf, it works as intended.
I'm not 100% sure I understand to what this config refers to (the MCU should be running at 160MHz, what clock is this exactly)
@soburi @carlescufi could I be missing something here?

**Expected behavior**
Should print every one second.

Zephyr revision: 4d76e26f173060d0fd0c9ccbed1e617fff615434
Platform: Windows
Blame: https://github.com/zephyrproject-rtos/zephyr/blame/ed040f187375cea20b5cceab2a3697a5e305980b/boards/riscv/stamp_c3/stamp_c3_defconfig#L8

**Impact**
Simple example does not work as intended.

sylvioalves, 2023-12-15:
Indeed `stamp_c3` is redefining HW clock cycles, causing this.

ndrs-pst, 2023-12-16:
@sylvioalves, you're right. One of my concerns when changing the clock frequency is the need to update both the clock-frequency in the DTS and CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC in prj.conf.

I'm not certain if it's possible to do a compile-time check within a source file (perhaps a dedicated dummy file for compile-time checks to avoid cluttering the include files).
Here's an idea:
```
#if (CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC != DT_PROP(DT_PATH(cpus, cpu_0), clock_frequency))
#error "SYS_CLOCK_HW_CYCLES_PER_SEC must match with CPU clock!"
#endif
```

Another approach, seen in other SoCs like atmel_sam, atmel_sam0, and gigadevice, is to define the default value in Kconfig.defconfig as the same as in the DTS:
```
config SYS_CLOCK_HW_CYCLES_PER_SEC
	default	$(dt_node_int_prop_int,/cpus/cpu@0,clock-frequency)
```

Yaxit, 2023-12-20:
@ndrs-pst I don't feel very familiar with the whole zephyr architecture just yet, but why would we need to update the DTS file as well?
If I understand, `SYS_CLOCK_HW_CYCLES_PER_SEC` is defined by the [SoC Kconfig](https://github.com/zephyrproject-rtos/zephyr/blob/a6fda002549f0be5fe928d7a6762abd0f262984e/soc/riscv/espressif_esp32/common/Kconfig.defconfig.series#L27C8-L27C35) and defaults to 16MHz.

By the way, since I will be working with this platform for a few weeks, let me know if a pull request would be welcomed.
In this time I will be trying to integrate ESP-NN accelerated kernels in TensorFlow Lite (it currently seems to be running locally), do you know if anyone should be involved on this topic?

ndrs-pst, 2023-12-20:
@Yaxit The CPU clock frequency setting depends on the SoC initialization code. As you can see, the actual values for setting the CPU clock frequency are defined by [CONFIG_ESP32C3_DEFAULT_CPU_FREQ_MHZ](https://github.com/zephyrproject-rtos/hal_espressif/blob/e98ce93e916eebdc4a0c5bceef864289447e3d63/components/esp_system/port/soc/esp32c3/clk.c#L123C71-L123C71) and [ESP_SOC_DEFAULT_CPU_FREQ_MHZ](https://github.com/zephyrproject-rtos/hal_espressif/blob/e98ce93e916eebdc4a0c5bceef864289447e3d63/zephyr/esp_shared/include/stubs.h#L27-L38).

These settings are not directly related to `SYS_CLOCK_HW_CYCLES_PER_SEC`.
However, higher-level functions, such as `k_msleep()`, depend on this value. If a different value is used, it could lead to issues, as you have now confirmed.

About TensorFlow Lite, I'm not sure maybe someone who involved in this folder?
https://github.com/zephyrproject-rtos/zephyr/commits/main/modules/tflite-micro

soburi, 2023-12-22:
I haven't tested it yet, but
https://github.com/zephyrproject-rtos/zephyr/commit/cff71c8d2b5b31c1e8d7a62371817da9177e002d

It looks like the following line should have been deleted along with this correction.
https://github.com/zephyrproject-rtos/zephyr/blob/f688532412432987682716de60bdf17f2afd4b38/boards/riscv/stamp_c3/stamp_c3_defconfig#L8

I think it will work if you delete it, but I haven't tested it yet.
(I created stamp_c3 mostly based on esp32c3_devkitm.)

ndrs-pst, 2023-12-22:
@soburi Looks like that's the way. So the PR from you can close this issue 😉.

Yaxit, 2023-12-23:
@soburi yes I did test it, it works when the line is removed.

sylvioalves, 2023-12-26:
Is that a PR to fix this already?

soburi, 2023-12-27:
> Is that a PR to fix this already?

Yes, The #66922 fixed this issue.

PR #66922 (2024-01-08): boards: riscv: stamp_c3: Update systimer configuration
The esp32c3 systimer has been fixed in PR #53453, but stamp_c3 was not included,
so we will update it to catch up with this fix

Fix: #66570
closed 2024-01-08 by carlescufi