zephyrproject-rtos/zephyr #66576: NXP mimxrt1020_evk Ethernet issue

DerekSnell, 2023-12-15:
This issue was originally reported by @danielwippermann in this [Discussion](https://github.com/zephyrproject-rtos/zephyr/discussions/63389#discussioncomment-7839157).  But I have replicated the issue as described, and confirmed this is a bug.

> On the MIMXRT1020_EVK, I can only get the dhcpv4_client sample to work if I had previously flashed and run an ethernet example using MCUXpresso and then (without pushing the RESET button or power-cycling the hardware) west flash-ing the dhcpv4_client sample on the same board. Then it runs perfectly fine, fetches an IP and can be pinged. If I then push RESET or power-cycle the board, it shows the same behaviour as the OP reported: it gets stuck attempting to contact the DHCPv4 server:

> Working example:
```
*** Booting Zephyr OS build zephyr-v3.5.0-2951-gbd227e19cff9 ***
[00:00:00.501,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:00.501,000] <inf> net_dhcpv4_client_sample: Start on ethernet@402d8000: index=1
[00:00:02.502,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
[00:00:08.052,000] <inf> net_dhcpv4: Received: 192.168.180.245
[00:00:08.052,000] <inf> net_dhcpv4_client_sample:    Address[1]: 192.168.180.245
[00:00:08.052,000] <inf> net_dhcpv4_client_sample:     Subnet[1]: 255.255.255.0
[00:00:08.053,000] <inf> net_dhcpv4_client_sample:     Router[1]: 192.168.180.1
[00:00:08.053,000] <inf> net_dhcpv4_client_sample: Lease time[1]: 600 seconds
uart:~$
```
> After pushing the RESET button or power-cycling the board:
```
*** Booting Zephyr OS build zephyr-v3.5.0-2951-gbd227e19cff9 ***
[00:00:00.501,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:00.501,000] <inf> net_dhcpv4_client_sample: Start on ethernet@402d8000: index=1
[00:00:02.502,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
uart:~$ net iface

Interface eth0 (0x80000ba8) (Ethernet) [1]
===================================
Link addr : XX:XX:XX:XX:XX:XX
MTU       : 1500
Flags     : AUTO_START,IPv4
Device    : ethernet@402d8000 (0x600173e8)
Ethernet capabilities supported:
	Virtual LAN
	Auto negotiation
	10 Mbits
	100 Mbits
IPv4 unicast addresses (max 1):
	<none>
IPv4 multicast addresses (max 2):
	224.0.0.1
IPv4 gateway : 0.0.0.0
IPv4 netmask : 0.0.0.0
DHCPv4 lease time : 0
DHCPv4 renew time : 0
DHCPv4 server     : 0.0.0.0
DHCPv4 requested  : 0.0.0.0
DHCPv4 state      : selecting
DHCPv4 attempts   : 69
uart:~$
```
I reproduced the issue using the same sample with Zephyr v3.5.0.  To get the Zephyr sample to work, I first programmed with the [MCUXpresso SDK lwip_dhcp](https://github.com/nxp-mcuxpresso/mcux-sdk-examples/tree/main/evkmimxrt1020/lwip_examples/lwip_dhcp/bm) example.  It seems the Zephyr app is not initializing something which is initialized by the MCUXpresso app.  

I also found reverting to Zephyr v3.4.0 resolves the issue, so this appears to be a regression issue.

henrikbrixandersen, 2023-12-19:
@DerekSnell Did you try bisecting this?

DerekSnell, 2023-12-19:
Hi @henrikbrixandersen ,
I have tried bisecting a couple times, but it has not yet helped me find a commit causing this issue.  Unfortunately, I found this issue is intermittent.  It has been easy for me to replicate the issue using v3.5.  And I have not yet seen the app fail using v3.4.  But when bisecting, some commits run the app several times successfully before it fails.  One commit succeeded 10 power cycles before I saw it fail.  The commits found by bisecting were trivial, and should not impact this app.  So I suspect I reported to bisect some commits as good, when they still have the issue.

It seems it will take more work to track this down.

danielwippermann, 2023-12-22:
Hi everone,

I did some manual bisecting (since the automatic one tends to land on issues that were fixed shortly after) and have found one commit where I can repeatedly switch back and forth around to make that problem appear and disappear:

- v3.5.0-rc1-165-g147cef6660f2: works
- v3.5.0-rc1-166-ge032f9520dfe: fails

The commit in question is e032f9520dfe545eb21d28cdd270d3b6baa26383 . But I cannot really tell why that should have the effect I observe.

I hope this info helps someone narrowing it down.

Thanks for all your help!

DerekSnell, 2023-12-26:
Hi @danielwippermann ,
Thanks for sharing your findings.  I must admit, debugging this issue has been frustrating.  I do see that different builds make a difference.  But I believe it is differences in timing causing the different behaviors.  And I do not think the changes in https://github.com/zephyrproject-rtos/zephyr/commit/e032f9520dfe545eb21d28cdd270d3b6baa26383 are the root cause of this issue.

I spent a few hours working on this issue today, so first let me share some of my observations.  When I first started working on this today, https://github.com/zephyrproject-rtos/zephyr/commit/e032f9520dfe545eb21d28cdd270d3b6baa26383 worked fine for me.  I reset and power cycled the board more than 20 times, and the DHCP sample worked great for me.  

Thinking the issue was still intermittent, I modified the DHCP sample, and added some code that reboots the MCU if the DHCP address is received.  But testing with that change, I found I broke the sample.  I also realized commit https://github.com/zephyrproject-rtos/zephyr/commit/e032f9520dfe545eb21d28cdd270d3b6baa26383 was no longer working for me.  It consistently failed, and for the rest of the day I did not see it succeed.  

But since my modified sample also failed, I reduced my changes to a single line of code that also breaks the sample.  At this point, commit https://github.com/zephyrproject-rtos/zephyr/commit/147cef6660f2f368457447a0fdc94ea06ce268dd always works for me, like you found.  But if I apply [0001-add-extra-LOG_INF.patch](https://github.com/zephyrproject-rtos/zephyr/files/13773820/0001-add-extra-LOG_INF.patch) to https://github.com/zephyrproject-rtos/zephyr/commit/147cef6660f2f368457447a0fdc94ea06ce268dd, the sample always fails.  This patch just adds the line below:
```
LOG_INF("Booting attempt  \n");
```
So I do see different builds make a difference.  But the changes in https://github.com/zephyrproject-rtos/zephyr/commit/e032f9520dfe545eb21d28cdd270d3b6baa26383 and in my patch do not seem like they should break this sample.  But I also see differences using the same exact source code, like today I saw https://github.com/zephyrproject-rtos/zephyr/commit/e032f9520dfe545eb21d28cdd270d3b6baa26383 work many times for me, until it stopped working.  And I did not even rebuild that commit when it started failing for me.  I was saving off different build directories so I could easily flash previous builds without rebuilding.  Therefore, I think something changed in my testing environment, like the ambient temperature, which impacted the sample timing slightly, and caused the issue to manifest.

Unfortunately, I think this will take more debugging to find the root cause of this issue.  Thanks again for your help with this.

nxpadamm, 2024-06-04:
Hi everyone,

Using a recent commit from main, I can't seem to be able to reproduce this issue. I've tried various combinations of reset, power cycling, debugger attached and detached. Issue possibly got fixed by the recent enet driver rewrite, it's quite an old issue actually.

Console logs for reference:

```
[00:00:03.837,000] <inf> phy_mc_ksz8081: PHY 0 is up
[00:00:03.837,000] <inf> phy_mc_ksz8081: PHY (0) Link speed 100 Mb, full duplex

*** Booting Zephyr OS build ca19e6bbaa54 ***
[00:00:03.841,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:03.841,000] <inf> net_dhcpv4_client_sample: Start on ethernet: index=1
[00:00:11.848,000] <inf> net_dhcpv4: Received: 192.168.0.229
[00:00:11.848,000] <inf> net_dhcpv4_client_sample:    Address[1]: 192.168.0.229
[00:00:11.848,000] <inf> net_dhcpv4_client_sample:     Subnet[1]: 255.255.255.0
[00:00:11.848,000] <inf> net_dhcpv4_client_sample:     Router[1]: 192.168.0.1
[00:00:11.848,000] <inf> net_dhcpv4_client_sample: Lease time[1]: 7200 seconds
uart:~$ 
uart:~$ net ping 192.168.0.33
PING 192.168.0.33
28 bytes from 192.168.0.33 to 192.168.0.229: icmp_seq=1 ttl=64 time=1 ms
28 bytes from 192.168.0.33 to 192.168.0.229: icmp_seq=2 ttl=64 time=0 ms
28 bytes from 192.168.0.33 to 192.168.0.229: icmp_seq=3 ttl=64 time=0 ms
```

decsny, 2024-06-04:
> Hi everyone,
>
> Using a recent commit from main, I can't seem to be able to reproduce this issue. I've tried various combinations of reset, power cycling, debugger attached and detached. Issue [...]

My understanding was that the issue only happens sometimes once every few hundred tries

decsny, 2024-07-01:
no longer appears to be an issue in 3.7 rc2

closed 2024-07-01 by decsny