zephyrproject-rtos/zephyr #66578: timer: kernel: cortex_m_systick timer jumps around in time

peter-mitsis, 2023-12-15:
**Describe the bug**
In a simple loop that repeatedly calls k_cycle_get_32(), I am observing multiple instances where a call jumps back in time. That is, consecutive calls to k_cycle_get_32() are not always returning increasing values.

This observation is **not** due to a 32-bit rollover.

I have observed this behaviour on both the frdm_k64f hardware and the mps2_an385 qemu platforms. When this _rewind_ occurs, the difference in the cycle counts is a little bit more than 0xf8000000. This number is significant because the cortex M systick timer is a 24-bit counter. This strongly hints that the cause is related to the rollover/reload of the timer's counter.

**To Reproduce**

Create a test project that contains the following test code ...

```
#include <zephyr/kernel.h>

int main(void)
{
        uint32_t start;
        uint32_t finish;

        while (1) {
                start = k_cycle_get_32();
                finish = k_cycle_get_32();

                if (start > finish) {
                        printk("[start:finish]=[%08u:%08u]\n",
                               start, finish);
                        printk("    *** Diff = %x\n", finish - start);
                }
        }

        return 0;
}
```
west build -p -b mps2_an385 _\<path to test project above\>_ -t run
    or
west build -p -b frdm_k64f _\<path to test project above\>_
west flash

**Expected behavior**
Successive calls to k_cycle_get_32() should return values that keep increasing (except for the 32-bit wraparound case).

**Impact**
As it currently stands, Zephyr can neither reliably report nor track cycle usage on platforms that use the cortex M systick timer driver.

**Logs and console output**

Sample output from running this test on the frdm_k64f board ...

```
*** Booting Zephyr OS build zephyr-v3.5.0-3148-gbf2bdcf6f79c ***
[start:finish]=[237440594:103224083]
    *** Diff = f80004c1
[start:finish]=[4285691618:4151475107]
    *** Diff = f80004c1
[start:finish]=[4038975341:3904758830]
    *** Diff = f80004c1
[start:finish]=[3792259044:3658042533]
    *** Diff = f80004c1
[start:finish]=[3672259713:3538043202]
    *** Diff = f80004c1
[start:finish]=[3298826515:3164610009]
    *** Diff = f80004c6

```

**Environment (please complete the following information):**
SHA: bf2bdcf6f79cf9711b1fdee3f2d91a5e61603f29  (Mid december 2023)
Zephyr SDK version: 0.16.3

gramsay0, 2023-12-16:
I'm not sure if that is a valid configuration.

For `mps2_an385` the 24bit 25MHz clock wraps every 2^24/25000000 = 0.67 seconds.

Setting `CONFIG_SYS_CLOCK_TICKS_PER_SEC=1` means 1 tick is 1.5 wraps of the system clock.

This calculation falls apart:
```
#define COUNTER_MAX 0x00ffffff

#define CYC_PER_TICK (sys_clock_hw_cycles_per_sec()	\
		      / CONFIG_SYS_CLOCK_TICKS_PER_SEC)
#define MAX_TICKS ((k_ticks_t)(COUNTER_MAX / CYC_PER_TICK) - 1)
#define MAX_CYCLES (MAX_TICKS * CYC_PER_TICK)
```
i.e. `MAX_TICKS` becomes `-1`.

There should be a static assert in there at least

henrikbrixandersen, 2024-01-02:
> I'm not sure if that is a valid configuration.

@peter-mitsis Any thoughts on this?

peter-mitsis, 2024-01-02:
Just got back from a holiday break. Getting familiar with this again ...

nashif, 2024-03-12:
@peter-mitsis  can you take another look?

ithinuel, 2024-06-26:
@peter-mitsis I tried to replicated the issue with `CONFIG_SYS_CLOCK_TICKS_PER_SEC=3` (as `CONFIG_SYS_CLOCK_TICKS_PER_SEC=2` would also yield an invalid `MAX_TICKS` value of 0) but could not get errors to show.

The condition for the printk was met a few times because of wraparound (difference typically ~0x3d).

Are you still experiencing this issue ? (note there had been a fix for a similar issue back in https://github.com/zephyrproject-rtos/zephyr/pull/63344)

ZhiyuanTang17, 2024-07-23:
I also encounter this issue when I porting my SoC.
Our SoC do has a “ticked” or “dumb” counter driver.
And when I run kernel\timer\timer_monotonic test, I will encounter:

`START - test_timer
k_ticks_to_cyc_floor32(1) = 32
sys_clock_hw_cycles_per_sec() = 32000
diff = -31 (t_last = 3967 : t_now = 3936);i = 175
diff = -31 (t_last = 8063 : t_now = 8032);i = 3821
diff = -31 (t_last = 9791 : t_now = 9760);i = 5300
diff = -31 (t_last = 10271 : t_now = 10240);i = 5674
diff = -31 (t_last = 11519 : t_now = 11488);i = 6687
diff = -31 (t_last = 11743 : t_now = 11712);i = 6860
diff = -31 (t_last = 12479 : t_now = 12448);i = 7464
diff = -31 (t_last = 15519 : t_now = 15488);i = 10033
diff = -31 (t_last = 17535 : t_now = 17504);i = 11938
diff = -31 (t_last = 17759 : t_now = 17728);i = 12054
diff = -31 (t_last = 19007 : t_now = 18976);i = 13119
diff = -31 (t_last = 19999 : t_now = 19968);i = 13899
diff = -31 (t_last = 21247 : t_now = 21216);i = 14964
diff = -31 (t_last = 23519 : t_now = 23488);i = 16869`

Seems the same fault: NOW_time>OLD_time.

gramsay0, 2024-07-23:
@ZhiyuanTang17 are you using the cortex_m_systick or some other system timer?

AFAICT there are no issues with cortex_m_systick jumping back in time when given a sensible configuration.

@ithinuel since the PR to add a static assert to detect misconfiguration has gone stale, do you think it is acceptable to just close this issue?

ithinuel, 2024-07-23:
@gramsay0  I think it is.

@ZhiyuanTang17 If you’re seeing a similar issue, please report in a new ticket because it is likely unrelated.

closed 2024-07-23 by ithinuel