zephyrproject-rtos/zephyr #66607: Lwm2m: blockwise transfer: device responds with block 0 instead of the correct block

marco85mars20, 2023-12-18:
**Describe the bug**
The following tcpdump shows the Leshan server requesting block 6 and correctly receives the requested block. However, then when requesting block 7, the device responds with block 0 (not sure if this is the correct term).
```
21:09:39.154676 IP6 2400:7660:ffff:ffff::1.5683 > 2400:7660:ffff:28:f43:d53f:b67b:cdb7.35860: UDP, length 25
	0x0000:  6004 8e56 0021 113f 2400 7660 ffff ffff  `..V.!.?$.v`....
	0x0010:  0000 0000 0000 0001 2400 7660 ffff 0028  ........$.v`...(
	0x0020:  0f43 d53f b67b cdb7 1633 8c14 0021 c1cd  .C.?.{...3...!..
	0x0030:  4801 c677 5802 246a 84ce f0ec b532 3632  H..wX.$j.....262
	0x0040:  3432 0130 622d 1661 64                   42.0b-.ad
21:09:39.159440 IP6 2400:7660:ffff:ffff::1.5683 > 2400:7660:ffff:28:f43:d53f:b67b:cdb7.35860: UDP, length 4
	0x0000:  6004 8e56 000c 113f 2400 7660 ffff ffff  `..V...?$.v`....
	0x0010:  0000 0000 0000 0001 2400 7660 ffff 0028  ........$.v`...(
	0x0020:  0f43 d53f b67b cdb7 1633 8c14 000c ce24  .C.?.{...3.....$
	0x0030:  6000 91c9                                `...
21:09:39.348508 IP6 2400:7660:ffff:28:f43:d53f:b67b:cdb7.35860 > 2400:7660:ffff:ffff::1.5683: UDP, length 274
	0x0000:  6000 0000 011a 1140 2400 7660 ffff 0028  `......@$.v`...(
	0x0010:  0f43 d53f b67b cdb7 2400 7660 ffff ffff  .C.?.{..$.v`....
	0x0020:  0000 0000 0000 0001 8c14 1633 011a fa57  ...........3...W
	0x0030:  6845 c677 5802 246a 84ce f0ec c22d 16b1  hE.wX.$j.....-..
	0x0040:  6cff 88f5 f867 b7bd c3c5 2d5d c0d0 1d28  l....g....-]...(
	0x0050:  2028 03e3 f9ed 4a72 3914 1256 a061 4016  .(....Jr9..V.a@.
	0x0060:  aebe 58e3 4f6a 00ab 40cd 9b28 d2ef 4b94  ..X.Oj..@..(..K.
	0x0070:  b644 b174 6a17 6119 2e86 36da 6802 41c5  .D.tj.a...6.h.A.
	0x0080:  b7d4 d211 6e14 f3ed 1e4f e288 50bb 0105  ....n....O..P...
	0x0090:  b9d8 ae68 0343 4ed8 977f 30ca eda6 3337  ...h.CN...0...37
	0x00a0:  7b47 7059 4e0e 6860 7656 ba55 dfd9 a295  {GpYN.h`vV.U....
	0x00b0:  6dd9 83ae 7750 864e 74f9 d3ef 42e3 f0a6  m...wP.Nt...B...
	0x00c0:  044c 8636 e78a 5602 4e73 c1a0 6396 7b88  .L.6..V.Ns..c.{.
	0x00d0:  cf0e c28b 20b9 38d4 ae14 72d9 fad0 064e  ......8...r....N
	0x00e0:  ab3c d770 3438 5c33 039c 7341 264a 82ab  .<.p48\3..sA&J..
	0x00f0:  b5a8 196e 1861 64f9 8e0d 002b d867 ee4c  ...n.ad....+.g.L
	0x0100:  3f1a 622b 35ac 89d8 1fa1 a404 4bd6 802d  ?.b+5.......K..-
	0x0110:  5a67 cee3 8a06 8f4a f055 aead 753a 2a48  Zg.....J.U..u:*H
	0x0120:  eb6e 3bb5 2b17 73d6 1176 205f 4a64 0b40  .n;.+.s..v._Jd.@
	0x0130:  1f1f c374 c9c3 7229 0898 dba5 caee 84fc  ...t..r)........
	0x0140:  de94                                     ..
21:09:39.423687 IP6 2400:7660:ffff:ffff::1.5683 > 2400:7660:ffff:28:f43:d53f:b67b:cdb7.35860: UDP, length 25
	0x0000:  6004 8e56 0021 113f 2400 7660 ffff ffff  `..V.!.?$.v`....
	0x0010:  0000 0000 0000 0001 2400 7660 ffff 0028  ........$.v`...(
	0x0020:  0f43 d53f b67b cdb7 1633 8c14 0021 b1cc  .C.?.{...3...!..
	0x0030:  4801 c678 5802 246a 84ce f0ec b532 3632  H..xX.$j.....262
	0x0040:  3432 0130 622d 1661 74                   42.0b-.at
21:09:39.549546 IP6 2400:7660:ffff:28:f43:d53f:b67b:cdb7.35860 > 2400:7660:ffff:ffff::1.5683: UDP, length 274
	0x0000:  6000 0000 011a 1140 2400 7660 ffff 0028  `......@$.v`...(
	0x0010:  0f43 d53f b67b cdb7 2400 7660 ffff ffff  .C.?.{..$.v`....
	0x0020:  0000 0000 0000 0001 8c14 1633 011a 99d3  ...........3....
	0x0030:  6845 c678 5802 246a 84ce f0ec c22d 16b1  hE.xX.$j.....-..
	0x0040:  0cff c200 00a1 d001 1c94 ffd8 ffe0 0010  ................
	0x0050:  4a46 4946 0001 0100 0001 0001 0000 ffdb  JFIF............
	0x0060:  0084 000a 0707 0807 060a 0808 080b 0a0a  ................
	0x0070:  0b0e 1810 0e0d 0d0e 1d15 1611 1823 1f25  .............#.%
	0x0080:  2422 1f22 2126 2b37 2f26 2934 2921 2230  $"."!&+7/&)4)!"0
	0x0090:  4131 3439 3b3e 3e3e 252e 4449 433c 4837  A149;>>>%.DIC<H7
	0x00a0:  3d3e 3b01 0a0b 0b0e 0d0e 1c10 101c 3b28  =>;...........;(
	0x00b0:  2228 3b3b 3b3b 3b3b 3b3b 3b3b 3b3b 3b3b  "(;;;;;;;;;;;;;;
	0x00c0:  3b3b 3b3b 3b3b 3b3b 3b3b 3b3b 3b3b 3b3b  ;;;;;;;;;;;;;;;;
	0x00d0:  3b3b 3b3b 3b3b 3b3b 3b3b 3b3b 3b3b 3b3b  ;;;;;;;;;;;;;;;;
	0x00e0:  3b3b 3b3b ffc0 0011 0800 e000 e003 0111  ;;;;............
	0x00f0:  0002 1101 0311 01ff c401 a200 0001 0501  ................
	0x0100:  0101 0101 0100 0000 0000 0000 0001 0203  ................
	0x0110:  0405 0607 0809 0a0b 1000 0201 0303 0204  ................
	0x0120:  0305 0504 0400 0001 7d01 0203 0004 1105  ........}.......
	0x0130:  1221 3141 0613 5161 0722 7114 3281 91a1  .!1A..Qa."q.2...
	0x0140:  0823                                     .#
```

Testing on NordicSemi nRF52840 chip.
This appears to be an intermittent issue, most of the time data is retrieved correctly.
Using Leshan Server-Demo Version 2.0.0-SNAPSHOT
SDK: NCS version 2.4.1
- Zephyr up to commit: 4bbd91a9083a588002d4397577863e0c54ba7038

rlubos, 2023-12-19:
I'm sort of confused here, as if you indeed use NCS version 2.4.1, that revision had no generic support for block transfer in LwM2M library. Can you check the latest main? There were some heavy development regarding block transfer support since the revision you mention, including https://github.com/zephyrproject-rtos/zephyr/pull/54761, https://github.com/zephyrproject-rtos/zephyr/pull/61858 and corresponding bugfixes.

marco85mars20, 2023-12-21:
@rlubos apologies for the confusion, we have been cherry-picking commits related to the block-wise transfer feature, also from the 2 PRs you have mentioned.
Everything works as expected most of the time, but sometimes we have noticed glitches, and this is one of them, that has been captured only once and by chance while doing some investigations.
We will be performing further testing with the latest main and report back in case we spot related issues.

rlubos, 2024-01-31:
@marco85mars20 Any updates?

marco85mars20, 2024-02-01:
A few weeks ago we switched to NCS `v3.4.99-ncs1-1` (aka v4.5.1), and we haven't observed this issue yet, but as well we have not done proper testing yet. I believe we can close this issue for the moment. I will open a new one in case needed.

closed 2024-02-01 by marco85mars20