zephyrproject-rtos/zephyr #66639: IPv6 handle_mld_query net_pkt: Uncontiguous data cannot be linearized

KenMacD, 2023-12-18:
**Describe the bug**

When processing IPv6 packets an error is logged every time a ICMPv6/Multicast Listener Query (130) packet is received.

```
net_pkt: Uncontiguous data cannot be linearized
```

A slightly cleaned up stack from gdb is:

```
#0  net_pkt_get_data (pkt=pkt@entry=0x20022110 <_k_mem_slab_buf_rx_pkts+864>, access=access@entry=0x2001a348 <rx_stack+1272>) at /subsys/net/ip/net_pkt.c:2091
#1  0x0006b10c in handle_mld_query (pkt=0x20022110 <_k_mem_slab_buf_rx_pkts+864>, ip_hdr=0x2002024e <net_buf_data_rx_bufs+142>, icmp_hdr=0x2002027e <net_buf_data_rx_bufs+190>) at /subsys/net/ip/ipv6_mld.c:322
#2  0x0001ae34 in net_icmpv6_input (pkt=pkt@entry=0x20022110 <_k_mem_slab_buf_rx_pkts+864>, ip_hdr=ip_hdr@entry=0x2002024e <net_buf_data_rx_bufs+142>) at /subsys/net/ip/icmpv6.c:472
#3  0x0006a6ea in net_ipv6_input (pkt=pkt@entry=0x20022110 <_k_mem_slab_buf_rx_pkts+864>, is_loopback=is_loopback@entry=false) at /subsys/net/ip/ipv6.c:691
#4  0x0006692e in process_data (pkt=pkt@entry=0x20022110 <_k_mem_slab_buf_rx_pkts+864>, is_loopback=<optimized out>) at /subsys/net/ip/net_core.c:137
#5  0x00066948 in processing_data (pkt=0x20022110 <_k_mem_slab_buf_rx_pkts+864>, is_loopback=is_loopback@entry=false) at /subsys/net/ip/net_core.c:157
#6  0x00066972 in net_rx (iface=<optimized out>, pkt=<optimized out>) at /subsys/net/ip/net_core.c:373
#7  0x00066a84 in net_process_rx_packet (pkt=<optimized out>) at /subsys/net/ip/net_core.c:385
#8  0x00068f2c in tc_rx_handler (fifo=0x20003890 <rx_classes>) at /subsys/net/ip/net_tc.c:250
#9  0x000614e4 in z_thread_entry (entry=0x68f23 <tc_rx_handler>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>) at /lib/os/thread_entry.c:48
```

It seems similar in nature to issue #16034. While this appears in the NCS fork (using commit `2e2523e`), the code appears common with the same `NET_PKT_DATA_ACCESS_CONTIGUOUS_DEFINE(mld_access,` in the non-fork, so I suspect the issue is in Zephyr.

**To Reproduce**

Build using IPv6 and connect to a network where other hosts are sending these packets.

**Impact**

So far just the error message, but I'm hesitant to continue further in working with IPv6 while these packets result in an error.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain NCS+Zephyr
 - Commit 2e2523efe52a7ac89f0567b8798fd857b1e71ae3

jukkar, 2023-12-19:
It looks like your received MLD packet does not fit the allocated space in variable `mld_access` in `handle_mld_query()`.

rlubos, 2023-12-19:
Wouldn't that be the same issue we've addressed with https://github.com/zephyrproject-rtos/zephyr/pull/66532 (MLDv1 queries in the network)?

jukkar, 2023-12-19:
But wasn't that issue because MLDv1 packet is smaller than MLDv2, this case looks like the received bytes is larger than the `sizeof(struct net_icmpv6_mld_query)`.

jukkar, 2023-12-19:
@KenMacD could you verify what data you are receiving, pcap / wireshark output of the packet would be nice to see

rlubos, 2023-12-19:
> But wasn't that issue because MLDv1 packet is smaller than MLDv2, this case looks like the received bytes is larger than the `sizeof(struct net_icmpv6_mld_query)`.

Yes, and that was exactly the issue why we've seen `net_pkt: Uncontiguous data cannot be linearized`. MLDv1 query is smaller than MLDv2 one, so we had (if I recall correctly) 20 bytes left on a packet (which was a valid MLDv1 query), but tried to use MLDv2 access on it (which required 24 bytes of contiguous data) and that generated the error.

jukkar, 2023-12-19:
Ok, indeed. I assumed the reporter had that MLDv1 patch applied but I might have assumed wrong. @KenMacD could you verify that you have `net: ipv6_mld: Silently drop MLDv1 queries` patch applied and/or are you receiving MLDv1 packets when the print happens.

KenMacD, 2023-12-20:
I can look at this a little more closely soon. For now I'll attach a pcap of three packets. The first, the query, causes the issue. The two (and other) reports do not show the same behaviour. I'm not sure if that's because they're ignored elsewhere, or processed correctly. All have a length of `86`.

[mlq.pcapng.gz](https://github.com/zephyrproject-rtos/zephyr/files/13729815/mlq.pcapng.gz)

rlubos, 2023-12-21:
> With the patch applied I no longer see the errors. Hopefully someone can confirm those are MLDv1 packets, as I don't know how they're designated in the bytes.

Yes , those are MLDv1 packets. Query caused an error, as it shares the same ICMP code as MLDv2 query, but has a shorter header. MLD reports are not even processed by Zephyr, hence they did not generate any errors but were already silently dropped.

closed 2023-12-21 by KenMacD