zephyrproject-rtos/zephyr #66640: I can`t run RPMSG-lite in Zephyr 3.3 in imx7d-pico/m4 side.  Zephyr 1.13 Works. Issue in new APM changes in Zephyr

neuberfran, 2023-12-19:
[image]

Contextualizing:

1) I resolved the issues(Pic above) I had immediately when trying to migrate (west init, west update, west export) a project from zephyr 1.13 to zephyr 3.3.

[image]
2) Since then (as shown in the photo above) zephyr.bin (from zephyr3.3) has been defined to start on m4 of imx7d-pico in  U-boot in TCM memory 0X7F80000 and the messages  from main_remote_echo (in M4 UART6-Console is: Waiting for Master) stop at while (!rpmsg_lite_is_link_up(rl_instance)) {
**github project:** https://github.com/neuberfran/diegosueiro/blob/main/rpmsg-lite/zephyr/samples/subsys/ipc/rpmsg_lite/remote_echo/src/main_remote_echo.c#L82

Some things I've already tried to resolve:
1) Changes in Yocto/A7/Hardknott
2) Changes in Zephyr Side, int DTS files, Kconfig files, Soc.c file, Exclusions of things about libmetal and open-amp that could block RPMSG-lite
3) Changes zephyr_base And now I have rl_instance # 0, But issue now in this two lines **528 e 549:**
https://github.com/neuberfran/diegosueiro/blob/main/rpmsg-lite/lib/rpmsg_lite/rpmsg_lite.c#L549

in line 529 rpmsg_lite_dev and non-zero

But, in line 548 rpmsg_lite_dev->link_state is always equal to zero (and return 0 to function (!rpmsg_lite_is_link_up(rl_instance)). Why?

How to resolve my issue(Pls)?

[image]

MichalPrincNXP, 2023-12-19:
Hi @neuberfran 
if I understand it correctly, you are attempting to run the older rpmsg-lite v2.0.0 based example with the latest Zephyr version, correct? What is the boot order, is the Linux starting before the Zephyr or vice versa? When the Zephyr app stucks in rpmsg_lite_is_link_up(), I would check if all MU (Messaging Unit) interrupts are reached on the Zephyr side, have you tried to place a breakpoint into the MU isr?
We do not have the imx7d-pico board to try but we could take another NXP board with CortexA and CortexM cores and could get some rpmsg_lite example running. This will take some time, expecting some outputs in first weeks of 2024, does it make sense to you and is it acceptable?
Regards
Michal

neuberfran, 2023-12-19:
@MichalPrincNXP Yes, I stay trying older rpmsg-lite based example with the latest Zephyr version. I don't know if older rpmsg-lite version is 2.0.0.

The boot order is the traditional one in these cases. U-boot starts zephyr.bin (some messages appear on the m4 side and stop at that damn moment)

Also, I looked at the issue below 1 month ago and tried not to upload the tty module directly to  scorpion.conf but after booting Linux/A7/Yocto Hardknott using modprobe imx_rpmsg_tty. But not solved.
https://github.com/EmbeddedRPC/erpc-imx-demos/issues/2#issuecomment-375855176

You said:
I would check if all MU (Messaging Unit) interrupts are reached on the Zephyr side, have you tried to place a breakpoint into the MU isr?
Could you give more details on how to do this?

Today I will try this here:
https://docs.zephyrproject.org/latest/samples/drivers/ipm/ipm_imx/README.html#ipm-imx
https://review.coreboot.org/plugins/gitiles/zephyr-cros/+/7ca69b385de7896fc83ffe86723290c9a4417416%5E%21/

How unfortunate that you don't have this plate yet. Would anything else with A7 and M4 produce the same results?

Yes, I can wait a little longer because I still want to know what is stopping rpmsg-lite here.

Could you tell me something about lines 528 and 549 that I asked above?

E.g.: I stay trying rpmsg-lite (old and new version) in FreeRTOS here. With the new RPMSG-lite version I will can use something like this: rpmsg_lite_wait_for_link_up(my_rpmsg, RL_BLOCK);
But I have a compilation error

With Old rpmsg-lite version in FreeRTOS, stay stopping in rpmsg_lite_is_link_up() like here.

root@scorpion:~# **ls /sys/bus/virtio/devices/**
virtio0
root@scorpion:~# **ls /sys/bus/rpmsg/devices**
virtio0.rpmsg_ctrl.0.0	virtio0.rpmsg_ns.53.53
root@scorpion:~# **dmesg | grep -E "MU| rpmsg"**
[    0.635986] imx rpmsg driver is registered.
[    0.637319] MU is ready for cross core communication!
[    0.699771] virtio_rpmsg_bus virtio0: rpmsg host is online
[    0.716669] hw perfevents: enabled with armv7_cortex_a7 PMU driver, 5 counters available
root@scorpion:~# uname -a
Linux scorpion 5.10.72-2.2.0-next+g000eb85263cb #1 SMP PREEMPT Thu Jan 12 09:09:37 UTC 2023 armv7l armv7l armv7l GNU/Linux