zephyrproject-rtos/zephyr #66646: Best Master Clock identification in gPTP using multiple ports

mschappa, 2023-12-19:
The following problem occurred while running PTP on a multi-port target (RENESAS RZT2M) with two ports enabled but it shoul dalso happen with other Multi-Port devices. BMCA was configured in a way that the device (Dev-B) acts as bridge with a master port (Port2) and a slave port (Port1). Slave port connects to the GM (Dev-A). Master port connects to another device (Dev-C). The devices BMCA algorithms  have been configured using the according priorities:
 - Dev-A: 100
 - Dev-B: 150
 - Dev-C: 200

 That should result in the following PTP-Tree:
<pre>
 ┌─────────┐                ┌─────────┐              ┌─────────┐
 │         │                │         │              │         │
 │  DEV-A  ├────────────────►  DEV-B  ├──────────────►  DEV-C  │
 │         │                │         │              │         │
 └─────────┘                └─────────┘              └─────────┘
</pre>
However, Dev-C turns into a GM as well. We figured out that the announce messages sent from Dev-B inside contain wrong information. It includes information from its own configuration instead of forwarding the information about the GM. In that case Dec-C assumes having the better clock and therefore turns into a master as well. This results in a network topology having two GM.

The expected behavior is that Dev-B forwards the clock information from Dev-A via  announce messages on its other ports. 

Whil eanalyzing the problem we made some changes to *gptp_messages.c* as shown below what seems to solve the problem.

```diff
------------------------- gptp_messages.c -------------------------
index 06b9cb0..f6cf1c9 100644
@@ -502,6 +502,7 @@ struct net_pkt *gptp_prepare_announce(int port)
 	struct net_if *iface;
 	struct net_pkt *pkt;
 	struct gptp_hdr *hdr;
+	struct gptp_priority_vector *gm_prio;
 
 	NET_ASSERT((port >= GPTP_PORT_START) && (port <= GPTP_PORT_END));
 	global_ds = GPTP_GLOBAL_DS();
@@ -551,25 +552,12 @@ struct net_pkt *gptp_prepare_announce(int port)
 	ann->cur_utc_offset = htons(global_ds->current_utc_offset);
 	ann->time_source = global_ds->time_source;
 
+	gm_prio = &global_ds->gm_priority;
 	switch (GPTP_PORT_BMCA_DATA(port)->info_is) {
 	case GPTP_INFO_IS_MINE:
-		ann->root_system_id.grand_master_prio1 = default_ds->priority1;
-		ann->root_system_id.grand_master_prio2 = default_ds->priority2;
-
-		ann->root_system_id.clk_quality.clock_accuracy =
-			default_ds->clk_quality.clock_accuracy;
-		ann->root_system_id.clk_quality.clock_class = default_ds->clk_quality.clock_class;
-		ann->root_system_id.clk_quality.offset_scaled_log_var =
-			htons(default_ds->clk_quality.offset_scaled_log_var);
-
-		memcpy(&ann->root_system_id.grand_master_id,
-		       default_ds->clk_id,
-		       GPTP_CLOCK_ID_LEN);
-		break;
 	case GPTP_INFO_IS_RECEIVED:
 		memcpy(&ann->root_system_id,
-		       &GPTP_PORT_BMCA_DATA(port)->
-				master_priority.root_system_id,
+		       &gm_prio->root_system_id,
 		       sizeof(struct gptp_root_system_identity));
 		break;
 	default:
```

jukkar, 2023-12-19:
Thanks for the report. Could you send a pr as you have a fix proposal?

mschappa, 2023-12-20:
Of course: #66706

PR #66706 (2024-01-15): net:gptp: Always use GM PRIO root system id for announce messages
Fixes #66646
closed 2024-01-15 by carlescufi