zephyrproject-rtos/zephyr #66682: dts: unable to access the interrupt controller in `interrupts-extended`

ycsin, 2023-12-20:
**Is your enhancement proposal related to a problem? Please describe.**
As things stand today, there is no way for a driver to know it's parent interrupt controller if it is specified in the `interrupts-extended` instead of the `interrupt-parent` property.

For example, we can get the interrupt controller of a `uart` node can easily:

https://github.com/zephyrproject-rtos/zephyr/blob/6e678e3baee0c57fcf9a96e47b79f8b9a9e6eeaf/dts/riscv/virt.dtsi#L27-L34

with:

```c
DT_PROP(DT_NODELABEL(uart0), interrupt_parent)
```

but there's no way to get the interrupt controller for the following node:

https://github.com/zephyrproject-rtos/zephyr/blob/6e678e3baee0c57fcf9a96e47b79f8b9a9e6eeaf/dts/riscv/virt.dtsi#L165-L183

Even though the devicetree has all the info about the hardware, the application has no way to tap to those info. Having such ability would be able to simplify the generation of the lookup table for multi-level interrupt controllers, see [`multilevel_irq.c`](https://github.com/zephyrproject-rtos/zephyr/blob/main/arch/common/multilevel_irq.c).

**Describe the solution you'd like**
Output the interrupt controller node(s) in the `interrupts-extended` properties, alongside with the `interrupt number` and `priority` if available.

**Additional context**
- [Docs: Specifying interrupt information for devices](https://www.kernel.org/doc/Documentation/devicetree/bindings/interrupt-controller/interrupts.txt)
- https://github.com/zephyrproject-rtos/zephyr/pull/65288#issuecomment-1855412027

nordicjm, 2023-12-20:
Isn't this related to what @bjarki-trackunit explained on this weeks architecture working group call?

nordicjm, 2023-12-20:
> > Isn't this related to what @bjarki-trackunit explained on this weeks architecture working group call?
> 
> Not really sure what's discussed in the meeting, is there a pointer to that?

https://github.com/zephyrproject-rtos/zephyr/pull/66505

bjarki-trackunit, 2023-12-20:
Hi, this is in fact part of that PR, there are a few commits which update gen_defines.py and add the required macros to get the interrupt controller as specified by the devicetree section 2.4.1

See the additionans to devicetree.h (DT_INTC(nie_id) etc.) :)

ycsin, 2023-12-20:
> Hi, this is in fact part of that PR, there are a few commits which update gen_defines.py and add the required macros to get the interrupt controller as specified by the devicetree section 2.4.1
> 
> See the additions to devicetree.h (DT_INTC(node_id) etc.) :)

Just had a look at the (updated) RFC, seems like I could have sit back and relax!
In any case I think changes to the `gen_defines.py` and `devicetree.h` can be spin-off into a separate PR

bjarki-trackunit, 2023-12-20:
> Just had a look at the (updated) RFC, seems like I could have sit back and relax!
> In any case I think changes to the `gen_defines.py` and `devicetree.h` can be spin-off into a separate PR

agreed, I will split it into its own PR shortly :)

PR #66707 (2024-01-17): Add devicetree macros to get node's interrupt controller
This PR updates `gen_defines.py` to generate defines for nodes interrupt controllers, then updates `devicetree.h` with macros (`DT_INTC*`) which get the interrupt controller node.

The PR also adds tests for the new macros, and updates the dts documentation to include the new generated defines.

fixes: #66682
closed 2024-01-17 by carlescufi