zephyrproject-rtos/zephyr #66691: tests/bsim/bluetooth/mesh/tests_scripts/priv_beacon/proxy_adv_multi_subnet_coex.sh fails on main

aescolar, 2023-12-20:
**Describe the bug**
tests/bsim/bluetooth/mesh/tests_scripts/priv_beacon/proxy_adv_multi_subnet_coex.sh is failing on main

**To Reproduce**
Steps to reproduce the behavior:
1. Trigger this test in CI with a related PR
OR in Zephyr's root folder
1. nice tests/bsim/bluetooth/mesh/compile.sh
2. tests/bsim/bluetooth/mesh/tests_scripts/priv_beacon/proxy_adv_multi_subnet_coex.sh
3. See error

**Expected behavior**
No test failures in CI

**Impact**
Block development in main when the test is triggered

**Logs and console output**
https://github.com/zephyrproject-rtos/zephyr/actions/runs/7273217875/job/19816778491?pr=66667#step:12:498

**Environment (please complete the following information):**
 - OS: Linux, CI, or Ubuntu 22.04
 - Toolchain (e.g Zephyr SDK, ...): host gcc 11 or 12
 - Commit SHA or Version used: Today's main 6e678e3baee0c57fcf9a96e47b79f8b9a9e6eeaf

**Additional context**
Test introduced in https://github.com/zephyrproject-rtos/zephyr/pull/65914

aescolar, 2023-12-20:
CC: @PavelVPV @alxelax @Andrewpini

alxelax, 2023-12-20:
fix: https://github.com/zephyrproject-rtos/zephyr/pull/66689

closed 2023-12-20 by aescolar