zephyrproject-rtos/zephyr #66694: `dt-binding` files are not correctly parsed depending on the `compatible` list order

dgastonochoa, 2023-12-20:
**Description**

It seems `dt-binding` files are not detected by the device tree parser if their associated `compatible` appears not in first place in the list of `compatible`s of a node.  

**To Reproduce**
 - In the file `st,stm32-spi.yaml`, add a new property of any type. Let it be `my-property`. E.g.:

```
properties:
  my-property:
    required: true
    type: int
```

 - In the file `stm32h7.dtsi`, go to the `spi1` node and add to it the property above.
 - Build the spi_loopback tests: `west build -p auto -b nucleo_h753zi tests/drivers/spi/spi_loopback`. Verify this produces an error.
 - In the file `stm32h7.dtsi`, move `"st,stm32-spi"`, in the `compatible` list of `spi1`, from the last position to the first one.
 - Build again as above and verify no errors occur now.

**Expected behavior**
`dt-binding` files are detected correctly by the device tree parser no matter the position their respective `compatible` occupies in the list of `compatible`s of a node.

**Impact**
The impact is potentially big, as it could happen that 2 `dt-binding` files are modified and there is no way to put 2 in first place. 

**Logs and console output**
The error is:

`devicetree error: 'my-property' appears in /soc/spi@40013000 in [...]/zephyr/build/zephyr/zephyr.dts.pre, but is not declared in 'properties:' in [...]/zephyr/dts/bindings/spi/st,stm32h7-spi.yaml`

henrikbrixandersen, 2023-12-20:
This is actually well-described here: https://docs.zephyrproject.org/latest/build/dts/bindings-intro.html#other-ways-nodes-are-matched-to-bindings:

> Take this node as an example:
> 
> baz-device {
>      compatible = "foo-company,baz-device", "generic-baz-device";
> };
> 
> The baz-device node would get matched to a binding with a compatible: "generic-baz-device" line if the build system can’t find a binding with a compatible: "foo-company,baz-device" line.

In your example above, the compatible is set to `"st,stm32h7-spi", "st,stm32-spi-fifo", "st,stm32-spi"`, so the build system uses the first found compatible, namely `st,stm32h7-spi`. The `st,stm32-spi` binding is thus never evaluated _unless_ you change the order of compatibles, as you also mention.

dgastonochoa, 2023-12-20:
@henrikbrixandersen So this is the expected behavior? It is not a bug?

henrikbrixandersen, 2023-12-20:
> @henrikbrixandersen So this is the expected behavior? It is not a bug?

Correct.

closed 2023-12-20 by dgastonochoa