zephyrproject-rtos/zephyr #66697: RFC: net: Zephyr network services support with offloaded interfaces

rlubos, 2023-12-20:
## Introduction

Many TCP/UDP based services in Zephyr are implemented using `net_context` or even lower level APIs (DHCP clients, DNS resolver/responder, telnet shell backend etc.). The advantage of this approach over using sockets API is generally a lower memory footprint (no need for a thread to run the service, no need to allocate FD and sometimes `struct net_context`). However this comes at a cost - those services are not usable with offloaded interfaces.

### Problem description

Over years, this became an issue for several people, which resulted in enhancement requests, to rewrite certain parts of the networking stack:
* https://github.com/zephyrproject-rtos/zephyr/issues/15165
* https://github.com/zephyrproject-rtos/zephyr/issues/25858
* https://github.com/zephyrproject-rtos/zephyr/issues/29438

This also brings an architectual questions for new modules (like this [DHCP server feature request](https://github.com/zephyrproject-rtos/zephyr/issues/38284)). The feature request does not make it exlicit, but since most of the Wi-Fi solutions we have in Zephyr are offloaded, it seems pretty obvious that the requested DHCP server implementation should also work with offloaded interfaces.

## Detailed RFC

Before proposing any changes, I think a decission has to be made how to tackle this, hence opening this RFC to gather opinions. I feasibly see only two options here:
* Keep the implementations of the aforementioned services as they are, to keep them tailored for the native stack. This effectively means that offloaded interfaces won't have access to all network services in Zephyr. We shall then close all of the above feature requests as "won't fix". 

* Or rework aforementioned (and any future) network services to use socket API instead. This will come at a cost of increased memory footprint for those services (sorry I don't have any numbers at this point yet), affecting all native interfaces. But in return, we'd get better support for offloaded interfaces. The overall memory increment could possibly be mitigated by some wise design choices (like for example, instead of having each service to run in a separate thread, we could introduce `inetd`-like service with a single thread to monitor sockets from various network services).

So, to summarize, we need to decide, whether we agree for increased memory footprint for all types of interfaces, in favour of improved support for the offloaded ones, or whether to keep the status quo of the offloaded interfaces, agreeing they don't get accesss to all of the services avaialble for native ones. Any opinions on the topic are welcome.

rlubos, 2023-12-20:
CC @jukkar @carlescufi @pdgendt @tbursztyka @ssharks @rerickson1
Feel free to add anyone else that might have an opinion on this.

pdgendt, 2023-12-20:
At first thought I would prefer the latter option to rework the network services, as I suppose the socket API is also more in line with the linux world, which is an added bonus IMO.

rerickson1, 2023-12-20:
My vote is to re-work and use socket APIs. I don't have any hard numbers but it feels like we have more offloaded networking devices than native. Leaving out features for offloaded devices is a big limitation for IoT products.

tbursztyka, 2024-02-09:
I would also go for a re-work  towards using socket API.
Remember that dhcp client, telnet etc... were made at a time when socket API was not made or fully ready. Now that it is, let's use it exclusively.

This will obviously help support offloaded net stack enabled devices, but it will also help tweak&optimize the core api of the native net stack without impeding any users (services or else). I believe we can reach a point were using socket API and FDs will not bring much overhead, if at all.

jukkar, 2024-02-09:
I think we can slowly migrate the code use socket API, I just recently submitted a PR that converted the syslog exporter to use socket API. Further improvements in the other parts of the network stack will hopefully come in the future time permitting. The socket service API can help the conversion as it allows using socket API without creating a separate handler thread.