zephyrproject-rtos/zephyr #66701: doc: Documentation issue in 'guides/kconfig/tips' - broken hyperlink from kconfiglib to zephyrproject

ngenetzky-appareo, 2023-12-20:
**Describe the bug**

Broke hyperlink from https://pypi.org/project/kconfiglib/ to [zephyrproject](https://docs.zephyrproject.org/latest/guides/kconfig/tips.html).

Here is the text with the hyperlink:

> See [this page](https://docs.zephyrproject.org/latest/guides/kconfig/tips.html) for some Kconfig tips and best practices.

**Environment**

* Page: https://docs.zephyrproject.org/latest/guides/kconfig/tips.html
* Version: main
* SHA-1: 8c361404085f9ba407cce731076cbcc33ae86213

CC: @ulfalizer

kartben, 2023-12-20:
Hi @ngenetzky-appareo -- thanks for the report! It's not clear to me if there is anything to track on the Zephyr side of things? The "Kconfig - Tips and Best Practices" page doesn't have any broken links, correct? It looks as it's rather some broken links in the upstream Kconfiglib docs and published pypi package?

ngenetzky-appareo, 2023-12-20:
If this information has been moved and you know where that would be helpful
to add here or in the linked ticket.

I am not very familiar with the documentation generator you use, but it's
possibly that an "alias" or "redirect" could be added to fix the issue.

Zephyr has a much larger contributor base and higher quality documentation
than the other project. It is also possible that other projects have also
linked to zephyr's great documentation.

kartben, 2023-12-20:
Oh, I see! It used to be at a different URL, gotcha. Yes, we have a way to have redirects. Will make a PR :)

PR #66719 (2023-12-21): doc: kconfig: redirect: Add redirects for some old URLs
Documentation pages around Kconfig have moved around on a couple of occasions in the past [1] [2] and current redirects were dysfunctional (leading /) and incomplete, while an entire set of redirects was also missing. This notably fixes the "browse latest development version of this page" link on
https://docs.zephyrproject.org/2.7.5/guides/build/kconfig/tips.html or a link to the "Kconfig tips" page as found in Kconfiglib's current README (https://pypi.org/project/kconfiglib/)

[1] commit 5342bc64ddc2bb733bb08041b6029cb9299266eb
[2] commi [... cut]
commit 109c53da (2023-12-21): doc: kconfig: redirect: Add redirects for some old URLs

Documentation pages around Kconfig have moved around on a couple of
occasions in the past [1] [2] and current redirects were dysfonctional
(leading /) and incomplete, while an entire set of redirects was also
missing. This notably fixes the "browse latest development version of
this page" link on
https://docs.zephyrproject.org/2.7.5/guides/build/kconfig/tips.html or a
link to the "Kconfig tips" page as found in Kconfiglib's current README
(https://pypi.org/project/kconfiglib/)

[1] commit 5342bc64ddc2bb733bb08041b6029cb9299266eb [2] comm [... cut]
closed 2023-12-21 by fabiobaltieri