zephyrproject-rtos/zephyr #66710: tests/kernel/cache should not run  on stm32h7 or stm32f7 targets

FRASTM, 2023-12-20:
**Describe the bug**
Since  PR https://github.com/zephyrproject-rtos/zephyr/pull/65828
the tescase tests/kernel/cache/  is executed on stm32h7 and stm32h7 target boards and should probably not  (was not before).

The tests/kernel/cache/prj.conf has CONFIG_CACHE_MANAGEMENT=y and the tests/kernel/cache/sample.yaml is filtering on CONFIG_CACHE_MANAGEMENT
--> that will automatically set the CONFIG_CACHE_MANAGEMENT for the stm32h7 and stm32f7 targets
and resulting in a test failure 

Please also mention any information which could help others to understand
the problem you're facing:
   specific commit a2be813d99a4c2a9ba51d14810d2ee9bb635db9e
 

**To Reproduce**
Steps to reproduce the behavior:
1. west build  -p always -b stm32h747i_disco_m7 tests/kernel/cache/
2. west flash
3. See error

**Expected behavior**
Testcase not executed

**Impact**
CI failure 

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.5.0-3042-g8e5c3ce61646 ***
Running TESTSUITE cache_api
===================================================================
START - test_data_cache_api

```
--> endless loop

**Environment (please complete the following information):**
 - OS: ubuntu
 - Toolchain (e.g Zephyr SDK, ...) 0.16.3
 - Commit SHA or Version used  eb2cd314075bb3b28581c72bb777eee7ec5fd4bc

**Additional context**
building tests/kernel/cache/ for the stm32h7/stm32f7 target gives:
```
#define CONFIG_ICACHE_LINE_SIZE 32
#define CONFIG_DCACHE_LINE_SIZE 32
#define CONFIG_ARCH_HAS_NOCACHE_MEMORY_SUPPORT 1
#define CONFIG_CPU_HAS_DCACHE 1
#define CONFIG_CPU_HAS_ICACHE 1
#define CONFIG_DCACHE 1
#define CONFIG_ICACHE 1
#define CONFIG_CACHE_MANAGEMENT 1
#define CONFIG_ARCH_CACHE 1
#define CONFIG_CACHE 1
```

FRASTM, 2023-12-20:
Looks more or less like issue  https://github.com/zephyrproject-rtos/zephyr/issues/65863

FRASTM, 2024-01-04:
my understanding is that CONFIG_CACHE_MANAGEMENT is _=y_ by the https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/kernel/cache/prj.conf  but it _should not_ for arm M7

FRASTM, 2024-02-09:
One proposal:  to invalidate the L1 D cache before cleaning and invalidating it

In the ../modules/hal/cmsis/CMSIS/Core/Include/cachel1_armv7.h

```
diff --git a/CMSIS/Core/Include/cachel1_armv7.h b/CMSIS/Core/Include/cachel1_armv7.h
index abebc95..de504a3 100644
--- a/CMSIS/Core/Include/cachel1_armv7.h
+++ b/CMSIS/Core/Include/cachel1_armv7.h
@@ -226,6 +226,9 @@ __STATIC_FORCEINLINE void SCB_InvalidateDCache (void)
     SCB->CSSELR = 0U;                       /* select Level 1 data cache */
     __DSB();
 
+    SCB->CCR &= ~(uint32_t)SCB_CCR_DC_Msk;  /* disable Level 1 data cache */
+    __DSB();
+
     ccsidr = SCB->CCSIDR;
 
                                             /* invalidate D-Cache */
```

FRASTM, 2024-02-20:
It appears that the buffer used for the test case
`static ZTEST_BMEM uint8_t user_buffer[SIZE];`
is declared in  the SRAM0 area of the stm32f7 device

WHen this SRAM area is declared as **NOCACHE**: testcase PASSED

```
	sram0: memory@20010000 {
		compatible = "zephyr,memory-region", "mmio-sram";
		reg = <0x20010000 DT_SIZE_K(256)>;
		zephyr,memory-region = "SRAM0";
		zephyr,memory-attr = <( DT_MEM_ARM(ATTR_MPU_RAM_NOCACHE) )>;
	};
```

FRASTM, 2024-02-21:
@tbursztyka can you please give more detail about excluding " _some boards which either have their implementation broken or non-existent_."  What does it  mean

erwango, 2024-02-21:
@hakehuang Do you know if this tests is run and passed on your M7 based targets ?

erwango, 2024-02-23:
@microbuilder ^^
Not STM32 specific, seen also here: https://github.com/zephyrproject-rtos/zephyr/issues/69394

hakehuang, 2024-02-23:
> @hakehuang Do you know if this tests is run and passed on your M7 based targets ?

NO. it fails https://github.com/zephyrproject-rtos/zephyr/issues/69394?notification_referrer_id=NT_kwDOAAw1xrE5NTg1NjI1MDU2OjgwMDE5OA&notifications_query=reason%3Aparticipating#issuecomment-1961620256

FRASTM, 2024-02-29:
being fixed by https://github.com/zephyrproject-rtos/zephyr/pull/69556

erwango, 2024-03-01:
> being fixed by #69556

Merged

closed 2024-03-01 by erwango