zephyrproject-rtos/zephyr #66720: stm32f412cg unexpected power consumption for MCO2

niedzwiecki-dawid, 2023-12-20:
The STM32F412 chips have 2 MCO pins:
[image]

The MCO2 pin can have 4 different source clocks:
[image]

I've noticed that when I change the source from a default one (SYSCLK) to PLLI2S (PLLI2S not enabled) or HSE (not connected), the power consumption drops ~0,5mW.

I'm working with [google_dragonclaw](https://github.com/zephyrproject-rtos/zephyr/tree/main/boards/arm/google_dragonclaw) board, which uses STM32F412CG chip in UFQFPN48 package. This chip version doesn't even have the MCO2 pin (PC9).
[image]

This behavior can't be observed for the MCO1 pin in my case. 

The STM32 HAL doesn't even define bits for MCO2 configuration for [STM32F412cx](https://github.com/zephyrproject-rtos/hal_stm32/blob/main/stm32cube/stm32f4xx/soc/stm32f412cx.h#L8485) boards. I guess it is done this way because none of the STM32F412cx chips don't have the MCO2 pin. This doesn't even allow changing the source with the config `CONFIG_CLOCK_STM32_MCO2_SRC_x`.

Unfortunately, there is no nucleo board with that version of chip to confirm that issue.

Would you be able to somehow confirm that this is expected behavior or it is kind of silicon bug?

erwango, 2023-12-21:
> Would you be able to somehow confirm that this is expected behavior or it is kind of silicon bug?

Nothing is specified in Errata sheet around such behavior. I'd suggesting asking st.community.

niedzwiecki-dawid, 2023-12-21:
Will do. Do you think it is a good approach from HAL side that it doesn't allow changing the MCO2 bits when the PC9 pin is not present?

erwango, 2023-12-21:
@niedzwiecki-dawid To be sure I understand correctly: What are you doing to solve the point ?
Also, IIUC, by default, you don't have any setting on PC9, right ?

niedzwiecki-dawid, 2023-12-21:
I would like to set the MCO2 source clock to HSE, because it saves 0,5mW
[image]

It is not related to MCO1 in any way.

>Also, IIUC, by default, you don't have any setting on PC9, right ?

My chip doesn't have the PC9 pin, so I don't configure it.

erwango, 2023-12-21:
So, from zephyr perspective, what would be the change required ?
I guess you're able to do a: CONFIG_CLOCK_STM32_MCO2_SRC_HSE=y, right ?

Or maybe this doesn't compile because of missing HAL defines ?

niedzwiecki-dawid, 2023-12-21:
>Or maybe this doesn't compile because of missing HAL defines ?
Yes, the MCO2 bits are [not defined](https://github.com/zephyrproject-rtos/hal_stm32/blob/main/stm32cube/stm32f4xx/soc/stm32f412cx.h#L8485).

niedzwiecki-dawid, 2023-12-21:
Ok, I will.

One side question, also in the clock area. Is there any way to set the PLLR division factor for STM32F412 chips? I've notice that increasing the division factor also reduces power consumption, which of course make sense and have confirmation in an application note regarding power consumption.

[image]

closed 2023-12-21 by niedzwiecki-dawid

erwango, 2023-12-21:
> One side question, also in the clock area. Is there any way to set the PLLR division factor for STM32F412 chips? I've notice that increasing the division factor also reduces power consumption, which of course make sense and have confirmation in an application note regarding power consumption.

It isn't indeed. What you'd need to do is to add the matching prop in `dts/bindings/clock/st,stm32f4-pll-clock.yaml`, and update `include/zephyr/drivers/clock_control/stm32_clock_control.h`. The c driver should already be compatible.
One tricky part might be that this PLL R output might not be available on all F4 series, so you might have to create a f412-pll dedicated binding, similarly to what was done for `dts/bindings/clock/st,stm32f412-plli2s-clock.yaml`

niedzwiecki-dawid, 2023-12-21:
Ok, thanks for the info. I wasn't sure if I've missed something and it can be done without changes.

niedzwiecki-dawid, 2023-12-21:
FYI ST community [post](https://community.st.com/t5/stm32-mcus-products/stm32412cg-ufqfpn48-mco2-unexpected-power-consumption/m-p/622038) regarding MCO2.

niedzwiecki-dawid, 2024-02-28:
@erwango Is there any procedure to cherry-pick https://github.com/STMicroelectronics/cmsis_device_f4/commit/2e1724e592931693ce4dfc43e2fd6b3bc375d6a0 ?

niedzwiecki-dawid, 2024-02-28:
Great, thanks

niedzwiecki-dawid, 2024-05-07:
@erwango The MCO2 [commit](https://github.com/STMicroelectronics/cmsis_device_f4/commit/2e1724e592931693ce4dfc43e2fd6b3bc375d6a0)  has been merged some time ago, but I still can't see it in [hal_stm32](https://github.com/zephyrproject-rtos/hal_stm32/). Is there anything I should do to pull the commit?

erwango, 2024-05-07:
> @erwango The MCO2 [commit](https://github.com/STMicroelectronics/cmsis_device_f4/commit/2e1724e592931693ce4dfc43e2fd6b3bc375d6a0) has been merged some time ago, but I still can't see it in [...]

You can push a PR to update west.yaml.
But this is also on going with https://github.com/zephyrproject-rtos/zephyr/pull/71484

niedzwiecki-dawid, 2024-05-07:
Ok, thanks :-)