zephyrproject-rtos/zephyr #66727: gpio_pin_get_config doesn't return the GPIO_ACTIVE_LOW bit

jeremybettis, 2023-12-20:
If you call for example `gpio_pin_configure(gpio_dev, pin, GPIO_OUTPUT_INACTIVE | GPIO_ACTIVE_LOW)` and then immediately call `gpio_pin_get_config`, the returned value will be `GPIO_OUTPUT_HIGH`.  But I expected to get either `GPIO_OUTPUT_HIGH | GPIO_ACTIVE_LOW` or `GPIO_OUTPUT_INACTIVE | GPIO_ACTIVE_LOW`.

This is a problem if you desire to round trip the config. For example, if you call `gpio_pin_get_config` to save the current config, then `gpio_pin_configure` for a short time, and then put it back to the flags you got from `gpio_pin_get_config`

henrikbrixandersen, 2024-01-02:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

You neglected to include essential information such as the hardware platform used, the Zephyr SHA or version used, and sample code for reproducing the issue.

jeremybettis, 2024-01-02:
I discovered this while writing a unit test. It is not specific to any platform, and it still true at today's HEAD.

henrikbrixandersen, 2024-01-02:
> I discovered this while writing a unit test. It is not specific to any platform, and it still true at today's HEAD.

Well, that unit test must be using a specific GPIO driver. Which GPIO driver is used?

Posting a minimal example would greatly help in understanding this instead of leaving the guesswork up to people reading through this issue.

henrikbrixandersen, 2024-01-16:
Please reopen once more information is provided.

closed 2024-01-16 by henrikbrixandersen