zephyrproject-rtos/zephyr #66728: Syslog logging with RFC5424 support

psztoch, 2023-12-20:
Current syslog logging subsystem support UDP only.
UDP is very bad protocol for logging. Black holes in journal if we have unstable network, what is common after on site power supply cycle (off/on).
Please add support for: meta sequenceId and meta sysUpTime.
This will not improve the completeness of the logs, but it will ensure that there are holes in the log.
More in RFC: https://www.rfc-editor.org/rfc/rfc5424.html#section-7.3

We have defined timeQuality in RFC too. It will be nice to possibility of mark logs with tzKnown and isSynced.
https://www.rfc-editor.org/rfc/rfc5424.html#section-7.1

Support for:
* origin software
* origin swVersion
will be valuable feature too.

nordic-krch, 2024-01-29:
@jukkar as you've added that backend maybe you could comment?

jukkar, 2024-01-30:
Thanks for the info @psztoch. Supporting TCP logging has not been a priority but lets see if I could dedicate some time for it. I will probably also convert the net_context API usage there to BSD sockets at the same time.

jukkar, 2024-01-30:
> UDP is very bad protocol for logging. Black holes in journal if we have unstable network, what is common after on site power supply cycle (off/on).

BTW, using TCP is not perfect either. If there are lot of prints and slow network and/or too few network buffers, we might run of network buffers during send and be forced to drop packets, but that is pretty much business as usual.

jukkar, 2024-01-30:
Syslog over TCP is described here https://www.rfc-editor.org/rfc/rfc6587

jukkar, 2024-01-30:
Initial step here https://github.com/zephyrproject-rtos/zephyr/pull/68301 where the current syslog backend is converted to use BSD socket API

psztoch, 2024-01-31:
> @psztoch please take a look if #68307 would work for you. I did not implement any additional fields to the message at this point, this is just to make TCP work with the syslog.

Support for TCP is big step forward, because zephyr device has TCP confirmation about log delivery.
A temporary network problem will cause the packet to be retransmitted.
The certainty of log completeness will increase by up to 10 times.

Unfortunately, there will always be a grain of uncertainty as to whether we have everything. Didn't the buffer overflow and we lost something...
Adding sequenceId dispels these doubts.

In my opinion, adding additional metadata to syslog and TCP support is equally valuable and necessary.

jukkar, 2024-02-01:
FYI @psztoch, I added structured data support to the latest version of #68307

commit 3c8bd58a (2024-03-12): log: net: Add TCP support to syslog transmitter

Allow user to configure the syslog net backend to use TCP
instead of UDP. The syslog server address for TCP needs to
have "tcp://" URI in front of the address, for example the
tcp://192.0.2.2 server address would use TCP as a transport.
If there is no URI, then UDP would be used by default.

See the relevant RFC at https://www.rfc-editor.org/rfc/rfc6587
for details.

Fixes #66728
closed 2024-03-12 by fabiobaltieri