zephyrproject-rtos/zephyr #66742: twister: tests: Unexpected files left after Twister tests

golowanow, 2023-12-21:
**Describe the bug**

Unexpected files left in the source tree after Twister's  test execution (./scripts/tests/twister and ./scripts/tests/twister_bllackbox)

**To Reproduce**
Steps to reproduce the behavior:
```
$ pytest $ZEPHYR_BASE/scripts/tests/twister
$ git status
```
```
$ pytest $ZEPHYR_BASE/scripts/tests/twister_blackbox
$ git status 
```

**Expected behavior**
no extra files left in the source tree after the test execution

**Impact**
confusion, working commit possible poisoning with unexpected files.

**Logs and console output**

```
$pytest $ZEPHYR_BASE/scripts/tests/twister
...
========== 537 passed, 273 warnings in 30.36s =
$ git status
On branch main
Your branch is up to date with 'upstream/main'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        mock_platform/
        scripts/tests/twister/test_data/demo_board_2/
```
```
$pytest $ZEPHYR_BASE/scripts/tests/twister_blackbox
==== 22 passed, 4 warnings in 186.53s (0:03:06) ==
$ git status 
On branch main
Your branch is up to date with 'upstream/main'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        OUT_DIR/
        TEST_DIR/
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.16.4
 - main branch 8c83185f6e3d5b41e

golowanow, 2023-12-21:
@LukaszMrugala, one more observation in regard of `scripts/tests/twister_blackbox` execution: 
it creates quite a few (62) ./twister-out.* directories which also remains, what's the value to always keep these ?

LukaszMrugala, 2023-12-22:
> @LukaszMrugala, one more observation in regard of `scripts/tests/twister_blackbox` execution: it creates quite a few (62) ./twister-out.* directories which also remains, what's the value to always keep these ?

That's true. `twister-out` folders were deemed as acceptable to leave, as they hold the exact results of every test. Their removal by the user is also easy (`rm -rf ./twister-out*`), so I deemed it safe to leave.

On the other hand, as the number of `twister-out`s rises, their usefulness wanes. Twister only allows for 100 such folders to exist, so in a short order they won't hold the detailed results anymore.

I will look at some clearing fixture.

golowanow, 2023-12-22:
just consider someone who prepares PR already having few own twister-out directories, so it becomes a problem to decide what to delete manually.

> I will look at some clearing fixture.

it might be a pytest parameter to keep logs (with default - don't keep)

PR #66752 (2024-01-10): scripts: tests: Removal of straggling folders
Current blackbox tests leave two folders, `OUT_DIR` and `TEST_DIR` after they are finished.
This change deletes them appropriately.

Fixes #66742
closed 2024-01-10 by nashif