zephyrproject-rtos/zephyr #66761: Posix threads: 16-bit stack size limit

Frauschi, 2023-12-21:
Hi all,

I just saw that with #66511 merged yesterday, the stack size of a POSIX thread is now caped at 64 kB (uint16 max). We currently run multiple Zephyr applications with threads that need a stack size larger than that (roughly 128 kB due to an external library). As our target platforms have plenty of RAM available, we see no problem with larger stack sizes.

As #66511 is already merged, I opened this issue. @cfriedt mentioned in the discussion that he intends to add an option for larger stacks. Is this change intended for the near future? And is help required for the change?

Kind regards,
Tobias

cfriedt, 2023-12-21:
> I just saw that with #66511 merged yesterday, the stack size of a POSIX thread is now caped at 64 kB (uint16 max). We currently run multiple Zephyr applications with threads that need a stack size [...]

Aha!

I knew three would be at least 1 user 😅

Would be great to get the right reviewers on POSIX PRs because ATM, I'm just hand-picking people.

Typically there are zero reviewers automatically populated on many POSIX PRs, which seems wrong. @nashif, is there a better way to do that? Do reviewers mainly get populated from collaborators, or previous editors of a file? Collaborators welcome (ahem, @ycsin).

> As #66511 is already merged, I opened this issue. @cfriedt mentioned in the discussion that he intends to add an option for larger stacks. Is this change intended for the near future? And is help required for the change?

Would be happy to add a new option for larger size stacks. The structure itself would not change, but the stack size would be implicitly shifted left. This would coincide roughly with how `PAGESIZE` can be used. It might not be defined immediately as such, but it will be configurable in any case for pthread stacks.

To keep things small, I would probably default a page to 64-256 bytes (so 6-8 bits). That would effectively extend the maximum thread stack size up to 4M-16M (22-24 bits), for example rounded up by granularity of `PAGESIZE`.

Does that work for you @Frauschi ?

Frauschi, 2023-12-21:
> That would effectively extend the maximum thread stack size up to 4M-16M (22-24 bits), for example rounded up by granularity of `PAGESIZE`.
> 
> Does that work for you @Frauschi ?

This would definitely work for our use cases, yes. 

> The structure itself would not change, but the stack size would be implicitly shifted left. This would coincide roughly with how `PAGESIZE` can be used. It might not be defined immediately as such, [...]

Just out of curiosity, is all this effort really necessary/worth it „just“ to save the 2 bytes in the `pthread_attr_t` structure instead of keeping the stack size a `uint32_t`? Or do I miss something here?

cfriedt, 2023-12-22:
> Just out of curiosity, is all this effort really necessary/worth it „just“ to save the 2 bytes in the `pthread_attr_t` structure instead of keeping the stack size a `uint32_t`? Or do I miss something here?

@Frauschi - Maybe? Part of Zephyr's philosophy is to keep things as small as possible / practical. And there is no sense in wasting stack space on (typical) devices where such space is precious.

Also, I believe by optimizing for space we're saving a full 4 bytes by using a single 32-bit field for both stacksize and guardsize which would have otherwise been mostly unused.

In total, I think the structure is still 16 bytes (2 words) on 64-bit architectures and 12 bytes (3 words) on 32-bit architectures, compared to 48 and 36 previously, so a pretty significant difference, IMHO.

Frauschi, 2023-12-22:
Thanks for the thorough explanation, sounds reasonable. 😄

Also thanks for the quick fix in #66901, looks promising. Will test it tomorrow!

PR #66901 (2023-12-24): posix: pthread: support stack sizes larger than 65k
A previous size optimization capped the `pthread_attr_t` stacksize property at 65536. Some Zephyr users felt that was not large enough for specific use cases.

Modify `struct pthread_attr` to support large stack sizes by default with the flexibility to allow users to vary the number of bits used for both stacksizes and guardsizes. Additionally, make the public definition easier to parse for non posix areas, and make a private definition that has the correct bitfields defined.

The default guardsize remains zero sinze Zephyr's stack allocators [... cut]
closed 2023-12-24 by cfriedt