zephyrproject-rtos/zephyr #66806: [Coverity CID: 338094] Uninitialized array index read in drivers/bluetooth/hci/spi.c

zephyrbot, 2023-12-21:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/drivers/bluetooth/hci/spi.c

Category: Memory - illegal accesses
Function: `bt_spi_get_header`
Component: Bluetooth
CID: [338094](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=338094)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/bluetooth/hci/spi.c#L220

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_UNINIT). ([CWE-457](http://cwe.mitre.org/data/definitions/457.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

jhedberg, 2023-12-22:
@HoZHel looks like this was introduced by #65172. Could you take a look at it? I'd have assigned this to you, but it seems like you're not (yet) a member of the zephyrproject-rtos GitHub organization.

jhedberg, 2023-12-22:
Here's the detailed report, in case you don't have access:
```
static int bt_spi_get_header(uint8_t op, uint16_t *size)
189{
190        uint8_t header_master[5] = {op, 0, 0, 0, 0};
191        uint8_t header_slave[5];
   	1. var_decl: Declaring variable size_offset without initializer.
192        uint8_t size_offset, attempts;
193        int ret;
194
   	2. Condition op == 11, taking false branch.
195        if (op == SPI_READ) {
196                if (!IS_IRQ_HIGH) {
197                        *size = 0;
198                        return 0;
199                }
200                size_offset = STATUS_HEADER_TOREAD;
   	3. Condition op == 10, taking false branch.
201        } else if (op == SPI_WRITE) {
202                size_offset = STATUS_HEADER_TOWRITE;
203        }
204        attempts = IRQ_HIGH_MAX_READ;
205        do {
   	4. Condition op == 11, taking false branch.
206                if (op == SPI_READ) {
207                        /* Keep checking that IRQ is still high, if we need to read */
208                        if (!IS_IRQ_HIGH) {
209                                *size = 0;
210                                return 0;
211                        }
212                }
213                assert_cs();
214                ret = bt_spi_transceive(header_master, 5, header_slave, 5);
   	5. Condition ret, taking false branch.
215                if (ret) {
216                        /* SPI transaction failed */
217                        break;
218                }
219
   	6. Condition header_slave[0] == 2, taking true branch.
   	
CID 338094 (#1 of 1): Uninitialized array index read (UNINIT)
7. uninit_use: Using uninitialized value size_offset.
220                *size = (header_slave[STATUS_HEADER_READY] == READY_NOW) ?
221                                header_slave[size_offset] : 0;
222                attempts--;
223        } while ((*size == 0) && attempts);
224
```

HoZHel, 2023-12-22:
> @HoZHel looks like this was introduced by #65172. Could you take a look at it? I'd have assigned this to you, but it seems like you're not (yet) a member of the zephyrproject-rtos GitHub organization.

Yes, that's right. I'll do a PR to fix it.

PR #67102 (2024-01-04): drivers: bluetooth: hci: Fix uninitialized array index, issue #66806
Fix issue #66806 related to uninitialized array index, caused by PR #65172.

Fixes #66806
closed 2024-01-04 by carlescufi