zephyrproject-rtos/zephyr #66861: posix: implement `environ`

cfriedt, 2023-12-22:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

The `environ` global variable is required as part of `POSIX_SINGLE_PROCESS` [Subprofiling Option Group](https://pubs.opengroup.org/onlinepubs/9699919799/xrat/V4_subprofiles.html).

https://pubs.opengroup.org/onlinepubs/9699919799/functions/environ.html

moonlight83340, 2023-12-22:
Would like to try it but where should I start ?
I'm not sure on how it's implement in POSIX.

Sh0g0-1758, 2024-01-19:
hey @cfriedt , I am working on implementing the getenv and setenv issues. I would like to take this one up too. Can you please assign me it.

kartben, 2024-01-19:
> hey @cfriedt , I am working on implementing the getenv and setenv issues. I would like to take this one up too. Can you please assign me it.

There is already a PR #66762 so I would recommend you provide feedback there :)

closed 2024-03-08 by cfriedt