zephyrproject-rtos/zephyr #66870: net_buf_append_bytes() usage may cause a failed assertion

konradderda, 2023-12-22:
**Describe the bug**
Depending on the number of bytes provided as `net_buf_append_bytes()`'s it can allocate additional buffers under the hood. However, when the buffer pool is of type 'fixed' the default allocation method within this function calls `net_buf_alloc_len()` which tries to allocate a given number of bytes and it contains an assertion which checks whether the allocated size is bigger or equal to requested one. 
```
#if __ASSERT_ON
		size_t req_size = size;
#endif
		timeout = sys_timepoint_timeout(end);
		buf->__buf = data_alloc(buf, &size, timeout);
		if (!buf->__buf) {
			NET_BUF_ERR("%s():%d: Failed to allocate data",
				    func, line);
			net_buf_destroy(buf);
			return NULL;
		}

#if __ASSERT_ON
		NET_BUF_ASSERT(req_size <= size);
#endif
```
For fixed buffer pools this assertion can be hit quite easily if the number of bytes provided to `net_buf_append_bytes()` fills more than 2 fixed buffers.

**To Reproduce**
The following snippet presents the test case:
```
NET_BUF_POOL_FIXED_DEFINE(fixed_pool, 4, 64, 0, NULL);

int test(void) {
        struct net_buf *buf;
        uint8_t data[256];

        buf = net_buf_alloc(&fixed_pool, K_NO_WAIT);
        net_buf_append_bytes(buf, sizeof(data), data, K_NO_WAIT, NULL, NULL);
}
```

**Expected behavior**
`net_buf_append_bytes()` should allocate multiple `struct net_buf`s without assertion.

**Impact**
Current implementation may cause unexpected faults, e.g. while processing data from the network.

**Additional context**
This issue has a workaround - user can provide `allocate_cb` argument while calling `net_buf_append_bytes()` and provide a custom allocator but one will probably find out after debugging this issue.

jukkar, 2023-12-22:
@konradderda good catch! Would you be able to submit a fix for this?

konradderda, 2023-12-22:
It was my idea at first but it's a bit tricky (if my understanding is correct):
The default allocation goes by:
```
                        /* Allocate from the original pool if no callback has
			 * been provided.
			 */
			pool = net_buf_pool_get(buf->pool_id);
			frag = net_buf_alloc_len(pool, len, timeout);
```
The `net_buf_alloc_len()` contains the assertion which is triggered when `data_alloc()` inside it returns less raw bytes than requested. At the moment, there is no way to determine that the given pool's type.

The possible solutions I am thinking off:
1. Change the public function `net_buf_alloc_len()` to provide a chain of buffers when needed (but it feels a little wrong)
2. Provide some kind of additional information - either like 'is_fixed' flag or an additional pointer to buffer allocation

jhedberg, 2024-01-17:
@konradderda @jukkar I took a quick look at the related code. Perhaps `struct net_buf_data_alloc` needs another member to indicate the maximum size that can be allocated at one time, which would then be used by `net_buf_append_bytes()` to make decisions on how to fragment the buffers. I suppose that would also mean removing the `data_size` member from `struct net_buf_pool_fixed`, since it'd become redundant. Anyway, this is just my first thoughts after spending a short time analysing the issue.

konradderda, 2024-02-01:
@jhedberg  @jukkar I've opened the PR: https://github.com/zephyrproject-rtos/zephyr/pull/68413

PR #68413 (2024-02-06): net: buf: resolve issues with fixed size network buffer pool
This PR fixes issue discussed in https://github.com/zephyrproject-rtos/zephyr/issues/66870 (fix proposed by @jhedberg) with new test case. 
Additionally, it resolves an additional issue where fixed buffer would set it's size to requested length instead of the actual size (with additional check in existing unit test) .
 
Fixes #66870