zephyrproject-rtos/zephyr #66880: doc: Documentation python requirements will not install on Ubuntu 23.10

byzantic, 2023-12-22:
**The python requirements step fails when setting up a workspace on Ubuntu 23.10**

I was trying to follow the getting started section on setting up a `west` workspace, on Ubuntu 23.10. Note that I was using the `pipenv` tool rather than using `venv`, since `pipenv` is now the officially approved tool for organising environments.

The process failed at the requirements install stage. I.e. at the last command of the following sequence:
```
mkdir zephyrproject
pipenv --python 3 install west
pipenv run west init .
pipenv run west update
pipenv run west zephyr-export
pipenv run install -r zephyr/scripts/requirements.txt
```
The failure occurred when trying to install `python-magic-bin` from the error messages it looks as though this requires python <= 3.7, whereas the Ubuntu default is 3.11. I _did_ try to restrict the python version using the `--python` flag to `pipenv`, but couldn't seem to make this work.

If the version restriction is a hard limit for the development envrionment, I think it would be worthwhile mentioning this.

**Environment**

* Page: `develop/getting_started/index`
* Version: main
* SHA-1: e91040917c4ea23008df0b544f3837705086985b

kartben, 2023-12-22:
I am not sure if this is `pipenv` related, but I was able to install everything just fine using the Getting Started instructions on ubuntu:23.10 Docker image. 
Given our minimal supported Python version is 3.8, "the error messages it looks as though this requires python <= 3.7" sounds rather weird. I am not familiar with pipenv at all, but shouldn't you be doing `pipenv shell` at some point? and shouldn't the last line in your sequence be:  `pipenv install -r zephyr/scripts/requirements.txt`?

byzantic, 2023-12-22:
Hi Kartben,

Thanks for responding. Err .. you are quite correct, the last line should be `pipenv install -r zephyr/scripts/requirements.txt`. You can either use `pipenv shell` and then all subsequent commands use the environment, or `pipenv run <command>` will run the rest of the command line within the envrionment.

This was the output from the requirements install:
```
pipenv install -r zephyr/scripts/requirements.txt
Requirements file provided! Importing into Pipfile...
Pipfile.lock (8deb4a) out of date, updating to (47611a)...
Locking [packages] dependencies...
Building requirements...
Resolving dependencies...
Resolving dependencies...
✘ Locking Failed!
⠧ Locking...
CRITICAL:pipenv.patched.pip._internal.resolution.resolvelib.factory:Ignored the following versions that require a different python version: 1.9.5 Requires-Python >=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <3.7
CRITICAL:pipenv.patched.pip._internal.resolution.resolvelib.factory:Could not find a version that satisfies the requirement python-magic-bin (from versions: none)
[ResolutionFailure]:   File "/usr/lib/python3/dist-packages/pipenv/resolver.py", line 704, in _main
[ResolutionFailure]:       resolve_packages(
[ResolutionFailure]:   File "/usr/lib/python3/dist-packages/pipenv/resolver.py", line 662, in resolve_packages
[ResolutionFailure]:       results, resolver = resolve(
[ResolutionFailure]:       ^^^^^^^^
[ResolutionFailure]:   File "/usr/lib/python3/dist-packages/pipenv/resolver.py", line 642, in resolve
[ResolutionFailure]:       return resolve_deps(
[ResolutionFailure]:       ^^^^^^^^^^^^^
[ResolutionFailure]:   File "/usr/lib/python3/dist-packages/pipenv/utils/resolver.py", line 1167, in resolve_deps
[ResolutionFailure]:       results, hashes, markers_lookup, resolver, skipped = actually_resolve_deps(
[ResolutionFailure]:       ^^^^^^^^^^^^^^^^^^^^^^
[ResolutionFailure]:   File "/usr/lib/python3/dist-packages/pipenv/utils/resolver.py", line 948, in actually_resolve_deps
[ResolutionFailure]:       resolver.resolve()
[ResolutionFailure]:   File "/usr/lib/python3/dist-packages/pipenv/utils/resolver.py", line 690, in resolve
[ResolutionFailure]:       raise ResolutionFailure(message=str(e))
[pipenv.exceptions.ResolutionFailure]: Warning: Your dependencies could not be resolved. You likely have a mismatch in your sub-dependencies.
  You can use $ pipenv run pip install <requirement_name> to bypass this mechanism, then run $ pipenv graph to inspect the versions actually installed in the virtualenv.
  Hint: try $ pipenv lock --pre if it is a pre-release dependency.
ERROR: No matching distribution found for python-magic-bin
```
Yes, I would have been surprised if you required a python version < 3.7! Looking at the error messages, it looks like the resolver is part of `pipenv`, so I will try the force-fix they suggest and report back

byzantic, 2023-12-22:
Confirmed that this is an issue with `pipenv` which doesn't seem to respect platform markers. Looks like it might be  a regression of https://github.com/pypa/pipenv/issues/2378 or maybe caused by the use of nested requirements file.

Managed to get this working by commenting out the offending lines in `zephyr/scripts/requirements-base.txt` and `zephyr/scripts/requirements-compliance.txt`. However, note that every time you run the command `pipenv install -r requirements.txt` it places the requirement in `Pipfile` cumulatively, so you need to purge these from the Pipfile.

kartben, 2024-01-09:
@byzantic are you ok if we turn this issue into an enhancement (i.e. "document the use of pipenv"). It's not really a bug in the documentation since you were not really following the recommended instructions ;) Would be happy to review a PR though!

BTW, what do you mean by "since pipenv is now the officially approved tool for organising environments"? https://docs.python.org/3.12/installing/index.html#key-terms is still referring to pip and venv?

kartben, 2024-01-10:
Actually, I am tempted to just close this -- I couldn't find any official statement that pipenv is in any way an "official" or "recommended" tool, so it would probably make sense to stick to supporting/recommending the use of pip + venv for now.

byzantic, 2024-01-10:
Sure, happy to close it; as you say, your documentation stipulates the use of `pip` and `env`.

OK, it's probably not _mandated_ to use `pipenv`, but `pipenv` is the officially endorsed way of controlling environments, and `Ubuntu`, as of 23.10, forces you to use user environments. Having said that, the problem I encountered is definitely a bug in `pipenv`, which means that it is difficult to ensure backwards compatibility using `pipenv`.

Sigh. One can't help feeling a bit despondent at the state of Python environment control, but of course this is no fault of yours.

closed 2024-03-12 by nashif