zephyrproject-rtos/zephyr #66900: MCUBoot Build Fails with slot1 on External Flash

xudongzheng, 2023-12-22:
**Describe the bug**

MCUBoot fails to build when slot1 is on external flash.

**To Reproduce**

```
cd samples/application_development/sysbuild/with_mcuboot && west build -b pinetime_devkit0 --sysbuild --domain mcuboot
```

This results in the following build error:

```
CMake Error at CMakeLists.txt:396 (if):
  if given arguments:

    "4096" "GREATER"

  Unknown arguments specified

-- Configuring incomplete, errors occurred!
```

I added a couple of print messages to https://github.com/mcu-tools/mcuboot/blob/12e2b6398b8a268b43ee9f9efa492be99250fb8c/boot/zephyr/CMakeLists.txt#L396

```
+        message("slot0_flash ${slot0_flash}")
+        message("slot1_flash ${slot1_flash}")
+        message("erase_size_slot0 ${erase_size_slot0}")
+        message("erase_size_slot1 ${erase_size_slot1}")
+        message("write_size_slot0 ${write_size_slot0}")
+        message("write_size_slot1 ${write_size_slot1}")
```

This shows the following:

```
slot0_flash /soc/flash-controller@4001e000/flash@0
slot1_flash /soc/spi@40004000/xt25fb32@0
erase_size_slot0 4096
erase_size_slot1
write_size_slot0 4
write_size_slot1
```

**Expected behavior**

MCUBoot should build

**Impact**

Using older Zephyr

thedjnK, 2023-12-22:
Is fixed by https://github.com/mcu-tools/mcuboot/pull/1877

ndrs-pst, 2023-12-26:
@xudongzheng, I can confirm that replacing the file `boot/zephyr/CMakeLists.txt` with the version from 
https://github.com/mcu-tools/mcuboot/pull/1877 resolves the issue.
Additionally, since **SPI_NRFX** depends on MULTITHREADING, make sure to set CONFIG_MULTITHREADING=y in `sysbuild/mcuboot.conf` as well."

xudongzheng, 2023-12-27:
Thanks! I can also confirm that 1877 addresses the issue.

closed 2023-12-27 by xudongzheng